jgromes/RadioLib #897: Sleep mode + Standby stops transmission of message - Heltec Cubecell

Terrence77, 2023-12-07:
When I put my radio to sleep and then wake it by using standby, it no longer sends a message, or I should say my receiver not longer receives the message.

**platformio.ini file**

```
[env:cubecell_AB02A_ant]
framework = arduino
board = cubecell_node
platform = heltec-cubecell
monitor_speed = 115200
lib_deps =
  jgromes/RadioLib @ ^6.2.0
build_src_filter = 
	+<SX126x_Transmit.cpp>
```

**BuildOptUser.h**
Set to verbose, but not much verbosity.

```
#if !defined(_RADIOLIB_USER_BUILD_OPTIONS_H)
#define _RADIOLIB_USER_BUILD_OPTIONS_H

//#define RADIOLIB_DEBUG    (1)
#define RADIOLIB_VERBOSE  (1)

#endif

```

**Output**

```
Processing cubecell_AB02A_ant (framework: arduino; board: cubecell_node; platform: heltec-cubecell)
---------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/heltec-cubecell/cubecell_node.html
PLATFORM: Heltec CubeCell (1.1.0) > Heltec CubeCell-1/2AA Node (HTCC-AB02A)
HARDWARE: ASR6502 48MHz, 16KB RAM, 128KB Flash
PACKAGES: 
 - framework-arduinocubecell @ 1.6.0 
 - tool-cubecellelftool @ 0.0.1 
 - tool-cubecellflash @ 0.0.1 
 - tool-cubecellflash6601 @ 0.0.1 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RadioLib @ 6.3.0
Building in release mode
...
Linking .pio\build\cubecell_AB02A_ant\firmware.elf
Checking size .pio\build\cubecell_AB02A_ant\firmware.elf
Building .pio\build\cubecell_AB02A_ant\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.9% (used 1784 bytes from 16384 bytes)
Flash: [===       ]  34.1% (used 44660 bytes from 131072 bytes)
Building .pio\build\cubecell_AB02A_ant\firmware.cyacd
Configuring upload protocol...
AVAILABLE: serial
CURRENT: upload_protocol = serial
Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\cubecell_AB02A_ant\firmware.cyacd
Initialising bootloader.
Silicon ID 0x256a11b5, revision 0.
Verifying rows.
Array 0: first row 34, last row 511.
Starting upload.
...
Checksum verifies OK.
Rebooting.
Total upload time 6.63s
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
before setupradio
SetupRadio Initializing ... SetupRadio success!
Sending first packet ...
after setupradio
```

**Here is the code.**
I have tried all 3 stand by modes, but not luck
I also noted that I have to manually set transmittedFlag back to true for it to fall into if statement.

```
#include <RadioLib.h>
//
//this is for the CubeCell !!!! not the heltec_wifi_lora_32_V3
//
SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

int transmissionState = RADIOLIB_ERR_NONE;
volatile bool transmittedFlag = false;
int msgCount = 0;
int loopDelay = 2500;

void SetupRadio();
void SendMessage();
void setFlag(void);

void setup() {
  Serial.begin(115200);
  delay(3000); //take a break while serial gets it's act together.
  Serial.println("before setupradio");
  SetupRadio();
  Serial.println("after setupradio");
}

void loop() {
  Serial.println("loop.................");

  SendMessage();
  Serial.println("after sendmessage");

  radio.sleep();
  delay(loopDelay);

  radio.standby();
  //radio.standby(RADIOLIB_SX126X_STANDBY_RC, true);
  //radio.standby(RADIOLIB_SX126X_STANDBY_XOSC, true);
  transmittedFlag = true;
}

void SendMessage()
{
  if(transmittedFlag) {
    transmittedFlag = false;

    if (transmissionState == RADIOLIB_ERR_NONE) {
      Serial.println("\ttransmission finished!\r\n");
    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }

    radio.finishTransmit();
    Serial.print(F("Sending: "));
    String msg = "Hello! #" + String(msgCount++);
    transmissionState = radio.startTransmit(msg);
    Serial.println(msg);
  }

}
void SetupRadio()
{
  Serial.print(F("SetupRadio Initializing ... "));
  int state = radio.begin(908.0, 500.0, 10, 5, 0x12, 10, 8);

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("SetupRadio success!"));
  } else {
    Serial.print(F("SetupRadio failed, code "));
    Serial.println(state);
    while (true);
  }

  radio.setPacketSentAction(setFlag);
  Serial.println(F("Sending first packet ... "));
  transmissionState = radio.startTransmit("Hello!");
}
void setFlag(void) {
  transmittedFlag = true;
}

```

tpspencer, 2023-12-07:
Thank you for those suggestions.

Terrence77, 2023-12-09:
At your advice, I started over on the original example code and adding sleep work as it should.

I have some work to do on my wrapper class to get it working, but I have the underlying code that works.

Thank you for the great RadioLib.

For any interested here is the working code.

```
#include <RadioLib.h>

SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);
int transmissionState = RADIOLIB_ERR_NONE;
volatile bool transmittedFlag = false;

void setFlag(void) {
  transmittedFlag = true;
}

void setup() {
  Serial.begin(9600);
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin(908.0, 500.0, 10, 5, 0x12, 10, 8);

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  radio.setPacketSentAction(setFlag);

  Serial.print(F("[SX1262] Sending first packet ... "));
  transmissionState = radio.startTransmit("Hello!");
}

int count = 0;

void loop() {

  if(transmittedFlag) {

    transmittedFlag = false;

    if (transmissionState == RADIOLIB_ERR_NONE) {
      Serial.println(F("transmission finished!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }
    radio.finishTransmit();

    radio.sleep();
    delay(2500);
    radio.standby();
   
    Serial.print(F("Sending another packet ... "));
    String str = "Hello! #" + String(count++);
    Serial.println(str);
    transmissionState = radio.startTransmit(str);
  }
}
```

jgromes, 2023-12-09:
Glad you were able to resolve it!

closed 2023-12-09 by jgromes