jgromes/RadioLib #898: Semtech SX1233

victorfmin66, 2023-12-07:
Hello,
I would like to know if you can support in the future the Semtech SX1233 chip. The particularity of this transceiver is that it can have a bitrate up to 600kbps. This is approximately between 5 to 10 times higher than other transceivers such as SX1262/8. There are multiple modulations available such as (G)FSK (G)MSK or OOK that would be interesting to implement.

Thank you in advance. 

Victor

jgromes, 2023-12-07:
RadioLib already supports SX1231, browsing through the datasheet I would say that SX1233 is just a different version of that, so it should be pretty straight-forward to add the support. However, I don't have the SX1233 transceiver, so I can't do any testing. 

@victorfmin66do you have access to the hardware to do some tests?

victorfmin66, 2023-12-07:
Yes, I have designed a PoC with the SX1233 and Arduino. I would be very happy to help with the testing if you are good with that. I'm open to design other hardware PoC's with different microcontrollers such as an ESP32 or ESP8266 if required.

victorfmin66, 2023-12-07:
I have the Arduino Mega with the ATmega 2560. Is there any microcontroller that you think would be better to test with? I have ESP32, ESP8266 and Raspberry pi pico.

jgromes, 2023-12-09:
@victorfmin66 I compared the SX1231 and SX1233 datasheets and it looks like the high bit rate is the only difference between the two. I added support for SX1233 including the high bit rate mode, let me know if you get a chance to test it.

closed 2023-12-09 by jgromes