jgromes/RadioLib #907: ESP32 SX1262 no OTAA answer

derGraph, 2023-12-19:
**Sketch that is causing the module fail**

```c++
#include <RadioLib.h>
#include <ss_oled.h>

static uint8_t ucBackBuffer[1024];

#define SDA_PIN 17
#define SCL_PIN 18
// Set this to -1 to disable or the GPIO pin number connected to the reset
// line of your display if it requires an external reset
#define RESET_PIN 21
// let ss_oled figure out the display address
#define OLED_ADDR -1
// don't rotate the display
#define FLIP180 0
// don't invert the display
#define INVERT 0
// Bit-Bang the I2C bus
#define USE_HW_I2C 0

// Change these if you're using a different OLED display
#define MY_OLED OLED_128x64
#define OLED_WIDTH 128
#define OLED_HEIGHT 64

// SX1262 has the following connections:
// NSS pin:   8
// DIO0 pin:  13
// RESET pin: 12
// DIO1 pin:  14
SX1262 radio = new Module(8, 13, 12, 14);

// create the node instance on the EU-868 band
// using the radio module and the encryption key
// make sure you are using the correct band
// based on your geographical location!
LoRaWANNode node(&radio, &EU868);
SSOLED ssoled;
String monitor[8] = {"1","2","3","4","5","6","7","8"};

void Oled_println(String);
void Oled_print(String);

void setup() {
  Serial.begin(9600);

  int rc;
  rc = oledInit(&ssoled, MY_OLED, OLED_ADDR, FLIP180, INVERT, USE_HW_I2C, SDA_PIN, SCL_PIN, RESET_PIN, 400000L); // use standard I2C bus at 400Khz
  

  // initialize SX1278 with default settings
  Oled_println("[SX1262] Initializing ... ");
  int state = radio.begin(868.0, 125.0, 12, 7, RADIOLIB_SX126X_SYNC_WORD_PRIVATE, 17, 8, 1.6, false);

  if(state == RADIOLIB_ERR_NONE) {
    Oled_println("success!");
  } else {
    Oled_print("failed, code ");
    Oled_println(String(state));
    while(true);
  }

  // application identifier - pre-LoRaWAN 1.1.0, this was called appEUI
  // when adding new end device in TTN, you will have to enter this number
  // you can pick any number you want, but it has to be unique
  uint64_t joinEUI = 0x0000000000000000;

  // device identifier - this number can be anything
[... 618 lines cut ...]
RadioLib Debug Info
Version:  6.3.0.0
Platform: ESP32
Compiled: Dec 19 2023 20:25:26

Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.

M       SX126x
success!
[LoRaWAN] Attempting over-the-air activation ... 

Channel frequency UL = 868.099976 MHz
DR 72: LORA (SF: 10, BW: 125.000000, CR: 5)
Timeout in 556032 us
Join-request sent <-- Rx Delay start

Channel frequency DL = 868.099976 MHz
DR 72: LORA (SF: 10, BW: 125.000000, CR: 5)
Opening Rx1 window (226848 us timeout)... <-- Rx Delay end 
closing
```
Is the problem that I am not in range (I have tested previously 10m to a Gateway from TTN-Mapper) or that something isn't working. 

Thanks!

jgromes, 2023-12-20:
> GPIO pre-transfer timeout, is it connected?
GPIO pre-transfer timeout, is it connected?

This suggests there could be some hardware issue, though the module itself seems to be communicating (mostly) fine. Other than that, you seem to always be wipign the persistent storage. That will lead to reuse of previous nonces, which your application server might be rejecting (just a guess though). From RadioLib end, everything seems to be correct. The only suspicious thing is:

```
// SX1262 has the following connections:
// NSS pin:   8
// DIO0 pin:  13
// RESET pin: 12
// DIO1 pin:  14
SX1262 radio = new Module(8, 13, 12, 14);
```

Is the pinout actually correct for SX1262? The comment seems to be copied from some SX127x module, since SX126x does not have DIO0 pin.

derGraph, 2023-12-20:
I just edited the comments to match my config. DIO0=Busy
And how do I change the SF in the otaa sequence?

jgromes, 2023-12-20:
> DIO0=Busy

So your constructor actually looks like this?

`SX1262 radio = new Module(NSS, BUSY, RESET, DIO1);`

If that's the case, then you have BUSY and DIO1 swapped. Since DIO1 is used to detected received packets, that could explain why you don't get a reply.

> And how do I change the SF in the otaa sequence?

It's stated in the example:

```
  // a specific starting-datarate can be selected in dynamic bands (e.g. EU868):
  /* 
    uint8_t joinDr = 4;
    state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey, joinDr);
  */
```

You will need to consult the LoRaWAN regional specification to find out which SF corresponds to a specific data rate value.

derGraph, 2023-12-20:
I tried changing the pins and have removed the `node.wipe()` now getting error code -6, will try near a lora gateway tomorrow...

derGraph, 2023-12-31:
Ok, so I got a connection to a gateway of a friend of mine, it looks like this:
[image]
but i dont recieve an answer. How do i have to Format the nwkKey and the appKey (LSB, MSB ...) from the TTN Console for my device.
Thanks

jgromes, 2023-12-31:
The gateway just forwards this to the TTN servers, so the more interesting information will be found there. If it's dropping the join request for some reason, it should report it (you might have to enable some verbose mode in the node console).

> How do i have to Format the nwkKey and the appKey (LSB, MSB ...) from the TTN Console for my device.

It should be the exact same as what you set in TTN.

derGraph, 2023-12-31:
Can you please tell me if i have this translated correctly (I will reset the device after):
[image]
[image]

jgromes, 2023-12-31:
I fail to understand why you swapped the order of the keys in the byte. Like I said above, it has to be exactly the same.

Also JoinEUI 000... is pretty weird, though I guess not strictly forbidden.

derGraph, 2023-12-31:
I tried putting the byte in the other order but it didn't work.

jgromes, 2023-12-31:
I would suggest trying to find out whgat is the cause for this behavior, instead of guessing. The byte order in keys really isn't the culprit here.

> In the console i recieved this

Judging by the gateway status message, I'm guessing that's the gateway console. You probably won't be able to tell why the end device is not joining from there, since the gateway just forwards messages between the TTN servers and your end devices. What does the end device console say? If you enable verbose stream it should tell you e.g. if some nonces need to be reset.

derGraph, 2023-12-31:
You have correctly assumed the console of the gateway, in the console of the device i cant find anything.

StevenCellist, 2024-01-02:
@derGraph really make sure that your keys match. On the pictures you shared, the devEUIs differ between the TTN console and your code.
You can use the formatting and copy-buttons on the TTN console to just copy-paste the keys, and in that case you cannot make any typos. If the join-request appears in the gateway console, also look at the device console (probably with verbose mode enabled), to see if there is any problem reported there.
But it won't end up at your device console if the devEUI is incorrect, because then it can't match the device which is based on the devEUI...

jgromes, 2024-01-13:
Closing due to inactivity.

closed 2024-01-13 by jgromes