jgromes/RadioLib #909: SX1276::begin hangs after repeated calls

ststolz, 2023-12-20:
**Check the wiki**

Wiki states that method SX1276::begin

> Must be called at least once from Arduino sketch to initialize the module.
> - (https://jgromes.github.io/RadioLib/class_s_x1276.html#ac0f792c2fee6aac9b554104c5b1e5ae7)

**Describe the bug**

After repeated call, program hangs in method SX1276::begin. I call it every time there is some error with sending or receiving to be sure it is set up. What I read from Wiki, this should be ok. 

**To Reproduce**

Call SX1276::begin in a loop until code hangs.

**Expected behavior**

I can call SX1276::begin as often I want.

**Additional info (please complete):**
 - MCU: xmc2go Infineon
 - Wireless module type RFM95
 - Arduino IDE version: platformio
 - Library version jgromes/RadioLib@6.1.0

jgromes, 2023-12-20:
As I stated in #910, you are using an unsupported MCU. Despite that, debug+verbose (as requested by the issue template) could be useful to figure what exactly is happening, since the description is very vague.

ststolz, 2023-12-20:
After further investigation, found that the bug was not within SX1276::begin and has nothing to do with RadioLib. So closing.

closed 2023-12-20 by ststolz