jgromes/RadioLib #910: EEPROM.h: No such file or directory on 6.3.0

ststolz, 2023-12-20:
**Describe the bug**

Building 6.3.0 for board xmc1100_xmc2go, [platform infineonxmc](https://registry.platformio.org/platforms/infineon/infineonxmc) gives:

> .pio/libdeps/xmc1100_xmc2go/RadioLib/src/ArduinoHal.cpp:6:10: fatal error: EEPROM.h: No such file or directory

**To Reproduce**

Build some code on this platform with lib >= 6.2.0

**Expected behavior**

Should build.

**Additional info (please complete):**
 - MCU: xmc2go Infineon
 - Wireless module type RFM95
 - Arduino IDE version: platformio
 - Library version jgromes/RadioLib@6.3.0

jgromes, 2023-12-20:
Infineon XMC is not supported by RadioLib - see the list of supported platforms in README. Had you enabled debug output, you would have seen that during compilation, RadioLib probably reports you are using an unknown platform. Because RadioLib is missing the platform definition, it also doesn't know that XMC doesn't provide an EEPROM library, hence the error.

jgromes, 2024-01-05:
Closed due to inactivity.

closed 2024-01-05 by jgromes