jgromes/RadioLib #913: LoRaWAN: implement LinkCheck and TimeReq

StevenCellist, 2023-12-22:
Currently, the two MAC commands that should be available to a LoRaWAN enduser are not implemented (LinkCheck to retrieve margin and gateway count, and TimeReq to hear the retrieve datetime).
I plan to add this in early January.

StevenCellist, 2024-01-02:
[image]
Two birds with one stone.. requesting the device to send the MAC requests using the Verification suite that I am also implementing. Which can manipulate the device through downlink commands. More later...

jgromes, 2024-01-13:
@StevenCellist this was resolved by merging #918 and can be closed, right?

StevenCellist, 2024-01-13:
Yes! Fully working.

closed 2024-01-13 by StevenCellist