jgromes/RadioLib #920: Si4063 Support Recommendation

maxsrobotics, 2024-01-07:
Wanted to suggest support for the Si4063 module from Silicon Labs. The datasheet can be found here: https://www.silabs.com/documents/public/data-sheets/Si4063-60-C.pdf. Many projects use it, including RS41ng from Project Horus, and the pAVAR9 project. It includes low-power management and sleep modes, which is always handy. It would be helpful as is supported until 2033 and not many Arduino libraries are made for it, so this could attract popularity.

Just an idea to consider, not a burden!

Thanks,
Max

jgromes, 2024-01-07:
Looks like an interesting little transceiver, do you happen to know of any development kits? I have an Si4463 kit off Ebay, but don't know what are the major differences between Si4463 and Si4063. There's also an open issue for Si4463 support (#655), so if the implementation is smart enough (Si4x63?) that would be two chips supported for the price of one ;)

At the moment, I'm working on LR11x0 support, so if anyone is willing to help out with this, that would be amazing.

maxsrobotics, 2024-01-07:
From what I'm seeing in the programming interface (https://www.farnell.com/datasheets/1889753.pdf), it seems as though the Si446x is mostly compatible with the Si4063. Page 2 also shows test dev boards. The programming interface also shows a detailed reference for the signals.