jgromes/RadioLib #924: BW and FHSS problem

AlbertoManske, 2024-01-11:
Hi, I'm facing a curious problem when using the FHSS example with an RFM95W radio, in the following radio configurations: radio.begin(channels[0], 62.5, 11, 7, 0x34, 20, 12, 6);
I'm sending 20 bytes to a board that returns another 20 bytes, always the same value because I was testing the library, it happened that the bytes were wrong sometimes, one board I'm using STM32 and another board ESP32, to make sure it wasn't a or another board with a problem, I placed a third one to listen to both messages, and the following happened, board A sent it to B, B received it with errors, but sometimes board C received it without any error, and the opposite also happened, so the Instability wasn't just on one board.
I was testing various things, parameters and such, but it was when I changed the BW that I saw the problem being resolved, for some reason when I change to 250khz or 62.5khz the problem stops, but with 125khz the problem occurs.
Do you happen to have any idea what might be going on here?

jgromes, 2024-01-12:
In the first sentence you stated that you are using 62.5 kHz and you are seeing the issue; then in the last paragraph you said that it only happens with 125 kHz. So which one is it?

> in the following radio configurations: radio.begin(channels[0], 62.5, 11, 7, 0x34, 20, 12, 6);

You seem to be using the LoRaWAN sync word for your non-LoRaWAN application. Also you set the gain to 6, which is the lowest gain setting - why? Can you show your code?

Generally I don't see a reason why this would be happening with any given bandwidth, apart from the fact that more narrow bands will place higher demands on the oscillator stability since the margin for frequency error from the crystal will be more narrow.

jgromes, 2024-01-26:
Closed due to inactivity.

closed 2024-01-26 by jgromes