jgromes/RadioLib #925: SX1278 Low Battery Action

Brynod, 2024-01-11:
Good evening,

First of all, many thanks for your excellent work with this library! The extensive examples have made this library a pleasure to use.

Looking at the datasheet for the SX1278, it appears that it is possible to set an action for "LOW BATTERY" warning. I have looked through the library and searched the wiki, but i can find no relevance to any kind of "SET_DIO_ACTION" for this event. 

Has this feature been implemented in this library? If not, is there a way in whihc i can write a register to enable DIO0 to change state when this low battery condition is reached?

Apologies if this has already been answered...

Many thanks in advance.

Regards
Brynod

jgromes, 2024-01-12:
At the moment, this is not implemented - simply because there was no such request. Although it may be a good idea to add this, since other modules like SX126x implement some sort of low battery indicator interrupt as well if I recall correctly.

> is there a way in which I can write a register to enable DIO0 to change state

Yes, RadioLib has a godmode which can be enabled by defining `RADIOLIB_GODMODE` e.g. in `src/BuildOptUser.h`. When enabled, it will allow you access to all low-level functions including raw register writes, using something like this:

```c++
Module* mod = radio.getMod();
mod->SPIsetRegValue(address, value); // see Module class documentation for details
```

Brynod, 2024-01-12:
Many thanks for the reply sir! I will give this a try over the weekend and report back with my findings

Thanks again for the excellent library!

jgromes, 2024-01-13:
@Brynod FYI I added this as `setLowBatteryThrehold(level, pin)`. Example usage: `radio.setLowBatteryThrehold(RADIOLIB_SX127X_LOW_BAT_THRESHOLD_2_045_V, 0);` to enable 2.045 V detector at pin DIO0. One caveat: the datasheet is not entirely clear whether this functionality is only available for FSK modem. Please let me know if there are any issues.

closed 2024-01-13 by jgromes