jgromes/RadioLib #948: RadioLib  6.4.1 LoRaWAN isseu

nmaas87, 2024-01-28:
**Describe the bug**
Short feedback for you (@StevenCellist / @jgromes) and the PR #942 , I tried a STM32WL55 module with the latest 6.4.1 patch, as well as with 6.4.0 and 6.3.0 in EU868 state with the LoRaWAN reference example.

6.3.0 - works flawlessly
6.4.0 - node.restore() does not work, but initial joining/OTAA does work
6.4.1 - initial joining/OTAA does not work with reference to the region code / status / error code:

````
[LoRaWAN] Resuming previous session ... failed, code -1101
[LoRaWAN] Attempting over-the-air activation ... failed, code -1106
````

I have not tried other modules yet, but maybe there is a bigger issue here at hand? 

Debug output below:

````
[STM32WLx] Initializing ... 
RadioLib Debug Info
Version:  6.4.1.0
Platform: Arduino STM32 (official)
Compiled: Jan 28 2024 12:17:58

CMDW	80	
SI	0	
SO	AA	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	A2	
CMDR	1D	3	20	
SI	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
SO	A2	53	58	31	32	36	31	20	54	4B	46	20	31	41	31	31	0	
Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 54 4b 46 20 31 41 31 31 00 | SX1261 TKF 1A11.

M	SX126x
CMDW	80	
SI	0	
SO	AA	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	A2	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	A2	
CMDW	80	
SI	0	
SO	A2	
CMDR	C0	
SI	0	
SO	A2	
CMDR	17	
SI	0	0	0	
SO	A2	0	20	
CMDR	C0	
SI	0	
SO	A2	
CMDW	7	
SI	0	0	
SO	A2	A2	
CMDR	C0	
[... 523 lines cut ...]
  }

  // on EEPROM enabled boards, you can save the current session
  // by calling "saveSession" which allows retrieving the session after reboot or deepsleep
  node.saveSession();
 

  // wait before sending another packet
  uint32_t minimumDelay = 60000;                  // try to send once every minute
  uint32_t interval = node.timeUntilUplink();     // calculate minimum duty cycle delay (per law!)
	uint32_t delayMs = max(interval, minimumDelay); // cannot send faster than duty cycle allows

  delay(delayMs);
}
````

**Expected behavior**
Starting connection to LoRaWAN as usual

**Additional info (please complete):**
 - MCU: [STM32WL55 NUCLEO-64]
 - Link to Arduino core: [https://github.com/stm32duino/Arduino_Core_STM32, Version 2.7.1]
 - Wireless module type [STM32WL55 NUCLEO-64]
 - Arduino IDE version  [1.8.19]
 - Library version [4.6.1]

StevenCellist, 2024-01-28:
Your debug output appears incomplete, there should be much more before it actually fails with -1106. Also, please remove all (or most of) the SPI mess to keep logs readable and the page scrollable :)

nmaas87, 2024-01-28:
Maybe like that?
Oh yeah, sorry I saw it was missing most of it...

````
[STM32WLx] Initializing ...
RadioLib Debug Info
Version:  6.4.1.0
Platform: Arduino STM32 (official)
Compiled: Jan 28 2024 12:17:58

Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 54 4b 46 20 31 41 31 31 00 | SX1261 TKF 1A11.

success!
[LoRaWAN] Attempting over-the-air activation ... Didn't restore session (checksum: 0, mode: 0)
First 16 bytes of NVM:
0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
Wiping EEPROM and starting a clean session
Setting up dynamic channels
Channel UL/DL 0 frequency =  MHz
Channel UL/DL 1 frequency =  MHz
Channel UL/DL 2 frequency =  MHz
exe MAC CID = 03, len = 4
ADR REQ: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 00, nbTrans = 0
DR 72: LORA (SF: 10, BW: , CR: 5)

ADR failed to configure dataRate = 2!

ADR ANS: status = 0x05
exe MAC CID = 04, len = 1
Max duty cycle: 1/2^7
exe MAC CID = 05, len = 4
Rx param REQ: rx1DrOffset = 0, rx2DataRate = 0, freq =

Rx param ANS: status = 0x07
exe MAC CID = 08, len = 1
RX timing: delay = 1 sec
exe MAC CID = 09, len = 1
TX timing: dlDwell = 0, ulDwell = 0, maxEirp = 16 dBm
exe MAC CID = 0c, len = 1
ADR param setup: limitExp = 6, delayExp = 5
exe MAC CID = 0f, len = 1
Rejoin setup: maxTime = 15, maxCount = 10

There are no channels defined - are you in ABP mode with no defined subband?
failed, code -1106
````

StevenCellist, 2024-01-28:
> ADR failed to configure dataRate = 2!

@jgromes, could you have a look? Still on mobile, haven't had time to look at the actual fix you added.

jgromes, 2024-01-28:
There was an error in the data rate check, my fault. @nmaas87 I pushed a fix, could you try it out?

nmaas87, 2024-01-28:
Thanks @jgromes - trying it right now and getting back to you

nmaas87, 2024-01-28:
Yes, thanks a lot @jgromes / @StevenCellist  - that fixed it :).
Can be closed! :)

jgromes, 2024-01-28:
@nmaas87 thanks for testing it out. Unfortuantely the bug pretty much breaks LoraWAN in 6.4.1, so we will need to release 6.4.2 patch ...

closed 2024-01-28 by jgromes

StevenCellist, 2024-01-28:
Hopefully it's so fast after 6.4.1 that nobody notices / bothers :) and thanks you two!!

altinem, 2024-02-12:
hello,
I have a Heltec wireless stick v3 board. I succesfully initilized SX1262 chip  and OTAA activated the board with TTN. I am getting LORAWAN sending uplink packet... failed code -4. I have looked all over the forum and could not find a solution  why I am getting this error.  I have a sensecap M2 gateway setup as packet forwarder. The gateway is connected to the TTN without issues. Could someone help me why I am getting this error message.

jgromes, 2024-02-12:
@altinem did you at least search for what the -4 error means? It's on this page: https://jgromes.github.io/RadioLib/group__status__codes.html, literally the first result I got when googling "radiolib error codes". In there, you will find that -4 is `RADIOLIB_ERR_PACKET_TOO_LONG`, so my guess is that you attempted to uplink a packet longer than 255 bytes, which the SX126x cannot do. I'm only guessing though since you have not provided the code you are running.

Also, please don't revive old threads that have been resolved and are rather obviously unrelated to your problem. If your problem persists, open a new Issue or Discussion **and include all the necessary information, such as the code you are actually running**.