jgromes/RadioLib #951: ABP does not work on SX1262 EU868 with Platformio - There are no channels defined - are you in ABP mode with no defined subband?

MPMcIntyre, 2024-02-01:
- [Background](#background)
- [Device and library info](#device-and-library-info)
- [Behaviour: "There are no channels defined - are you in ABP mode with no defined subband?"](#behaviour-there-are-no-channels-defined---are-you-in-abp-mode-with-no-defined-subband)
- [Sketch](#sketch)
- [Debug mode output -- Failure on restart](#debug-mode-output----failure-on-restart)

# Background

I have taken the LoRaWAN example and managed to connect to TTN.

I have set up two devices in TTN to test this issue for APB mode and OTAA mode respectively.

Using the LoRaWAN MAC V1.1.0 I ran into the [DevNonce is too smal](https://www.thethingsnetwork.org/forum/t/devnonce-is-too-small-what-to-do/46330) issue, which is not the topic of discussion here

# Device and library info

- LoRaWAN specs: LoRaWAN V1.0.4
- LoRa Network: The Things Network
- Controller: [ESP32 - Devkit V4](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html)
- LoRa module: [WaveShare Lora Node (HF) [SX1262]](https://www.waveshare.com/wiki/Core1262-868M)
- Frequency plan: EU868
- IDE Environment: VSCode + Platformio (Library pulled from platformio)
- RadioLib version: 6.4.0
- Framework:
  - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
  - tool-esptoolpy @ 1.40501.0 (4.5.1)
  - tool-mkfatfs @ 2.0.1
  - tool-mklittlefs @ 1.203.210628 (2.3)
  - tool-mkspiffs @ 2.230.0 (2.30)
  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
- And also:
  - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)

# Behaviour: "There are no channels defined - are you in ABP mode with no defined subband?"

The initial OTAA is successful and successfully transmits data. However, once the device looses power (reset) it tries to connect using the APB mode as it now has the correct keys. This is where the error occurs. I can also confirm this is the default behaviour when using ABP mode directly.

The only way I can get around this is to allow the device in TTN settings to reset the nonces and force the OTAA to run every time it starts up, which is not ideal.

# Sketch

Please note that the ESP sleep function puts it into deep sleep and essentially restarts on wake. This is why it is essential for the connection to re-establish after a power down event.

```c++
#include <Arduino.h>
#include <low_power.h>
#include <Ticker.h> 
// This was needed for compilation issues
#include <RadioLib.h>
#include <pins.h>

// ------------------- SETTINGS -------------------
#define TCXO_PRESENT 1   // Temperature compensated crystal oscillator present
#define USE_OTAA 1       // 1 for OTAA, 0 for ABP
#define SPREAD_FACTOR 9U // LoRaWAN SF
#define SLEEP 0          // Sleep between transmissions
// ------------------- SETTINGS -------------------

SX1262 radio = new Module(loraPins.cs, loraPins.dio1, loraPins.reset, loraPins.busy);
LoRaWANNode node(&radio, &EU868);

#if USE_OTAA
uint64_t deviceEUI = 0x1122333444455555;                                                                               // Redacted
uint64_t joinEUI = 0x1111111111111111;                                                                                 // Redacted
uint8_t nwkKey[16] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}; // Redacted
uint8_t appKey[16] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}; // Redacted
#else
uint32_t devAddr = 0x12345678;
uint64_t deviceEUI = 0x1122333444455555;
uint8_t nwkSKey[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}; // Redacted
[... 587 lines cut ...]
(12:26:32.393) SI	0
(12:26:32.393) SO	22
(12:26:32.393) CMDR	1D	7	36
(12:26:32.393) SI	0	0
(12:26:32.393) SO	A2	D
(12:26:32.393) CMDR	C0
(12:26:32.393) SI	0
(12:26:32.393) SO	22
(12:26:32.393) CMDW	D	7	36
(12:26:32.393) SI	D
(12:26:32.393) SO	A2
(12:26:32.393) CMDW	8C
(12:26:32.393) SI	0	8	0	FF	1	0
(12:26:32.393) SO	A2	A2	A2	A2	A2	A2
(12:26:32.393) CMDR	C0
(12:26:32.393) SI	0
(12:26:32.393) SO	22
(12:26:32.393) [main.cpp:129->setup()]: successfully started node!
(12:26:32.423) [LoRaWAN] Sending uplink packet ... There are no channels defined - are you in ABP mode with no defined subband?
(12:26:32.496)
(12:26:32.496) Channel frequency UL = 0.000000 MHz
(12:26:32.496) failed, code -12
```

Please let me know if I can be of more assistance

StevenCellist, 2024-02-01:
Well.. why not try to solve the DevNonce issue? This in turn will very likely solve the issue you find here: RadioLib is LoRaWAN v1.1 from the ground up! LoRaWAN v1.0.x is available as sort of a fallback but by no means guaranteed - which you currently found out. Maybe if I ever have a lot of spare time I will see if there are small bugs to be fixed to get 1.0.x running, but 1.1 is where it is at.

If you encounter a valid reason for 1.1 to fail working I will happily assist, but unfortunately for you I am not solving 1.0.x issues anywhere soon. BTW: don't just revert to an older version if it doesn't work, better dig in and make it work, because [it can work](https://ttnmapper.org/experiments/?experiment=MapQuester-beta&startdate=2024-01-25&enddate=2024-01-25&gateways=on&lines=on&points=on)!

StevenCellist, 2024-02-01:
Oh, and also, please run RadioLib 6.4.2, 6.4.0 is actually borked w.r.t. LoRaWAN. Switching to 6.4.2 will likely _also_ solve the issue you encountered!

MPMcIntyre, 2024-02-01:
Thanks for your fast reply! Ok, will try to get 1.1 working first then :)

StevenCellist, 2024-02-01:
Quick note that you're probably best off with also re-creating your device on TTN for a fresh set of keys, so that after switching to 6.4.2 the DevNonce counter will be at 0. Then as long as you don't change keys and do call node.saveSession() at appropriate interval (during development probably best after every up/downlink), you _should_ be able to have an easy life. But there are some particularities - devil is in the detail.

HeadBoffin, 2024-02-01:
It looks like you are sort of mixing OTAA and ABP with the same DevEUI and it's not totally clear if you are joining and then refreshing to use ABP but it's pulling the session from NVM. I'm not sure we've tested that combo (that's polite for "no chance we coded that situation"). OTAA on v1.1 works on ESP32 devices. Everything else is unknown/unverified but there are reports that plain ABP works. So if you want to crack on with making a device, I'd stick to OTAA on v1.1. If you want to join the testing fray, that would be cool. But with separate setups on the LNS for the two modes. And ABP will need the f_cnt resets setting for Dev mode.

MPMcIntyre, 2024-02-01:
> It looks like you are sort of mixing OTAA and ABP with the same DevEUI and it's not totally clear if you are joining and then refreshing to use ABP but it's pulling the session from NVM. I'm not [...]

I apologise for the ambiguous code. The device EUI is random as I dont want sensitve info in public, I merely copied it for the APB. My intention was also not to have the device act as both APB or OTAA but simply have a quick setup where I could though in the keys, select a mode and test it. 

I managed to get OTAA v1.1 working on TTN now, will test further. Just to be sure, I added node.saveSession() after each uplink/downlink and on startup a node.restore() after the node is started. The connection seems a bit unstable as I keep recieving -707 errors which then lead to -1108 error code when trying to send the message. Here is a snippet of the debug log

```
(16:00:38.682) Channel frequency UL = 868.299988 MHz

(16:00:38.682) CMDW	98	

(16:00:38.682) SI	D7	DB	

(16:00:38.755) SO	A2	A2	

(16:00:38.755) CMDR	C0	

(16:00:38.755) SI	0	

(16:00:38.755) SO	22	

(16:00:38.755) CMDW	86	

(16:00:38.755) SI	36	44	CC	C0	

(16:00:38.755) SO	A2	A2	A2	A2	

(16:00:38.755) CMDR	C0	

(16:00:38.755) SI	0	

(16:00:38.755) SO	22	

(16:00:38.755) DR 24: LORA (SF: 7, BW: 125.000000, CR: 5)

(16:00:38.755) CMDR	11	

(16:00:38.755) SI	0	0	

(16:00:38.755) SO	A2	1	

(16:00:38.755) CMDR	C0	

(16:00:38.755) SI	0	

(16:00:38.755) SO	22	

(16:00:38.755) CMDR	11	

(16:00:38.755) SI	0	0	

(16:00:38.755) SO	A2	1	

(16:00:38.755) CMDR	C0	

(16:00:38.755) SI	0	

(16:00:38.755) SO	22	

(16:00:38.755) CMDW	8B	

(16:00:38.755) SI	7	4	3	0	

(16:00:38.755) SO	A2	A2	A2	A2	

(16:00:38.755) CMDR	C0	

(16:00:38.755) SI	0	

(16:00:38.755) SO	22	
[... 538 lines cut ...]

(16:00:43.974) SO	A6	

(16:00:43.974) CMDW	98	

(16:00:43.974) SI	D8	DB	

(16:00:43.974) SO	A2	A2	

(16:00:43.974) CMDR	C0	

(16:00:43.974) SI	0	

(16:00:43.974) SO	2A	

(16:00:43.974) CMDW	86	

(16:00:43.974) SI	36	58	66	80	

(16:00:44.003) SO	AA	AA	AA	AA	

(16:00:44.003) failed, code -707
```

I will continue testing, but if you spot something or need more logs let me know.

StevenCellist, 2024-02-01:
Given your custom XTAL configuration and the [matching description](https://github.com/jgromes/RadioLib/blob/547328f375998c663fd88f9b9f1143abc281de6d/src/TypeDef.h#L412C1-L420C70) which you can find, I'm not surprised.

MPMcIntyre, 2024-02-01:
Yeah, the description helped me to fix the issue when it was just the LoRa chip not initialising correctly, hence the custom config, but now the chip does initialise, it gives a -707 when sending and recieving from the network.

I have tested it further, the module successfully sends an uplink to TTN after starting, the -707 comes after the initial message. So for my purpouses it will do as the device basically reboots after deep sleep. But if the sleep is replaced with a delay and the device remains powered on, it will eventually result in a -707 which will prevent any future messages from comming out.

MPMcIntyre, 2024-02-01:
# Logs

1. V1.1 - Delay.log - Logs for the always on, delay for 60 seconds
2. V1.1 - Deep sleep.log - Logs for the deep sleep for 60 seconds, reboot on wake

Notes: 
- Sometimes the -707 error does not occur after the initial uplink for the deep sleep logs, seems to be at random.
- The sensor in question is a simple ultrasonic sensor that uses 2 digital pins, no interference detected 

Let me know if I can help any further.

[v1.1 - Deep sleep.log](https://github.com/jgromes/RadioLib/files/14127765/v1.1.-.Deep.sleep.log)
[v1.1 - Delay.log](https://github.com/jgromes/RadioLib/files/14127767/v1.1.-.Delay.log)

StevenCellist, 2024-02-01:
Given the -707 error and the custom config, I'm 99% certain this is a hardware/crystal issue, not a problem with the library, especially not the LW stack. Maybe @jgromes can suggest something, and I'd suggest converting this to a discussion.

MPMcIntyre, 2024-02-01:
Cool! I do appreciate your input and effort. I will have another dev board to test on tomorrow, also with the sx1262. Hopefully that will resolve the issue, but I will keep you informed.

HeadBoffin, 2024-02-01:
Presumably the TCXO define is because you are swapping between boards, one with and one without? The WaveShare has TCXO so if you only have the one radio, perhaps remove the code that breaks it?

Or if you do have two radios, to make it easier to track down issues, any chance you can run to two separate code bases - or four even, depending on if you want ABP on both boards as well as OTAA?! Or have defines like BOARD_A_ON_OTAA, BOARD_A_ON_ABP etc

**Couple of FYI's:**
> The device EUI is random as I dont want sensitve info in public,
It's sent in plain text during a join and appears in the logs of any gateway that hears it along with the JoinEUI. Only things ending in Key should be kept private.

And 60 seconds isn't within TTN Fair Use Policy

MPMcIntyre, 2024-02-01:
I plan on swapping to a different model in future but the tests were done on the same board. The latest dump with lora v1.1 were all OTAA, however. 

Ah, thanks for the heads up, will add in a button and test on button press instead.

MPMcIntyre, 2024-02-02:
Tested on the new devboard, also with sx1262 and TCXO. Settings are a bit different but I do not recieve the same error and the connection is stable. @StevenCellist  I think you are correct and we can put this down as a hardware issue.

Thanks again for the support!

closed 2024-02-02 by MPMcIntyre

StevenCellist, 2024-02-02:
Glad to hear, good luck on your adventures!

jgromes, 2024-02-02:
A bit late, but here's what I think is actually going on.

The -707 error happens when attempting to change frequency to 869.525024 MHz. This also triggers image rejection calibration. This calibration actually fails, though for some reason this is not caught by the library, hence why it errors during the next SPI transfer.

So, why is this calibration failing, and why don't the previous calibrations fail? Checking previous calibration comands (CMDW 98 in the log), all those that pass have arguments 0xD7, 0xDB. But the ones that fail have 0xD8, 0xDB. These arguments specify the frequency range in which to perform the calibration, and are calculated from the frequency we want to set. The calculation is from LR1110 datasheet, the SX1262 datasheet does not actually specify it, it only has some commonly used bands. The equations for LR1110 yield the same numbers, and so I though it should be safe to use it. So perhaps there are some corner cases that need to be investigated ... or perhaps my assumption that the calibration is the same for LR1110 and SX126x doesn't really hold up. I'll see if I can replicate the issue without LoRaWAN stack.

As a SW developer, I too can be prone to blaming the HW, let's cut it some slack - at least for now ;)

jgromes, 2024-02-03:
So despite that promising lead described above, I was not able to replicate this with a DRF1262G, the module seems to have no issue calibrating and setting the frequency to 869.525024 MHz:

```
CMDW    98
SI              D8      DB
SO      A2      A2      A2
CMDW    86
SI              36      58      66      80
SO      A2      A2      A2      A2      A2
```

So ... hardware issue it is :)