jgromes/RadioLib #960: High % of CRC Errors in SF12 and similar configurations,

Tom-E, 2024-02-10:
Hi, I've been testing different configurations for SX1262 to send+receive, and find that SF12 500khz for example results in a ~20% CRC error rate! If I adjust down to SF9 500khz, that rate will drop to about 6%. If I use lower banwdiths like 62khz and SF5, I can get down to 1% CRC error rate.

Best I understand, the Lilygo T3S3 SX1262 module I'm using has a TCXO and this is what semtech recommends for best performance, and Radiolib leverages it by default.

How can I debug this and better understand how to get lower rate of CRC errors while using configurations like SF12 500khz?

Thank you for any help,

jgromes, 2024-02-11:
High spreading factor combined with high bandwidth (SF12 BW500) is a rather odd configuration, since you are pushing sensitivity up with the high spreading factor, while simulatenously decreasing it with high bandwidth - why would you want to do that? And do you know if the Lilygo module can handle this properly? It most likely uses filters in it RF path, can it handle signal with that bandwidth? My experience is that most of these off-the-shelf modules work OK-ish for common settings (e.g. SF10 BW125), but not for corner cases. And SF12 BW500 is absolutely a corner case.

Another question would be what is this correlated with? High bandwidth? High spreading factor? Both? Some other variable? You only have 3 data points, I don't think that's enough. Ideally you would measure all SF/BW configurations.

I'm assuming all of your test are done on a benchtop as opposed to in the field, since those different configurations will have significantly different range.

> Best I understand, the Lilygo T3S3 SX1262 module I'm using has a TCXO and this is what semtech recommends for best performance

TCXO precision has significant effect on narrow bandwidth LoRa (<62.5 kHz), because in narrow band with cheap crystal, it is easier to have sufficient mismatch between the transmitter and receiver frequencies - IIRC LoRa should be able to tolerate up to 25% BW frequency mismatch (assuming no Doppler shift). TCXO fixes that and allows your device to work in higher temperature range, but that's about it.

> How can I debug this and better understand how to get lower rate of CRC errors while using configurations like SF12 500khz?

I can think of only one thing that is related specifically to 500 kHz bandwidth and that is a fix for receiver sensitvity descrcibed in SX1262 datasheet chapter 15 "Known limitations". RadioLib does apply this fix, so you could try disabling it by commenting out this line:

https://github.com/jgromes/RadioLib/blob/e4267760f97f0447d829286c0f51b9e4748b4d8c/src/modules/SX126x/SX126x.cpp#L558

Other than that, I think you would be better off asking this in Semtech forums, and finding out more information about the hardware used, measuring the signal with with spectrum analyzer/SDR to see if it is being deformed by anything in the RF path etc.

jgromes, 2024-02-26:
Converting to discussion due to not really being an issue (and inactivity).