jgromes/RadioLib #961: SX1278 LoRa exmples compiling. Transmitter emits signal but Receiver getting timeout in a loop

kocinski15, 2024-02-11:
Doing simple test of library usability i tried a pair:
[SX127x_Receive_Blocking]  / [SX127x_transmit_Blocking]  in LoRa mode
Mcu used is STM32 with Arduino lib. 
Code compiles fine. Transmitting side seemd to emit some signal visible on the SDR receiver in the correct band.
Receiver goes into an endless loop.
Eample code used as it is.
Tested on two different  Sx modules. Ra-01 and RFM98HW
Modules work fine when used with different library eg. LoRaRF.
Arduino latest / linbary lates as of this moment.
Library does not allow any control over paket size, or ai I missing something?

jgromes, 2024-02-11:
Please fill in the information requested by the issue template.

> Receiver goes into an endless loop of Timeout error.

Timeout is not an error, it just means that the receiver did not hear anything in the time window it was receiving, that's it. Did you try with interrupts?

> Library does not allow any control over paket size, or am I missing something?

I don't know what you are trying to say here - the library transmits the packets you provide, if you reqeust to send 15 bytes it will send a 15 byte packet + preamble/header/CRCc, which again is set by yourself. You have absolute control over the packet size being sent.

kocinski15, 2024-02-12:
> > Receiver goes into an endless loop of Timeout error.
> 
> Timeout is not an error, it just means that the receiver did not hear anything in the time window it was receiving, that's it. Did you try with interrupts?

No I haven't tried with interrupts. I would like to stick with one interrupt line DIO0 which is required. 
Without DIO0 declared and connected transmitter signals Timeout. So in the test setup both sides use DIO0.

Data is being sent in the loop by the transmitter device every second and once it is received by the receiving side SX1278  it should be seen in the nearest receiver's loop . It isn't. It always ends up with continuous timeout regardless of trasmitter being on or off.

jgromes, 2024-02-12:
> is it interrupt or poll ?

In interrupt mode, it's an interrupt, as the name suggest. In blocking mode, the pin is polled.

>No I haven't tried with interrupts. I would like to stick with one interrupt (... ) line DIO0 which is required. 

Packet reception is signaled on DIO0. On the contrary, timeout is signaled on DIO1, or handled by software if you did not provide DIO1.

> Polling mode on the receiving side is fine for me.

Then you will see timeouts. That's just what blocking mode *is*.

> once it is received by the receiving side SX1278 it should be seen in the nearest receiver's loop

That's not how this works. If the receiver is not receiving while you are transmitting, then you will not get the packet. And when the receiver times out, it will no longer receive, until reception is restarted. That's the base of the blocking mode. With interrupts, the receiver is listening continuously and triggers an interrupt when it receives a packet.

Also, this is not an issue in the library, but a generic question about usage, so I'm converting this thread to a discussion.