jgromes/RadioLib #982: SetPowerOutput(20, false) not working

jacasais, 2024-02-20:
Dear all,

Good day

I am using radiolib with an SX1278 Module, i can transmit and receive without problem but we tested on our spectrum analizer the power output for the module and is alwas 10dBm, even that on the arduino code is set to: if (radio.setOutputPower(20, false) == RADIOLIB_ERR_INVALID_OUTPUT_POWER) {
    Serial.println(F("Selected output power is invalid for this module!"));
    while (true);
  }
Of course we do not get any error that the output power is not correct, unless we set it to for example 22, but even with this setting, is outputting only 10 dBm.
We use this pins on our SX1278:
SX1278 radio = new Module(7, 2, 9, 3);
Does anyone know how we can fix this?

Thank you.

jgromes, 2024-02-20:
Since oyu have not posted your complete code (and ignored the issue template in the process), it's very hard to guess, but I will attempt to anyway: did you increase the overcurent protection limit? I'd guess not, it's set to 60 mA by default, not enough for +20 dBm output.

jacasais, 2024-02-20:
Dear jgromes,
Sorry for deleting the template is the first time i post on GitHub and i did not really know how to proceed.
Thank you for your answer.
I have tried with this setting for the current limit: 
 if (radio.setOutputPower(20, false) == RADIOLIB_ERR_INVALID_OUTPUT_POWER) {
    Serial.println(F("Selected output power is invalid for this module!"));
    while (true);
  }
  if (radio.setCurrentLimit(240) == RADIOLIB_ERR_INVALID_CURRENT_LIMIT) {
    Serial.println(F("Selected current limit is invalid for this module!"));
    while (true);
  }
But still only getting 10 dBm, the module is XL1278-SMT maybe is somethig to do with this?

Thank you for your answer highly appreciated

jgromes, 2024-02-20:
I am still missing basic information, such as the code, which Arduino you are using etc.

> the module is XL1278-SMT

No idea what that is. Quick google search several different Chinese modules. Do you know whether it uses the RFO or the PA_BOOST pin? Is the power source sufficient? How are you measuring the output?

jacasais, 2024-02-20:
#include <RadioLib.h>
#include <WiFi.h>
#include <WebServer.h>
#include <SPIFFS.h>

// Configuración WiFi y servidor web
const char* ssid = "wifi";
const char* password = "123456789";
WebServer server(80);
//unsigned long wifiStartTime;
//const unsigned long wifiTimeout = 60000; // 60 segundos
bool isWifiActive = false;
unsigned long lastRequestTime = 0;
const unsigned long wifiTimeout = 60000; // 60 segundos
String lastConfigData = ""; //variable para leer los datos desde la pagina admin pulsando Leer

#define LED_PIN 1 
#define SENSOR_PIN 10 
unsigned long previousMillis = 0; // Variable para almacenar la última vez que se actualizó el LED
const long intervaloEncendido = 300; // Medio segundo
const long intervaloApagado = 2000; // 2 segundos

// Configuración del módulo LoRa SX1278
SX1278 radio = new Module(7, 2, 9, 3);
bool mensajeLoRaEnviado = false;

float frecuencia = 433.0; // Frecuencia inicial
float anchoBanda = 62.5; // Ancho de banda inicial
String SerialNumber = "433T0000101"; //Numero de serie ejemplo
String ID = "01"; //ID ejemplo maximo ID 99 

void setup() {
 if (!SPIFFS.begin(true)) {
        Serial.println("Error al montar el sistema de archivos");
        return;
    }

    server.on("/", HTTP_GET, handleRoot);

    // Agregar un punto de acceso para servir el archivo de imagen
    server.on("/pickpoint.png", HTTP_GET, []() {
        File file = SPIFFS.open("/pickpoint.png", "r");
        if (!file) {
            server.send(404, "text/plain", "Archivo no encontrado");
            return;
        }
        server.streamFile(file, "image/png");
        file.close();
    });  
// Configurar el servidor web para procesar datos y ademas
// Configura el ESP32 como punto de acceso
  WiFi.softAP(ssid, password);
  isWifiActive = true;
  IPAddress IP = WiFi.softAPIP();
  Serial.print("IP del AP: ");
  Serial.println(IP);
  server.on("/", HTTP_GET, handleRoot);
  server.on("/admin", HTTP_GET, handleAdminLoginForm);
  server.on("/admin", HTTP_POST, handleAdmin);
  server.on("/program", HTTP_POST, handleProgram);
  server.on("/setFrequency", HTTP_GET, []() {
    String frequency = server.arg("frequency");
    procesarComando("F" + frequency);
    server.send(200, "text/plain", "Frecuencia programada: " + frequency);
  });

  server.begin();
  pinMode(A0, INPUT); // Configurar el pin A0 para la lectura de voltaje
  pinMode(LED_PIN, OUTPUT); // Configura el pin del LED como salida
  pinMode(SENSOR_PIN, INPUT); // Configura el pin del sensor como entrada
  digitalWrite(LED_PIN, LOW); // Asegúrate de que el LED esté apagado al inicio

  Serial.begin(9600);

  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin(frecuencia, anchoBanda);
    // Configurar el Factor de Dispersión (Spreading Factor)
   // state = radio.setSpreadingFactor(12);
   // if (state != RADIOLIB_ERR_NONE) {
   //   Serial.println(F("Falló la configuración de SF!"));
   // }
    // Configurar el Código de Tasa (Coding Rate)
   // state = radio.setCodingRate(7);
   // if (state != RADIOLIB_ERR_NONE) {
   //   Serial.println(F("Falló la configuración de CR!"));
   // }
  if (radio.setOutputPower(20, false) == RADIOLIB_ERR_INVALID_OUTPUT_POWER) {
    Serial.println(F("Selected output power is invalid for this module!"));
    while (true);
  }
  if (radio.setCurrentLimit(240) == RADIOLIB_ERR_INVALID_CURRENT_LIMIT) {
    Serial.println(F("Selected current limit is invalid for this module!"));
    while (true);
  }
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("Initialization successful!"));
    digitalWrite(GPS, HIGH);
  } else {
    Serial.print(F("Initialization failed, code "));
    Serial.println(state);
    while (true);
  }

}

I use an ESP32C3Mini module connected to the XL1278-SMT lora module, i can transmit, no problem, the only problem is that i connect the lora to my spectrum analyzer to check the power output and is alwas 10 db no matter what i set on the code. The current is not the problem the lora gets the power from a 18650 battery so this is not the problem. On the specifications for this module, says that can transmitt up to 20,5 dBm.

Thank you sir.

jgromes, 2024-02-20:
> spectrum analyzer to check the power output and is alwas 10 db

10 dB is just a relative measure, what is the reference level? And what is the measurement setup? Spectrum analyzers are typically quite sensitive to input power and will get damaged if you exceed its limit, do you have some attenuators in path?

> The current is not the problem the lora gets the power from a 18650 battery

So you have the LoRa module connected directly to the battery? Seems unlikely, but if it really is the case it will most definitely destroy it, the absolute maximum the module can tolerate is 3.9V, while a fully charged Li-ion cell will be at 4.2V.

I have measured power output from SX1278 using RF power meter - not spectrum analyzer, an actual RF power meter with proper atteniuation. The output was as expected, so I am fairly confident it is not an issue in the library, but rather with your setup, or measurement, so I will convert this to a discussion.