jgromes/RadioLib #985: LLCC68 LoRaWAN failed with -1106 and -9

lioff, 2024-02-26:
**Describe the bug**
I am using the LoRaWAN example with an LLCC68 on EU868.
However, beginOTAA fails during checkADB. It seems to be related to the limited supported SF of LLCC68.

See output:

```
18:43:14.225 -> [Radio] Initializing ... 
18:43:14.225 -> RadioLib Debug Info
18:43:14.226 -> Version:  6.4.2.0
18:43:14.226 -> Platform: ESP32
18:43:14.226 -> Compiled: Feb 26 2024 18:42:26
18:43:14.226 -> 
18:43:14.226 -> CMDW	80	
18:43:14.226 -> SI	0	
18:43:14.226 -> SO	A2	
18:43:14.226 -> CMDR	C0	
18:43:14.226 -> SI	0	
18:43:14.258 -> SO	22	
18:43:14.258 -> CMDR	1D	3	20	
18:43:14.258 -> SI	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
18:43:14.258 -> SO	A2	4C	4C	43	43	36	38	20	56	32	48	20	32	48	30	32	0	
18:43:14.258 -> Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
18:43:14.258 -> 0000320 4c 4c 43 43 36 38 20 56 32 48 20 32 48 30 32 00 | LLCC68 V2H 2H02.
18:43:14.258 -> 
18:43:14.258 -> M	SX126x
18:43:14.258 -> CMDW	80	
18:43:14.258 -> SI	0	
18:43:14.258 -> SO	A2	
18:43:14.258 -> CMDR	C0	
18:43:14.258 -> SI	0	
18:43:14.258 -> SO	22	
18:43:14.258 -> CMDW	80	
18:43:14.258 -> SI	0	
18:43:14.258 -> SO	A2	
18:43:14.258 -> CMDR	C0	
18:43:14.258 -> SI	0	
18:43:14.258 -> SO	22	
18:43:14.258 -> CMDW	8F	
18:43:14.258 -> SI	0	0	
18:43:14.258 -> SO	A2	A2	
18:43:14.258 -> CMDR	C0	
18:43:14.258 -> SI	0	
18:43:14.258 -> SO	22	
18:43:14.258 -> CMDW	8A	
18:43:14.258 -> SI	1	
18:43:14.258 -> SO	A2	
18:43:14.258 -> CMDR	C0	
18:43:14.258 -> SI	0	
18:43:14.258 -> SO	22	
18:43:14.258 -> CMDW	93	
18:43:14.258 -> SI	20	
18:43:14.258 -> SO	A2	
18:43:14.289 -> CMDR	C0	
18:43:14.289 -> SI	0	
18:43:14.289 -> SO	22	
18:43:14.289 -> CMDW	88	
18:43:14.289 -> SI	3	16	A	0	0	0	0	
18:43:14.289 -> SO	A2	A2	A2	A2	A2	A2	A2	
18:43:14.289 -> CMDR	C0	
18:43:14.289 -> SI	0	
18:43:14.289 -> SO	22	
18:43:14.289 -> CMDW	2	
18:43:14.289 -> SI	43	FF	
18:43:14.289 -> SO	A2	A2	
18:43:14.289 -> CMDR	C0	
18:43:14.289 -> SI	0	
18:43:14.289 -> SO	22	
18:43:14.289 -> CMDW	8	
18:43:14.289 -> SI	0	0	0	0	0	0	0	0	
[... 476 lines cut ...]
    Serial.println(F("no downlink!"));

  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

  // on EEPROM enabled boards, you can save the current session
  // by calling "saveSession" which allows retrieving the session after reboot or deepsleep
  node.saveSession();

  // wait before sending another packet
  uint32_t minimumDelay = 60000;                   // try to send once every minute
  uint32_t interval = node.timeUntilUplink();      // calculate minimum duty cycle delay (per law!)
  uint32_t delayMs = max(interval, minimumDelay);  // cannot send faster than duty cycle allows

  delay(delayMs);
}
```

**Expected behavior**
Example should connect to TTN and send Hello World message.

**Additional info:**
 - I am using an ESP32 S2 (Wemos S3 Mini)

jgromes, 2024-02-26:
This has been discussed (at length) here: https://github.com/jgromes/RadioLib/discussions/946

It's been fixed in upstream, just not released yet. Are you using the last release? And if so, can you try with the current master?

lioff, 2024-02-27:
@jgromes: I used the latest release (6.4.2). I pulled the current master and get tge sane error:

```
18:52:13.358 -> [Radio] Initializing ... 
18:52:13.359 -> RadioLib Debug Info
18:52:13.359 -> Version:  6.4.2.0
18:52:13.359 -> Platform: ESP32
18:52:13.359 -> Compiled: Feb 27 2024 18:50:41
18:52:13.359 -> 
18:52:13.359 -> Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
18:52:13.359 -> 0000320 4c 4c 43 43 36 38 20 56 32 48 20 32 48 30 32 00 | LLCC68 V2H 2H02.
18:52:13.359 -> 
18:52:13.359 -> M	SX126x
18:52:13.359 -> success!
18:52:13.359 -> [LoRaWAN] Attempting over-the-air activation ... Didn't restore session (checksum: 0, mode: 0)
18:52:13.359 -> First 16 bytes of NVM:
18:52:13.391 -> 0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
18:52:13.391 -> Wiping EEPROM and starting a clean session
18:52:13.681 -> Setting up dynamic channels
18:52:13.681 -> Channel UL/DL 0 frequency = 868.099976 MHz
18:52:13.681 -> Channel UL/DL 1 frequency = 868.299988 MHz
18:52:13.681 -> Channel UL/DL 2 frequency = 868.500000 MHz
18:52:13.681 -> exe MAC CID = 03, len = 4
18:52:13.681 -> ADR REQ: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 00, nbTrans = 0
18:52:13.681 -> DR 72: LORA (SF: 10, BW: 125.000000, CR: 5)
18:52:13.681 -> ADR failed to configure dataRate 2, code -9!
18:52:13.681 -> ADR ANS: status = 0x05
18:52:13.681 -> exe MAC CID = 04, len = 1
18:52:13.681 -> Max duty cycle: 1/2^7
18:52:13.681 -> exe MAC CID = 05, len = 4
18:52:13.681 -> Rx param REQ: rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525024
18:52:13.712 -> Rx param ANS: status = 0x07
18:52:13.712 -> exe MAC CID = 08, len = 1
18:52:13.712 -> RX timing: delay = 1 sec
18:52:13.744 -> exe MAC CID = 09, len = 1
18:52:13.744 -> TX timing: dlDwell = 0, ulDwell = 0, maxEirp = 16 dBm
18:52:13.744 -> exe MAC CID = 0c, len = 1
18:52:13.744 -> ADR param setup: limitExp = 6, delayExp = 5
18:52:13.777 -> exe MAC CID = 0f, len = 1
18:52:13.777 -> Rejoin setup: maxTime = 15, maxCount = 10
18:52:13.777 -> There are no channels defined - are you in ABP mode with no defined subband?
```

Based on the discussion you mentioned I tried again with joinDr of 3:
```
state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey, 3);
```

Now it seems to attempt a join request with SF 9, reaches a timeout, and then fails in a second attempt with SF 12.
I am not sure if this is expected behavior (as the SF 9 request timed out).

Any idea what the problem could be?

lioff, 2024-02-27:
I see now that TTN actually receives (at least some) of the join requests. The first one appeared ok (even though I did not notice any change in the RadioLib debug output). Since then, however, I get the error "DevNonce is too small" in TTN console. I tried to reset DevNonce in TTN, but no change.

[image]

jgromes, 2024-02-27:
It looks like the original problem was resolved (TTN now obviously receives your join requests), so I will convert this to a discussion.

> Based on the discussion you mentioned I tried again with joinDr of 3:

Setting appropriate DR is required, since LLCC68 only supports a subset of spreading factors. 

> I see now that TTN actually receives (at least some) of the join requests. The first one appeared ok (even though I did not notice any change in the RadioLib debug output). 

So ... can you post the output? Was there a timeout, or did you receive the join accept message? Either way it might be worth it to wipe the persistent memory, and (temporarily) allow reusing dev nonces in TTN.