jgromes/RadioLib #987: IDF Component registry

nicklasb, 2024-02-26:
Hi, 
Saw that you added an ESP-IDF build step, just saying that it feels like you'd have very little more to do to add Radiolib to the Espressif IDF component library to make it easy for my library to depend on it there. 

I'd rather not have to make a dumb fork of my own (with the exact same content) and confuse everybody. 
Also, the IDF registry doesn't allow git repository references ([for a pretty good reason](https://github.com/espressif/idf-component-manager/issues/57)), so I cannot depend on it that way either. 

Anyway, it is not that hard, and would help the glorious Radiolib and its charming maintainer bedazzle and rule them all even more. :-)

nicklasb, 2024-02-27:
Would you like a PR for that?

jgromes, 2024-02-27:
I don't use ESP-IDF that much, so I'm not that familiar with the whole ecosystem - could you elaborate on what would be needed to add RadioLib to Espressif IDF component library? I imagine it's something like the Platformio package index ...?

By chance, I did recently start one proejct using ESP-IDF and RadioLib, so I did make it work there (pretty much just using the NonArduino/ESP-IDF example as the basis), but like I said I'm not really that familiar with ESP-IDF, so I have no idea if the way it's handled currently is correct.

> Would you like a PR for that?

If it makes the library easier to use in ESP-IDF, then that would be great!

closed 2024-02-28 by jgromes