jgromes/RadioLib #988: Heltec Wireless Stick Lite V3 - code -12 and then -1108

vladosam, 2024-02-27:
Hi. I'm trying to use radiolib library on board. It worked when I first tried some examples but with current code all I get is code -12 and on next data send -1108. I checked status codes and I see that code -12 is (The supplied frequency value is invalid for this module).  I'm not sure is it problem with code or maybe on server side (chirpstack v3). Any help is much appreciated.

Arduino code:
```
/*
  RadioLib LoRaWAN End Device Persistent Example

  This example assumes you have tried one of the OTAA or ABP
  examples and are familiar with the required keys and procedures.
  This example restores and saves a session such that you can use
  deepsleep or survive power cycles. Before you start, you will 
  have to register your device at https://www.thethingsnetwork.org/
  and join the network using either OTAA or ABP.
  Please refer to one of the other LoRaWAN examples for more
  information regarding joining a network.

  NOTE: LoRaWAN requires storing some parameters persistently!
        RadioLib does this by using EEPROM, by default
        starting at address 0 and using 448 bytes.
        If you already use EEPROM in your application,
        you will have to either avoid this range, or change it
        by setting a different start address by changing the value of
        RADIOLIB_HAL_PERSISTENT_STORAGE_BASE macro, either
        during build or in src/BuildOpt.h.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
#include <SPI.h>
SPIClass spi(SPI);
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);
// SX1262 has the following pin order:
// Module(NSS/CS, DIO1, RESET, BUSY)
SX1262 radio = new Module(8, 14, 12, 13, spi, spiSettings);

// SX1278 has the following pin order:
// Module(NSS/CS, DIO0, RESET, DIO1)
//SX1278 radio = new Module(10, 2, 9, 3);

// create the node instance on the EU-868 band
// using the radio module and the encryption key
// make sure you are using the correct band
// based on your geographical location!
LoRaWANNode node(&radio, &EU868);

// for fixed bands with subband selection
// such as US915 and AU915, you must specify
// the subband that matches the Frequency Plan
// that you selected on your LoRaWAN console
/*
  LoRaWANNode node(&radio, &US915, 2);
*/

void setup() {
  Serial.begin(9600);
	// SCK/CLK, MISO, MOSI, NSS/CS,
	spi.begin(9, 11, 10, 8);
  // initialize SX1278 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
[... 1206 lines cut ...]
SO	22	
CMDW	D	7	36	
SI	9	
SO	A2	
CMDW	8C	
SI	0	8	0	FF	1	1	
SO	A2	A2	A2	A2	A2	A2	
CMDR	C0	
SI	0	
SO	22	
CMDW	82	
SI	0	AA	E3	
SO	A2	A2	A2	
Opening Rx1 window (683552 us timeout)... <-- Rx Delay end 
closing
CMDR	12	
SI	0	0	0	
SO	A6	2	0	
CMDW	80	
SI	0	
SO	A6	
failed, code -12

[LoRaWAN] Sending uplink packet ... failed, code -1108
```

StevenCellist, 2024-02-27:
Hi @vladosam, I see that you configured your device as a LoRaWAN 1.0.x device.

> LoRaWAN session: v1.0

RadioLib specifically calls out for version 1.1.0 as the LoRaWAN version, as version 1.0.x is not tested and therefore definitely not guaranteed to work. Please register your device with version 1.1.0 and retry.

You may also have more luck by running the most recent version on GitHub (heck, maybe even the currently open PR), as opposed to version 6.4.2 that is currently out. There's a couple of bugfixes and improvements that are not out yet.

And if that still doesn't work, _please_ disable the VERBOSE flag first before posting additional logs, as the SPI output makes the logs way too long to read. If we need the SPI output, we'll let you know :)

vladosam, 2024-02-27:
Thanks for reply. I will set device to 1.1.0 and try again.

StevenCellist, 2024-02-27:
Quick note: I also see that you call out both `beginOTAA()` _and_ `restore()`. If you look at the current upstream examples, you'll notice that references to `restore()` are generally removed; you shouldn't need that call as long as you use `saveSession()`. In the current release 6.4.2 that is likely to cause some problems as well.

vladosam, 2024-02-27:
Progress. I changed device profile to lora 1.1 but now there is a new error code -6. Now I probably have join problem because I don't see device as active on server. I also remove `restore()` from code. 
```
[SX1262] Initializing ... 
RadioLib Debug Info
Version:  6.4.2.0
Platform: ESP32
Compiled: Feb 27 2024 09:44:01

Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.

M	SX126x
success!
[LoRaWAN] Attempting over-the-air activation ... Didn't restore session (checksum: 0, mode: 0)
First 16 bytes of NVM:
0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
Wiping EEPROM and starting a clean session
Setting up dynamic channels
Channel UL/DL 0 frequency = 868.099976 MHz
Channel UL/DL 1 frequency = 868.299988 MHz
Channel UL/DL 2 frequency = 868.500000 MHz
exe MAC CID = 03, len = 4
ADR REQ: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 00, nbTrans = 0
DR 72: LORA (SF: 10, BW: 125.000000, CR: 5)
ADR ANS: status = 0x07
exe MAC CID = 04, len = 1
Max duty cycle: 1/2^7
exe MAC CID = 05, len = 4
Rx param REQ: rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525024
Rx param ANS: status = 0x07
exe MAC CID = 08, len = 1
RX timing: delay = 1 sec
exe MAC CID = 09, len = 1
TX timing: dlDwell = 0, ulDwell = 0, maxEirp = 16 dBm
exe MAC CID = 0c, len = 1
ADR param setup: limitExp = 6, delayExp = 5
exe MAC CID = 0f, len = 1
Rejoin setup: maxTime = 15, maxCount = 10

Channel frequency UL = 868.099976 MHz
DR 72: LORA (SF: 10, BW: 125.000000, CR: 5)
Timeout in 556032 us
Join-request sent <-- Rx Delay start

Channel frequency DL = 868.099976 MHz
DR 72: LORA (SF: 10, BW: 125.000000, CR: 5)
Opening Rx1 window (226848 us timeout)... <-- Rx Delay end 
closing
DR 104: LORA (SF: 12, BW: 125.000000, CR: 5)
Opening Rx2 window (683552 us timeout)... <-- Rx Delay end 
closing
[LoRaWAN] Resuming previous session ... failed, code -6
```

vladosam, 2024-02-27:
I have access to gateway. It is Heltec HT-M01 GW with Samtec PacketForwarder_UI.
```
INFO: Received pkt from mote: 12AC648B (fcnt=17814)

JSON up: {"rxpk":[{"tmst":11275396,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":8.0,"rssi":-34,"size":23,"data":"AItkrBItlkVfAAAAAAAAAAABACmWQQ4="}]}
INFO: [up] PUSH_ACK received in 13 ms
```

StevenCellist, 2024-02-27:
Does a join-request arrive at your Chirpstack server (either at the gateway console on your server or in the gateway logs)?
If you do see it somewhere, the problem is likely that there is a problem with the DevNonce in the join-request, and you should check "Resets DevNonces" on your device console (no idea where that is on Chirpstack, I use TTN).
If you do not see any join-request, we will have to look a bit harder.

vladosam, 2024-02-27:
I don't see join requests on server. There were join requests when I used 1.0.2 lorawan version.

HeadBoffin, 2024-02-27:
> I don't see join requests on server. There were join requests when I used 1.0.2 lorawan version.

It's super very important to clarify if you see the JR in the gateway console but not in the device console as this will confirm the DevNonce situation.

In the meanwhile we can look at Chirpstack DevNonce config.

vladosam, 2024-02-27:
There is no JR on server. On GW this is what I see in console when I power up board.
```
JSON up: {"rxpk":[{"tmst":111289428,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":8.5,"rssi":-15,"size":23,"data":"AItkrBItlkVfAAAAAAAAAAABACmWQQ4="}]}
INFO: [up] PUSH_ACK received in 2 ms
INFO: host/sx1301 time offset=(1709071617s:260861 us) - drift=851 us
```

vladosam, 2024-02-27:
On  Chirpstack DevNonce is under Activation and it is empty for my device.
[image]

HeadBoffin, 2024-02-27:
> There is no JR on server.

So the JR isn't getting to the gateway server at all? 

Useful screen shot, thanks - good to know there is a clear DevNonce button

vladosam, 2024-02-27:
Now I found that my GW is acting strange. It looks like it lost connection to server. It's getting late here so I will test this more in the morning and I will get back to you. Thanks for your help.

vladosam, 2024-02-28:
It was GW misconfiguration that caused code -6 error. I probably made mistake trying to fix my first code -12 error. Now node successfully connects to server. Now I see [LoRaWAN] Frequency error:	-14100.16 Hz in serial console on node. Is that something unusual?

StevenCellist, 2024-02-28:
You can ignore that, it's some gimmick output that has zero meaning about LoRaWAN.

closed 2024-02-28 by StevenCellist