jgromes/RadioLib #997: Arduino Nano / ESP32 CC1101 : Error -6 when receiving

slater0013, 2024-03-01:
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

playing with CC1101_Receive/Transmit_Address on Arduino Nano boards and ESP32 boards.

**Hardware setup**
Arduino Nano as sketched here https://github.com/LSatan/SmartRC-CC1101-Driver-Lib?tab=readme-ov-file#wiring
- 3,3V and GND
- GDO0 on D6
- CSN on D10
- SCK on SCK/D13
- MOSI on D11
- MISO on D12
- GDO2 on D2

ESP32 WROOM 32D devkit v4 :
 - 3,3V and GND
 - GDO0 on D2
 - CSN on GPIO5
 - SCK on SCK/GPIO18
 - MOSI on MOSI/GPIO23
 - MISO on MOSI/GPIO19
 - GDO2 on GPIO4

Sending on ESP32 / init code : 
CC1101 radio = new Module(5, 2, RADIOLIB_NC, 4);

Receving on Arduino nano / init code : 
`CC1101 radio = new Module(10, 6, RADIOLIB_NC,2);`

Debug mode Enabled, fresh install of Arduino IDE and libraries.

Serial Monitor on ESP32 sending device : 

`CMD	W	35	0F
CMD	W	36	2F
R	75	1	
CMD	W	3B	0F
success!
[CC1101] Transmitting packet ... CMD	W	36	0F
R	75	1	
CMD	W	3B	0F
R	0	6	
W	0	6	
R	0	6	
W	3F	C	
R	7	66	
W	3F	0	
W	7F	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	
CMD	W	35	0F
CMD	W	36	2F
R	75	1	
CMD	W	3B	0F
success!
`

Serial Monitor on Arduino Nano receiving device : 

`[CC1101] Waiting for incoming transmission ... CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
R	2	46	
W	2	46	
R	2	46	
CMD	W	34	0F
CMD	W	36	1F
R	75	1	
CMD	W	3A	0F
failed, code -6
[CC1101] Waiting for incoming transmission ... CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
R	2	46	
W	2	46	
R	2	46	
CMD	W	34	0F
CMD	W	36	1F
R	75	1	
CMD	W	3A	0F
failed, code -6
[CC1101] Waiting for incoming transmission ... CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
R	2	46	
W	2	46	
R	2	46	
CMD	W	34	0F
CMD	W	36	1F
R	75	1	
CMD	W	3A	0F
failed, code -6`

Any help appreciated ! Goal project is to build a powerpoint/presentation remote working on a quite long range.

Thanks,
GS

slater0013, 2024-03-01:
Waiting like a minute, I noticed one partly/received packet in debug mode in the middle of all messages.

I modified the radio.transmit with a shorter message : 

`  int state = radio.transmit("GorD");`

And now on the receiveing side : 

`[CC1101] Waiting for incoming transmission ... CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
R	2	46	
W	2	46	
R	2	46	
CMD	W	34	0F
R	3F	4	
R	7	66	
R	3F	0	
R	7F	47	6F	72	66	
R	7	66	
R	3F	98	
R	3F	98	
R	17	30	
CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
success!
[CC1101] Data:		Gorf
[CC1101] RSSI:		-126.00 dBm
[CC1101] LQI:		24
[CC1101] Waiting for incoming transmission ... CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
R	2	46	
W	2	46	
R	2	46	
CMD	W	34	0F
CMD	W	36	1F
R	75	1	
CMD	W	3A	0F
failed, code -6`

You can see that the message was slightly modified during transmission. 

Tried again with "HelloWorld" string without the debug mode : 

`[CC1101] Waiting for incoming transmission ... failed, code -6
[CC1101] Waiting for incoming transmission ... failed, code -6
[CC1101] Waiting for incoming transmission ... success!
[CC1101] Data:		HelloWorlt
[CC1101] RSSI:		-106.00 dBm
[CC1101] LQI:		64
[CC1101] Waiting for incoming transmission ... failed, code -6
[CC1101] Waiting for incoming transmission ... failed, code -6
[CC1101] Waiting for incoming transmission ... success!
[CC1101] Data:		HelloWorlr
[CC1101] RSSI:		-105.50 dBm
[CC1101] LQI:		65
[CC1101] Waiting for incoming transmission ... failed, code -6
[CC1101] Waiting for incoming transmission ... failed, code -6
[CC1101] Waiting for incoming transmission ... failed, code -6`

Could it be related to some "timing" problem like Rx window too short to get the entire message ?

GS

jgromes, 2024-03-02:
Did you read what -6 means? It's shown here: https://jgromes.github.io/RadioLib/group__status__codes.html

-6 is `RADIOLIB_ERR_RX_TIMEOUT`, which is to be expected in blocking mode. I would suggest trying interrupt mode.

slater0013, 2024-03-02:
Thanks @jgromes but are
https://github.com/jgromes/RadioLib/blob/master/examples/CC1101/CC1101_Receive_Address/
and 
https://github.com/jgromes/RadioLib/tree/master/examples/CC1101/CC1101_Transmit_Address
dealing with blocking mode ?

I can definitely give a try to the 2 other exemples 
https://github.com/jgromes/RadioLib/tree/master/examples/CC1101/CC1101_Receive_Interrupt
and 
https://github.com/jgromes/RadioLib/tree/master/examples/CC1101/CC1101_Transmit_Interrupt

if this is what you mean ?

GS.

jgromes, 2024-03-02:
> if this is what you mean ?

Yes, that's what I meant. I was under the impression that the CC1101 examples also included explanation of the timeouts in blocking mode, but apparently not. That's an oversight on my part, the examples should be clearer in this regard. The purpose of `Receive_Address` and `Transmit_Address` is only to showcase how to use addressing on CC1101. I guess a note to that effect in these and other examples would be useful.

slater0013, 2024-03-04:
Ok, following your advice, tried to use the CC1101 interrupt exemples : 

Juste copy/paste of the Tx and Rx files, just adapting the pinout regarding Arduino Nano and ESP32 boards.

Sending devices (Arduino Nano) seems to be ok : 

```
23:20:08.195 -> [CC1101] Sending another packet ... CMD	W	36	0F
23:20:08.229 -> R	75	1	
23:20:08.229 -> CMD	W	3B	0F
23:20:08.262 -> R	0	6	
23:20:08.262 -> W	0	6	
23:20:08.262 -> R	0	6	
23:20:08.262 -> W	3F	F	
23:20:08.294 -> R	7	64	
23:20:08.294 -> W	7F	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	20	23	36	
23:20:08.326 -> CMD	W	35	0F
23:20:08.359 -> transmission finished!
23:20:08.392 -> CMD	W	36	2F
23:20:08.392 -> R	75	1	
23:20:08.392 -> CMD	W	3B	0F
23:20:09.347 -> [CC1101] Sending another packet ... CMD	W	36	0F
23:20:09.381 -> R	75	1	
23:20:09.381 -> CMD	W	3B	0F
23:20:09.413 -> R	0	6	
23:20:09.413 -> W	0	6	
23:20:09.413 -> R	0	6	
23:20:09.413 -> W	3F	F	
23:20:09.445 -> R	7	64	
23:20:09.445 -> W	7F	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	20	23	37	
23:20:09.478 -> CMD	W	35	0F
23:20:09.511 -> transmission finished!
23:20:09.543 -> CMD	W	36	2F
23:20:09.543 -> R	75	1	
23:20:09.543 -> CMD	W	3B	0F
```

But On the receiving side (ESP32) a first empty message is received and then nothing. Even when the Tx device is off, rebooting the Rx device always shows a first empty message, strange...

```
23:19:15.793 -> CMD	W	36	1F
23:19:15.825 -> R	75	1	
23:19:15.825 -> CMD	W	3A	0F
23:19:15.825 -> [CC1101] Received packet!
23:19:15.857 -> [CC1101] Data:		
23:19:15.890 -> [CC1101] RSSI:		-64.00 dBm
23:19:15.922 -> [CC1101] LQI:		0
23:19:15.922 -> CMD	W	36	0F
23:19:15.953 -> R	75	1	
23:19:15.953 -> CMD	W	3A	0F
23:19:15.953 -> R	2	46	
23:19:15.953 -> W	2	46	
23:19:15.986 -> R	2	46	
23:19:15.986 -> CMD	W	34	0F
```

The comments states : 
```
   To successfully receive data, the following settings have to be the same
   on both transmitter and receiver:
    - carrier frequency
    - bit rate
    - frequency deviation
    - sync word
```

But I guessed not changing anything to the default values would result in being the same values on both sides ! Am I right ?

I don't think my 434.0 MHz is saturated, but tried to switch to 433.300 MHz or 434.400 MHz with no more success :( But always the same empty packet when booting the Rx device.

Either Tx or Rx deivce (or both) is lying but how to debug... ?

thanks,
GS

slater0013, 2024-03-06:
Ok, tried swapping the boards. (I can also test 2 ESP32 and 2 Nano if that's interesting) ! Yes, tha arduino nano is the old ATmega328P board.

Arduino Nano receiving side : 
````
[CC1101] Initializing ... 
RadioLib Debug Info
Version:  6.4.2.0[CC1101] Initializing ... 
RadioLib Debug Info
Version:  6.4.2.0
Platform: Arduino AVR
Compiled: Mar  4 2024 23:11:26

R	71	14	
M	CC1101
CMD	W	30	0F
CMD	W	36	0F
R	75	1	
R	18	4	
W	18	14	
R	18	14	
R	18	14	
W	18	14	
R	18	14	
R	2	3F	
W	2	2E	
R	2	2E	
R	0	29	
W	0	2E	
R	0	2E	
R	7	4	
W	7	4	
R	7	4	
R	8	45	
W	8	5	
R	8	5	
R	8	5	
W	8	5	
R	8	5	
CMD	W	36	0F
R	D	1E	
W	D	10	
R	D	10	
R	E	C4	
W	E	B5	
R	E	B5	
R	F	EC	
W	F	2B	
R	F	2B	
R	3E	C6	
W	3E	C0	
R	3E	C0	
CMD	W	36	0F
R	10	8C	
W	10	87	
R	10	87	
R	11	22	
W	11	83	
R	11	83	
CMD	W	36	0F
R	10	87	
W	10	F7	
R	10	F7	
CMD	W	36	0F
R	15	47	
W	15	17	
R	15	17	
R	15	17	
W	15	14	
R	15	14	
R	3E	C0	
W	3E	C0	
R	3E	C0	
R	8	5	
W	8	5	
R	8	5	
R	6	FF	
W	6	3F	
R	6	3F	
R	7	4	
W	7	64	
R	7	64	
R	13	22	
W	13	2	
R	13	2	
CMD	W	36	0F
R	75	1	
R	12	2	
W	12	2	
R	12	2	
CMD	W	36	0F
R	75	1	
R	12	2	
W	12	2	
R	12	2	
R	8	5	
W	8	5	
R	8	5	
R	12	2	
W	12	2	
R	12	2	
R	4	D3	
W	4	12	
R	4	12	
R	5	91	
W	5	AD	
R	5	AD	
CMD	W	3A	0F
CMD	W	3B	0F
success!
[CC1101] Starting to listen ... CMD	W	36	0F
R	75	1	
CMD	W	3A	0F
R	2	2E	
W	2	46	
R	2	46	
CMD	W	34	0F
success!
```
No more empty packet at the begining. But... no packet even after :( 

ESP32 debug console (Tx device now) : 

```
[CC1101] Initializing ... 
RadioLib Debug Info
Version:  6.4.2.0
Platform: ESP32
Compiled: Mar  4 2024 23:12:05

R	71	14	
M	CC1101
CMD	W	30	0F
CMD	W	36	0F
R	75	1	
R	18	4	
W	18	14	
R	18	14	
R	18	14	
W	18	14	
R	18	14	
R	2	3F	
W	2	2E	
R	2	2E	
R	0	29	
W	0	2E	
R	0	2E	
R	7	4	
W	7	4	
R	7	4	
R	8	45	
W	8	5	
R	8	5	
R	8	5	
W	8	5	
R	8	5	
CMD	W	36	0F
R	D	1E	
W	D	10	
R	D	10	
R	E	C4	
W	E	B5	
R	E	B5	
R	F	EC	
W	F	2B	
R	F	2B	
R	3E	C6	
W	3E	C0	
R	3E	C0	
CMD	W	36	0F
R	10	8C	
W	10	87	
R	10	87	
R	11	22	
W	11	83	
R	11	83	
CMD	W	36	0F
R	10	87	
W	10	F7	
R	10	F7	
CMD	W	36	0F
R	15	47	
W	15	17	
R	15	17	
R	15	17	
W	15	14	
R	15	14	
R	3E	C0	
W	3E	C0	
R	3E	C0	
R	8	5	
W	8	5	
R	8	5	
R	6	FF	
W	6	3F	
R	6	3F	
R	7	4	
W	7	64	
R	7	64	
R	13	22	
W	13	2	
R	13	2	
CMD	W	36	0F
R	75	1	
R	12	2	
W	12	2	
R	12	2	
CMD	W	36	0F
R	75	1	
R	12	2	
W	12	2	
R	12	2	
R	8	5	
W	8	5	
R	8	5	
R	12	2	
W	12	2	
R	12	2	
R	4	D3	
W	4	12	
R	4	12	
R	5	91	
W	5	AD	
R	5	AD	
CMD	W	3A	0F
CMD	W	3B	0F
success!
[CC1101] Sending first packet ... CMD	W	36	0F
R	75	1	
CMD	W	3B	0F
R	0	2E	
W	0	6	
R	0	6	
W	3F	C	
R	7	64	
W	7F	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	
CMD	W	35	0F
transmission finished!
CMD	W	36	0F
R	75	1	
CMD	W	3B	0F
[CC1101] Sending another packet ... CMD	W	36	0F
R	75	1	
CMD	W	3B	0F
R	0	6	
W	0	6	
R	0	6	
W	3F	F	
R	7	64	
W	7F	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	20	23	30	
CMD	W	35	0F
transmission finished!
```

ESP32 keeps transmiting packet over and over every second.

GS.

slater0013, 2024-03-06:
Ok just tried with 2 arduino nano boards. same thing. :(

slater0013, 2024-03-10:
Yes I use dedicated 3.3V output from nano board !
Do you think it could miss some power ? How do you power them ?

I ordered them from Ali (8pin connection) if you have any recommendation or a better designed module for what I need to do (simple remote control)…

Thanks,
GS

jgromes, 2024-03-10:
I'm not talking about the power source, I'm talking about the votlage level on the Arduino digital pins. It sends out 5V signals, that exceeds the CC1101 maximum votlage rating of 3.9V. You need a logic level converter, such as this one: https://www.sparkfun.com/products/12009

Closing as I wasn't able to replicate the issue with neither Uno nor ESP32.

closed 2024-03-10 by jgromes

slater0013, 2024-03-10:
Just one last comment, Arduino nano and ESP32 have 3V3 Pins, where I connect the modules.

slater0013, 2024-03-10:
Awww... I think I just understand what you mean... I need to double check the datasheet ❌

slater0013, 2024-03-10:
According to https://www.ebyte.com/en/downpdf.aspx?id=165

Communucation Level : 3.3V maxi (5V TTL It may be at risk of burning down) 

oops... the first CC1101 I tested (SmartRC-CC1101) just have a warning at top Notes I did not read : A logic level converter is recommended for arduino. It also works well without. Use at your own risk.

My bad :(