jgromes/RadioLib #1001: NodeMCU (ESP8266) with SX1262 is not working

Jaishankar872, 2024-03-03:
This the connection(verified multiple times) I made with
NodeMCU (ESP8266) <->   SX1262 LoRa Module
-------------------------------------------
D5 (GPIO14)     <->   SCK
D6 (GPIO12)     <->   MISO
D7 (GPIO13)     <->   MOSI
D8 (GPIO15)     <->   NSS (Chip Select)
D2 (GPIO4)     <->   RST
D0 (GPIO16)    <->   DIO1
D1 (GPIO5)     <->   BUSY 

In the Example code, I updated pin function alone as mentioned below
// SX1262 has the following connections:
// NSS pin:   15
// DIO1 pin:  16
// NRST pin:  4 
// BUSY pin:  5
SX1262 radio = new Module(15, 16, 4, 5);

Ouput Log as
[SX1262] Initializing ... failed, code -707

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

----------------------------------------------------------------------------------------------------------------------------------------------------
PROBLEM is
             I tried all the way include the [Non-standard SPI setup](https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup) still the ESP8266 communication is failing & get the Watchdog timer reset again and again
       Please help to resolve this issue

jgromes, 2024-03-03:
I would suggest to start by finding out what -707 error code means. It can be found on this page: https://jgromes.github.io/RadioLib/group__status__codes.html

You will find that this error is indicitave of (among other things) incorrectly selected crystal.

The watchdog reset is only because after error, your code (which you failed to post) enters an infinite loop. The ESP8266 stops petting the watchdog because of that loop, and after a while, the watchdog times out and resets the ESP8266. This has nothing to do with the problem you're having, it's just a consequence of it.

> Please help to resolve this issue

That is rather hard to do - you say you have tried the non-standard SPI setups, yet you posted no code. Follow the issue template, I assure you it exists for a reason ...

Jaishankar872, 2024-03-04:
Thanks for your help

Now I understand the issue and also issue got resolved. Added the TCXO Parameter in code as below

-----------------------------------------------------------------------------------------------------------------------------------------------
  float lora_sx1262_freq = 878;
  float lora_sx1262_bw = 125;
  uint8_t lora_sx1262_sf = 7;
  uint8_t lora_sx1262_cr = 5;
  uint8_t lora_sx1262_syncWord = 0x3444;
  int8_t lora_sx1262_power = 20;
  uint16_t lora_sx1262_preambleLength = 12;
  float lora_sx1262_tcxoVoltage = 0;
  bool lora_sx1262_useRegulatorLDO = false;

  int state = radio.begin(lora_sx1262_freq, lora_sx1262_bw, lora_sx1262_sf, lora_sx1262_cr, lora_sx1262_syncWord, lora_sx1262_power, lora_sx1262_preambleLength, lora_sx1262_tcxoVoltage, lora_sx1262_useRegulatorLDO);

 
-----------------------------------------------------------------------------------------------------------------------------------------------