jgromes/RadioLib #1004: Heltec Wireless Stick Lite V3 not receiving join accept

cburza, 2024-03-06:
I have a Heltec Wireless Stick Lite V3 board flashed with the example code. The board uses a ESP32-S3 and a SX1262 module.

The node is set up with TTN which receives the join request and also responds with the join accept message. I have tested the board with Heltecs modified ESP32 platform code and their Lorawan library which joins the network without an issue.

I cannot get the board to read the response from the gateway and I cannot get past the -6 error code.

Here is a screenshot of the example code connecting to TTN.

[image]

**Sketch that is causing the module fail**

I am using platformio.

```ini
[env:heltec_wifi_lora_32_V3]
platform = espressif32@ 6.5.0
board = heltec_wifi_lora_32_V3
board_build.mcu = esp32s3
framework = arduino
lib_deps =
    jgromes/RadioLib@^6.4.2
monitor_speed = 115200
build_flags = -D RADIOLIB_DEBUG=1 -D RADIOLIB_VERBOSE=1
```

The main.ccp

```c++
#include <Arduino.h>
#include <Wire.h>
#include <RadioLib.h>

constexpr int SX1262_NSS = 8;
constexpr int SX1262_DIO1 = 14;
constexpr int SX1262_NRST = 12;
constexpr int SX1262_BUSY = 13;

SPIClass spi(SPI);
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);// NSS/CS, DIO1, RST, BUSY
SX1262 radio = new Module(SX1262_NSS, SX1262_DIO1, SX1262_NRST, SX1262_BUSY, spi, spiSettings);
LoRaWANNode node(&radio, &EU868);

void setup() {
    spi.begin(9, 11, 10, 8);
    Serial.begin(115200);
    delay(100);

    Serial.print(F("[SX1262] Initializing ... "));
    int state = radio.begin();
    if(state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while(true);
    }

    // application identifier - pre-LoRaWAN 1.1.0, this was called appEUI
    // when adding new end device in TTN, you will have to enter this number
    // you can pick any number you want, but it has to be unique
    uint64_t joinEUI = 0x0000000000000000;

    // device identifier - this number can be anything
    // when adding new end device in TTN, you can generate this number,
    // or you can set any value you want, provided it is also unique
    uint64_t devEUI = 0xredacted;

    // select some encryption keys which will be used to secure the communication
    // there are two of them - network key and application key
[... 1271 lines cut ...]
SI      0       
SO      A6      
CMDR    11      
SI      0       0       
SO      A2      1       
CMDR    C0      
SI      0       
SO      22      
CMDR    1D      7       36      
SI      0       0       
SO      A2      9       
CMDR    C0      
SI      0       
SO      22      
CMDW    D       7       36      
SI      D       
SO      A2      
CMDW    8C      
SI      0       8       0       FF      1       0       
SO      A2      A2      A2      A2      A2      A2      
CMDR    C0      
SI      0       
SO      22      
failed, code -6
```

HeadBoffin, 2024-03-06:
Overall this is deeply unlikely to be an actual RadioLib issue, partly because this Heltec model is part of the test suite for LW and partly because the join request is being transmitted. And your keys must be OK as the LNS is issuing a join accept.

Can you confirm which version of LW you registered the device on TTN with.

Do you have a gateway? If so, where is it in relation to the device? What is the RSSI & SNR of join request?

cburza, 2024-03-06:
@HeadBoffin It must have been a signal issue. I change my location and the join was received successfully.

closed 2024-03-06 by cburza

HeadBoffin, 2024-03-06:
You are just on the edge of LoRa's abilities so you'll find your data rate is quite slow which will restrict airtime.