jgromes/RadioLib #1010: e22-400m33s recieve interrupt

whoim2, 2024-03-10:
Hello. Have esp32s2 & this module.
The board is my design, for the E32-400M module and works with it. There is a 1278 chip. The current one is 1268.
I corrected the ping pong example from 1262 to 1268 and the following happens:
`[SX1262] Initializing ... success!
[SX1262] Sending first packet ... operationDone
transmission finished!
start_rcv
`
That is, the first sending is triggered, the dio1 interrupt is also triggered (the transfer is completed), the module starts receiving and never leaves it, although according to the logic of the example it should repeat the sending once a second. Or is this normal behavior for this example?

[Code here](https://pastebin.com/TkAV7TTm)

jgromes, 2024-03-10:
>  the module starts receiving and never leaves it, although according to the logic of the example it should repeat the sending once a second.

No, if the first module never hears back then it will not attempt to send another packet.

Looking at the code and the ping pong example itself, it seems to be missing `ICACHE_RAM_ATTR` for the interrupt service function `setFlag`. Can you try adding it?

```c++
#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  // we sent or received  packet, set the flag
  operationDone = true;
}
```

whoim2, 2024-03-10:
> No, if the first module never hears back then it will not attempt to send another packet.

Then there is no problem, I'm trying to use one module.
Actually the simple example I used was because my working code for 1278/1276 didn't work with 1268. This is an RC radio control system. ICACHE_RAM_ATTR is present there. It looks the same: the first transmission is in progress, but for some reason there is no interrupt, although in this example there is one (end of transmission).

Later I will try to refine your example for constant transmission on timeout (no response), and if everything works, I will look for the reason in my main code. Thanks!
I'm not closing the topic yet, I'll post the result soon.

jgromes, 2024-03-10:
It seems to me there is no actual issue in that case. Hence I will convert this to a discussion instead.