jgromes/RadioLib #1026: Add method to set io-home mode

leobel96, 2024-03-23:
I noticed that in `SX127x::configFSK()`, the io-homecontrol compatibility is set to off  (`RADIOLIB_SX127X_IO_HOME_OFF`) and there is no high level method/function to enable it. 

Is there a specific reason for that? If not, would it be possible to have a high level method to enable it?

Thank you!

jgromes, 2024-03-23:
It's disabled on purpose, there was never any reason to include io-homecontrol support in RadioLib. Do you see a need for it?

leobel96, 2024-03-23:
Yes, I'm trying to emulate an io-homecontrol remote and enabling the io-homecontrol mode seems like an obvious choice for my application 🙂

jgromes, 2024-03-24:
And do you knnow what else needs to be done to enable it? The SX127x datasheet doesn't really tell:

[image]

RadioLib has [god mode](https://github.com/jgromes/RadioLib/wiki/God-Mode) which you can use to directly access SPI registers. Feel free to experiment using that, but I think this is going to be slightly more complicated than just setting a single bit.

Velocet, 2024-03-25:
If you enable this bit then the following things happen:
- io-homecontrol UART encoded SyncWord can be used directly: 0xFF33. So there's some decoding going on...
- Automatic CRC Check incl. removal/insertion
- Now Variable Length Paket Mode can be used
- Correct Parsing/Insertion of PacketSize: only 5 bits from first payload byte are used to define size of paket
- SyncSize (Size of Sync word) = (SyncSize + 1 bytes) becomes (SyncSize bytes) if ioHomeOn=1
- `IoHomePowerFrame:1; // CRC LFSR init: 0=0x1d0f, 1=0x0000=powerlink`

This mode is (at least) available in the following model ranges:
- SX127x
- SX123x
- derivatives (which use the same registers): RFM66W, RFM95/96/97/98(W), etc.
- ADF7022

@leobel96
IMHO: Since newer LoRa32 boards all use a SX126x you should not invest any time into this mode. But for testing it is great. As jgromes suggested you can always use the light variant of the god mode and set the bit yourself. If you have any questions feel free to join our chat.

leobel96, 2024-03-25:
@Velocet I'm actually glad that you joined the chat: my idea was to use the great investigation you did (https://github.com/Velocet/iown-homecontrol) to come up with a library similar to https://github.com/cridp/IOwnHC_ESP8266_ESP32-SX1276_CC1101but more flexible (thanks to Radiolib) and with more features. I noticed that they were setting the io-homecontrol register, so I thought that it was something mandatory for this application. Reading the datasheet of the SX1276, it just says:
> Please contact your local Semtech representativefor details on its implementation.

jgromes, 2024-03-26:
Thanks for the additional info @Velocet!

I still don't think it's worth it to have additional methods to enable/disable io-homecontrol, especially since it's still unclear to me what other steps are needed apart from setting the regiter bit.

Since a workaround using low-level/godmode access exists, I will close this issue now. However, I also don't want any future libraries to rely on this workaround. So @leobel96 if in you get past the inital testing into some future library, I will be happy to accept PRs addressing this.

closed 2024-03-26 by jgromes

Velocet, 2024-03-27:
@leobel96 We can fork RadioLib and build upon the new LoraWan implementation. Just implement it like RadioLib did with LoraWan as a module/protocol. This way we could use upstream RadioLib with just some additions.

leobel96, 2024-03-27:
@Velocet sure. Is enabling the io-homecontrol mode mandatory though? Isn't it feasible just setting the right parameters (bw, freq, sync word, ...) that are currently accessible?

Velocet, 2024-03-28:
Nope, the ioHome mode is not needed but makes things easier for testing. You can just rely on this mode and build your iohc lib around it but from my point of view this would be sad as this could only be used from folks who happen to own a SX127x based LoRa32.

I forked RadioLib and will add you as a contributor? Sadly i won‘t have much time till the end of next week (got a lot of work to do). The plan is to write an abstraction for the RadioLib supported modules which sets the same Rx/Tx values for every module.

If you just want the raw values for iohc Rx/Tx:
Frequency: 868,95 MHz
Deviation: 19,2 kHz
BaudRate: 38400 bps
Encoding: NRZ
Preamble: 0x55 (at least three byte)
SyncWord: 0xFF33 (encode as UART with LSB first!)
Rx Bandwith: 25 - 50 kHz

Please join the Discord or Telegram if you have more questions as i don’t want to abuse this RadioLib issue for iohc related stuff.

leobel96, 2024-03-31:
@Velocet If the ioHome mode is not needed, I would rather not set it and try with the configuration you sent me (also considering that the ioHome mode is not documented at all). I had a look at the Telegram channel and I saw some working configurations there as well. Thank you a lot for your help!