jgromes/RadioLib #1030: ESP8266 store Session in EEPROM not supported (Issue)

Lasermax3000, 2024-03-24:
At first thanks for this very intuitve libary. It is really easy to understand and makes this sometimes complicated LORA WAN and Communication with different RadioModules very easy to understand and easy to debug (nice piece of Software :).

To my issue, i use the lastest Version 6.4.2 with a RFM95 (alias SX1276) Radio Module with a NodeMCU **ESP8266.**
I want to connect to the TTN by Joining with OTAA procedure. As i have noticed so far, you have to store some parameters after joning persistently in the EEPROM (Part of the ESP Flash Memory) to restore a previous joined session after Reboot or deepsleep. 

After a short look in the Libary source Code i found out that there is **no EEPROM storage for the ESP8266 implemented** jet. 
So i tried it myself and  edited only a few Lines as follows, and it seems to do the Job now. 
Hopefully i haven´t forgotten to edit something else anywhere in the Code...?
Maybe you could add this to future releases, as the ESP8266 has also an EEPROM support like the ESP32 already has.

[image]
[image]

jgromes, 2024-03-24:
Persistent memory management has been removed from RadioLib upstream and will be released in the next update (hopefully coming soon). See #1017 

After this, management of persistent memory will be up to the user, since the many platforms RadioLib supports differ significantly in the persistent memory they have.

StevenCellist, 2024-03-24:
To comfort you: we will provide some examples of how you can still use persistent sessions. I will make sure to add ESP8266 to that, but it may take a week or maybe two before I can get around to that. It is likely that this will end up in [this repository](https://github.com/radiolib-org/radiolib-persistence).
Until then, you can use the fix you applied yourself :)

HeadBoffin, 2024-03-24:
Could someone repost the fix - coloured text on black background is hard for my failing eyeballs to read.

Lasermax3000, 2024-03-26:
Ok thanks, i see you are already working on a better solution ;), so this issue can be closed.  
@HeadBoffin Sorry for the bad colouring,  but as this might be solved in future releases i think it´s not necessary to repost the fix.

closed 2024-03-26 by Lasermax3000

StevenCellist, 2024-03-27:
@Lasermax3000 please see https://github.com/radiolib-org/radiolib-persistence/issues/1