jgromes/RadioLib #1031: LoRaWAN - Frame Count Reset

mofm, 2024-03-24:
The example below works fine until 50th frame count and than reset frame counter. ChirpStack Network Server gives frame counter reset or rollover error. 
When I try it without deep sleep mode, it works fine.

```
#include <RadioLib.h>
#include <LoraMessage.h>
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include "boards.h"
#include "secrets.h"

#define SEALEVELPRESSURE_HPA (1013.25)
Adafruit_BME280 bme; // I2C

// Uplink message payload size
// The maximum allowed for all data rates is 51 bytes.
const uint8_t PAYLOAD_SIZE = 51;

// Uplink payload buffer
static uint8_t loraData[PAYLOAD_SIZE];

SX1262 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

// create the node instance on the EU-868 band
// using the radio module and the encryption key
// make sure you are using the correct band
// based on your geographical location!
LoRaWANNode node(&radio, &EU868);

// for fixed bands with subband selection
// such as US915 and AU915, you must specify
// the subband that matches the Frequency Plan
// that you selected on your LoRaWAN console
/*
  LoRaWANNode node(&radio, &US915, 2);
*/

void sensorBegin() {
    unsigned status;
    status = bme.begin(); // default settings
    if (!status) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1) delay(10);
    } else {
        Serial.println("BME280 sensor found!");
    }
}

void readSensor(LoraEncoder & encoder) {
    float temperature = bme.readTemperature();
    float humidity = bme.readHumidity();
    float pressure = bme.readPressure() / 100.0F;
    float altitude = bme.readAltitude(SEALEVELPRESSURE_HPA);
    encoder.writeRawFloat(temperature);
    encoder.writeRawFloat(humidity);
    encoder.writeRawFloat(pressure);
    encoder.writeRawFloat(altitude);
}

void helper_deep_sleep(int seconds = 0) {
    // It seems to make no sense to do a .begin() here, but in case the radio is
    // not interacted with at all before sleep, it will not respond to just
    // .sleep() and then consumes 800 µA more than it should in deep sleep.
    radio.begin();
    // 'false' here is to not have a warm start, we re-init the after sleep.
    radio.sleep(false);
    u8g2->sleepOn();
    SPI.end();
[... 181 lines cut ...]
    // by calling "saveSession" which allows retrieving the session after reboot or deepsleep
    node.saveSession();

    // Calculate minimum duty cycle delay (per FUP & law!)
    uint32_t interval = node.timeUntilUplink();

    // And then pick it or our MINIMUM_DELAY, whichever is greater
    uint32_t delayMs = max(interval, (uint32_t)MINIMUM_DELAY * 1000);
  
    Serial.printf("Next TX in %i s\n", delayMs/1000);

    Serial.println("Deep sleep now");
    // Make sure this still prints before sleep
    delay(10);

    // And off to bed we go
    helper_deep_sleep((delayMs / 1000) - 5);
}
```

**Additional info (please complete):**
 - MCU: ESP32S3
 - Wireless module type: SX1262
 - Arduino IDE version: 1.8.19
 - Library version: 6.4.2

StevenCellist, 2024-03-24:
This problem is solved in upstream, it was a problem I mistakenly introduced in 6.4.0. If you wait for one or two days, @jgromes will likely release 6.5.0 and it should be solved!

@ropg as you can see, it took this long before someone encountered this bug again ;)

mofm, 2024-03-24:
Thanks @StevenCellist . I'm waiting for the new release.

closed 2024-03-24 by mofm