jgromes/RadioLib #1037: LoRaWan - Raspberry Pi 3B error -6 on method LoRaWANNode::sendReceive()

SocioPJ, 2024-03-26:
Hi!

Im using Raspberry Pi 3B to transmit LoRaWAN messages but im getting the error -6 on method sendReceive(). The module im using is the SX1272.

Pinout: 
**NSS** pin:   8
**DIO0** pin:  17 
**DIO1** pin:  13 
**RST** pin:  4

The following code is: 

```c
#include <RadioLib.h>
#include "PiHal.h"
#include <string>
#include <stdio.h>

PiHal* hal = new PiHal(1);

SX1272 radio = new Module(hal, 8, 17, 4, 13);

LoRaWANNode node(&radio, &US915, 2);

uint32_t devAddr = 0x2******3;

uint64_t devEUI = 0x2**************3;

uint8_t nwkKey[] = { 0x2B, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0x3C };

uint8_t appKey[] = { 0x2B, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX, 0x3C };

int beginRadio(SX1272 radio){
  int state = radio.begin();
  if(state == RADIOLIB_ERR_NONE){
    printf("[Radio Begin]...ok!\n");
    return (state);
  } else {
    printf("[Radio Begin]...error: %i\n",state);
    return (-1);
  }
}

int beginABPLora(){
  int state = node.beginABP(devAddr, nwkKey, appKey);
  if(state >= RADIOLIB_ERR_NONE){
    printf("[LoRa ABP Begin]...ok!\n");
    return(state);
  } else{
    printf("[LoRa ABP Begin]...error: %i\n",state);
    return (-1);
  }
}
int main(){
  printf("[Radio]...ok!\n");
  printf("[LoRa Node]...ok!\n");

  int state = beginRadio(radio);
  state = beginABPLora();

  // node.setADR(true);
  // node.setDutyCycle(true, 1250);
  // node.setDwellTime(true, 1000);
  
  uint8_t count = 0;
  while(true){
    const char *strUp = "hello!";
    uint8_t strDown[256];
    size_t lenDown = 0;
    uint8_t port = 10;
    int16_t state = node.sendReceive(strUp, port, strDown, &lenDown);
    // int16_t state = node.uplink(strUp, port);
    if(state == RADIOLIB_ERR_NONE){
      printf("[LoRa sendReceive]...ok!\n");
      printf("[LoRaWAN] Data:\t\t");
      size_t dataDownLength = sizeof(strDown) / sizeof(strDown[0]);

      if(dataDownLength > 0){
        printf("%s",strDown);
      } else {
        printf("<MAC commands only>");
      }
    } else{
      printf("[LoRa sendReceive]...error: %i\n",state);
      return (-1);
    }

    uint32_t minimunDelay = 60000;
    uint32_t interval = node.timeUntilUplink();
    uint32_t delayMs = std::max(interval,minimunDelay);
    hal->delay(delayMs);
  }
  return(0);
}
```

The error im getting:

```bash
[Radio]...ok!
[LoRa Node]...ok!
[Radio Begin]...ok!
[LoRa ABP Begin]...ok!
[LoRa sendReceive]...error: -6
```

**Additional information:**
- Library version 6.4.2

StevenCellist, 2024-03-26:
Please enable the RADIOLIB_DEBUG flag and post the output you get of one complete uplink, including integrated timestamps from your terminal. I cannot help you if the log timestamps are missing.

SocioPJ, 2024-03-27:
Hi, tks for fast response!

The log output:

```bash
RadioLib Debug Info
Version:  6.4.2.0
Platform: Generic
Compiled: Mar 27 2024 08:36:23

M       SX127x
[Radio Begin]...ok!
Didn't restore session (checksum: 0, mode: 0)
First 16 bytes of NVM:
0000000 d8 26 b1 74 55 00 00 00 48 88 44 ca 7f 00 00 00 | .&.tU...H.D.....
Wiping EEPROM and starting a clean session
Setting up fixed channels (subband 2)
exe MAC CID = 03, len = 4
ADR REQ: dataRate = 0, txPower = 0, chMask = 0x0002, chMaskCntl = 07, nbTrans = 0
PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
ADR mask: clearing channels
mask[7] = 0x0002
UL: 65 1 904.600 (4 - 4) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
ADR ANS: status = 0x07
exe MAC CID = 03, len = 4
ADR REQ: dataRate = 0, txPower = 0, chMask = 0xff00, chMaskCntl = 00, nbTrans = 0
PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
mask[0] = 0xff00
UL: 65 1 904.600 (4 - 4) | DL: 0 0  0.000 (0 - 0)
UL: 8 1 903.900 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 9 1 904.100 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 10 1 904.300 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 11 1 904.500 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 12 1 904.700 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 13 1 904.900 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 14 1 905.100 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 15 1 905.300 (0 - 3) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
ADR ANS: status = 0x07
exe MAC CID = 03, len = 4
ADR REQ: dataRate = 0, txPower = 0, chMask = 0x0000, chMaskCntl = 00, nbTrans = 0
PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
ADR ANS: status = 0x07
exe MAC CID = 04, len = 1
Max duty cycle: 1/2^0
exe MAC CID = 05, len = 4
Rx param REQ: rx1DrOffset = 0, rx2DataRate = 8, freq = 923.299988
Rx param ANS: status = 0x07
exe MAC CID = 08, len = 1
RX timing: delay = 1 sec
exe MAC CID = 09, len = 1
TX timing: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
exe MAC CID = 0c, len = 1
ADR param setup: limitExp = 6, delayExp = 5
exe MAC CID = 0f, len = 1
Rejoin setup: maxTime = 15, maxCount = 10
[LoRa ABP Begin]...ok!

PHY: Frequency UL = 905.100 MHz
PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
FcntUp: 0
uplinkMsg pre-MIC:
0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0000010 40 43 04 01 20 80 00 00 0a c6 9a a6 fa 06 08 00 | @C.. ...........
0000020 00 00 00                                        | ...               
Uplink sent <-- Rx Delay start

PHY: Frequency DL = 926.900 MHz
PHY: SF = 10, BW = 500.000 kHz, CR = 4/5
Opening Rx1 window (72000 us timeout)... <-- Rx Delay end 
Closing Rx1 window
PHY: Frequency DL = 923.300 MHz
PHY: SF = 12, BW = 500.000 kHz, CR = 4/5
Opening Rx2 window (186000 us timeout)... <-- Rx Delay end 
Closing Rx2 window
[LoRa sendReceive]...error: -6
```

HeadBoffin, 2024-03-27:
Error -6 for sendReceive is all about the timings, which is why @StevenCellist said:

> including integrated timestamps from your terminal. I cannot help you if the log timestamps are missing.

SocioPJ, 2024-03-27:
```bash
09:14:45.995 -> [Radio]...ok!
09:14:46.001 -> [LoRa Node]...ok!
09:14:46.010 -> 
09:14:46.017 -> RadioLib Debug Info
09:14:46.024 -> Version:  6.4.2.0
09:14:46.032 -> Platform: Generic
09:14:46.038 -> Compiled: Mar 27 2024 09:05:58
09:14:46.045 -> 
09:14:46.053 -> M       SX127x
09:14:46.060 -> [Radio Begin]...ok!
09:14:46.067 -> Didn't restore session (checksum: 0, mode: 0)
09:14:46.074 -> First 16 bytes of NVM:
09:14:46.082 -> 0000000 d8 26 0e 79 55 00 00 00 58 14 38 e6 7f 00 00 00 | .&.yU...X.8.....
09:14:46.089 -> Wiping EEPROM and starting a clean session
09:14:46.096 -> Setting up fixed channels (subband 2)
09:14:46.103 -> exe MAC CID = 03, len = 4
09:14:46.110 -> ADR REQ: dataRate = 0, txPower = 0, chMask = 0x0002, chMaskCntl = 07, nbTrans = 0
09:14:46.117 -> PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
09:14:46.124 -> ADR mask: clearing channels
09:14:46.132 -> mask[7] = 0x0002
09:14:46.139 -> UL: 65 1 904.600 (4 - 4) | DL: 0 0  0.000 (0 - 0)
09:14:46.147 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.154 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.161 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.168 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.176 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.184 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.190 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.198 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.205 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.212 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.220 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.227 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.234 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.244 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.250 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.259 -> ADR ANS: status = 0x07
09:14:46.265 -> exe MAC CID = 03, len = 4
09:14:46.273 -> ADR REQ: dataRate = 0, txPower = 0, chMask = 0xff00, chMaskCntl = 00, nbTrans = 0
09:14:46.280 -> PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
09:14:46.287 -> mask[0] = 0xff00
09:14:46.294 -> UL: 65 1 904.600 (4 - 4) | DL: 0 0  0.000 (0 - 0)
09:14:46.301 -> UL: 8 1 903.900 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.308 -> UL: 9 1 904.100 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.315 -> UL: 10 1 904.300 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.323 -> UL: 11 1 904.500 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.329 -> UL: 12 1 904.700 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.337 -> UL: 13 1 904.900 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.343 -> UL: 14 1 905.100 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.352 -> UL: 15 1 905.300 (0 - 3) | DL: 0 0  0.000 (0 - 0)
09:14:46.359 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.366 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.374 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.380 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.387 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.395 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.403 -> UL: 255 0  0.000 (0 - 0) | DL: 0 0  0.000 (0 - 0)
09:14:46.410 -> ADR ANS: status = 0x07
09:14:46.418 -> exe MAC CID = 03, len = 4
09:14:46.425 -> ADR REQ: dataRate = 0, txPower = 0, chMask = 0x0000, chMaskCntl = 00, nbTrans = 0
09:14:46.432 -> PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
09:14:46.440 -> ADR ANS: status = 0x07
09:14:46.447 -> exe MAC CID = 04, len = 1
09:14:46.456 -> Max duty cycle: 1/2^0
09:14:46.464 -> exe MAC CID = 05, len = 4
09:14:46.470 -> Rx param REQ: rx1DrOffset = 0, rx2DataRate = 8, freq = 923.299988
09:14:46.478 -> Rx param ANS: status = 0x07
09:14:46.485 -> exe MAC CID = 08, len = 1
09:14:46.493 -> RX timing: delay = 1 sec
09:14:46.500 -> exe MAC CID = 09, len = 1
09:14:46.507 -> TX timing: dlDwell = 0, ulDwell = 1, maxEirp = 30 dBm
09:14:46.514 -> exe MAC CID = 0c, len = 1
09:14:46.521 -> ADR param setup: limitExp = 6, delayExp = 5
09:14:46.528 -> exe MAC CID = 0f, len = 1
09:14:46.535 -> Rejoin setup: maxTime = 15, maxCount = 10
09:14:46.543 -> [LoRa ABP Begin]...ok!
09:14:46.551 -> 
09:14:46.558 -> PHY: Frequency UL = 904.700 MHz
09:14:46.566 -> PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
09:14:46.573 -> FcntUp: 0
09:14:46.581 -> uplinkMsg pre-MIC:
09:14:46.587 -> 0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
09:14:46.595 -> 0000010 40 43 04 01 20 80 00 00 0a c6 9a a6 fa 06 08 00 | @C.. ...........
09:14:46.603 -> 0000020 00 00 00                                        | ...               
09:14:46.610 -> Uplink sent <-- Rx Delay start
09:14:46.617 -> 
09:14:46.625 -> PHY: Frequency DL = 925.700 MHz
09:14:46.632 -> PHY: SF = 10, BW = 500.000 kHz, CR = 4/5
09:14:46.640 -> Opening Rx1 window (72000 us timeout)... <-- Rx Delay end 
09:14:46.647 -> Closing Rx1 window
09:14:46.654 -> PHY: Frequency DL = 923.300 MHz
09:14:46.661 -> PHY: SF = 12, BW = 500.000 kHz, CR = 4/5
09:14:46.669 -> Opening Rx2 window (186000 us timeout)... <-- Rx Delay end 
09:14:46.676 -> Closing Rx2 window
09:14:46.683 -> [LoRa sendReceive]...error: -6
```

HeadBoffin, 2024-03-27:
OK, so there are some things we'll need to check out, but it's not clear if there is an issue with the uplink or you are just getting the normal message that says there wasn't a downlink that has previously been tagged as an error but is actually the expected result for the majority of uplinks.

Does the uplink arrive on the console, which LNS are you using, have you looked up the error code & does it make sense to you in the context of LoRaWAN?

SocioPJ, 2024-03-27:
First i though that the transmission is not arriving to the gateway and that's the reason it's not receiving the downlink, so i put a gateway on my desk and still not working. But then i check GQRX software to see if there is any uplink signal on 860 MHZ frequency (the lowest to this module and less signal traffic) and there is not.

StevenCellist, 2024-03-27:
Yeah so I've been looking at your code and RadioLib's Pi support in general, and figured that the problem is the thing that struck me the first: your `beginRadio(SX1272 radio)` function should use pass-by-reference instead of pass-by-value, or in other words: it needs to use a pointer: `beginRadio(SX1272 &radio)`.
And subsequently, calls to `radio` methods in that function need to use `->` instead of `.`, i.e. `radio->begin()`. But I expect you get the gist of it.

StevenCellist, 2024-03-27:
By the way - it also appears you registered your device as LoRaWAN v1.0.4, or you forgot to put in the other two keys. I _highly_, and I cannot stress it enough, _really_ recommend using LoRaWAN v1.1 instead. LoRaWAN v1.0.4 is not very much supported by RadioLib, especially for ABP. Your device will likely not work on v1.0.4.

SocioPJ, 2024-03-27:
Ok, by next week i'll change to OTAA and i say what happended.
Tks :)

HeadBoffin, 2024-03-27:
>  by next week i'll change to OTAA and i say what happended.

That as well, but LW 1.1, NOT v1.0.4 ...

StevenCellist, 2024-04-06:
@SocioPJ do you have an update on this? Any luck with your Pi adventures (with LW v1.1 & OTAA)?

SocioPJ, 2024-04-06:
Hi, srry for late response.
So basically the network server that I use don't support LW 1.1 so I have to continue with LW1.0.4.
I also discovery that my gateways doesn't send a downlink message when received a uplink, and that is the reason I receive the error -6 (my guess). I changed the LoRaWAN method to .uplink() and I was able to send one message but in the following messages I receive error -1108. I can't show the log right now since I am not with my pc. By wednesday i'll update with the log.

Although the first message was send successfully, the gateway didn't received the signal (I captured the signal on GQRX).

StevenCellist, 2024-04-06:
> the network server that I use don't support LW 1.1 so I have to continue with LW1.0.4

1.0.4 is really not supported, and that means you can't use Helium. You will have to use something other than RadioLib if you want to use Helium.

> my gateways doesn't send a downlink message when received a uplink, and that is the reason I receive the error -6 (my guess)

No, that is not the issue; the issue is in your code unfortunately, as I mentioned (you need to use a pointer to the radio in the beginRadio function).

> I changed the LoRaWAN method to .uplink() and I was able to send one message but in the following messages I receive error -1108

That is by design: LoRaWAN requires you to listen to potential downlinks; you cannot just skip a downlink!

StevenCellist, 2024-04-10:
Closing as OP unwilling to conform to given feedback, and instead bends universe against the law and intentionally breaking functionality.

closed 2024-04-10 by StevenCellist

HeadBoffin, 2024-04-10:
There's no point looking confused - making changes to code & settings that are known to work against an untested LNS using the wrong version to get them going on your own platform is somewhat counter-intuitive. You need to find the reason it doesn't work compared to a working setup, compare & contrast. Almost universally the TTN forum never recommends using a Pi as a device unless it is using a RAK811 (a popular HAT) or other AT based modem due to timing issues. And radical changes to the channel plan for a country have all sorts of consequences.

SocioPJ, 2024-04-10:
I apologize for not being clear in all aspects.
The LNS and gateway that I use have some modifications and are private. I'm not using public LNS. I wasn't the one who determined these changes, it was my boss at a job he did previously and he asked me to continue with these changes. Therefore, it was necessary to make this adaptation to Australian frequencies.
After making this adaptation, I was able to receive data from LoRa in my LNS.
I wasn't ignoring the feedback, but there are certain things I need to follow, like continue using ABP and LoRa v1.0.4. Regarding the code, I had redone it based on the examples for Raspberry.
I stopped using the sendReceive() method because my gateway does NOT send a downlink in response, so I thought it makes more sense to use uplink().
I understand that using the Pi is not recommended, but it has certain functionality that is necessary for my project.
Anyway, I started studying this a month ago, so I still don't know some LoRa concepts. If possible, I'll accept help with the errors I'm having, if not, thank you for everything :)

HeadBoffin, 2024-04-11:
**RadioLib is developed by volunteers.** There are no restrictions on using it for commercial projects AFAIK and if you were using TTS, LW1.1 and suitable known good tested MCU platform we'd not be here.

But you are using a RPi with an unsupported LW version on an LNS that could have any number of changes that we aren't aware of for a commercial project.

Phrases like "because my gateway does NOT send a downlink in response," indicate that you have some way to go before you understand what LW is all about and LW is not a system that falls neatly in to place even with known setups.

It's not appropriate to expect free assistance to help solve what can only be described as a disaster that's potentially not finished unfolding. And FWIW, it's not fair on you for someone to expect you to learn new tech AND use a non-standard setup.

You may want to hunt out a Pi implementation of LMIC which is only 1.0.3. Or there is a python version, not sure what version that supports. Or commission someone to sort this out but I'd suspect they will ask a lot of WHY questions (like why change the LNS, why use ABP, why use a Pi as a device) and then recommend reverting to standard kit to get the project done.

Good luck!