jgromes/RadioLib #1042: RP2040-LoRa Development kit

ObiLeek, 2024-03-30:
I've been puzzling for a few days how to make this development kit work with this library. Here is the link to the wiki: [https://www.waveshare.com/wiki/RP2040-LoRa](https://www.waveshare.com/wiki/RP2040-LoRa)

I'm trying the sample examples with a modified pinout:
```
#define NSS_PIN 13 // SPI Chip Selection (low active)
#define DIO1_PIN 16 // SX1262 interrupt output pin or special function pin
#define RESET_PIN 23 // Reset pin, low active
#define BUSY_PIN 18 // SX1262 BUSY Pin
```

But it just won't and won't work, I keep getting an error:
`failed, code -2
`

jgromes, 2024-03-31:
I would suggest to fill in the "module not working" template. In there, you will find a link to RadioLib [Troubleshooting Guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide), which I believe will help you greatly ;)

ObiLeek, 2024-03-31:
That was the first thing I did, but to no success.

HeadBoffin, 2024-03-31:
> That was the first thing I did, but to no success.

If you filled the template in, can you post it here, that's the first step in getting some meaningful assistance.

GUVWAF, 2024-03-31:
Similar to #729, for this module you have to use hardware device SPI1 (instead of the default SPI0) and configure the pins for SPI1 accordingly.

HeadBoffin, 2024-03-31:
Ah, that's where that nugget of information had disappeared to, thanks for finding it.

ObiLeek, 2024-04-01:
Thank you, that was exactly right. In my particular case, like this:

```
Module* mod
MbedSPI SPI1(24, 15, 14);

void setup() {
      mod = new Module(LORA_CS, LORA_DIO1, LORA_RST, LORA_BUSY, SPI1);
}
```

In my specific case with this modul [LINK](https://www.waveshare.com/wiki/RP2040-LoRa),  `tcxoVoltage `had to be set to zero.

closed 2024-04-01 by jgromes