jgromes/RadioLib #1049: LoRaWAN ESP32 (Heltec WSL V3) reboot with stack error

michapr, 2024-04-02:
**Describe the bug**
I'm using for test the "LoRaWAN_Reference.ino" with the only modification node.setADR(true);
When call these two command requests: link_check and device_time the ESP32 will reboot after delay()
```
  if(fcntUp % 64 == 0) {
    Serial.println(F("[LoRaWAN] Requesting LinkCheck and DeviceTime"));
    node.sendMacCommandReq(RADIOLIB_LORAWAN_MAC_LINK_CHECK);
    node.sendMacCommandReq(RADIOLIB_LORAWAN_MAC_DEVICE_TIME);
    state = node.sendReceive(uplinkPayload, sizeof(uplinkPayload), Port, downlinkPayload, &downlinkSize, true, &uplinkDetails, &downlinkDetails);
``` 

Debug log:
```
09:29:46.961 -> 
09:29:46.961 -> Stack smashing protect failure!
09:29:46.961 -> 
09:29:46.961 -> 
09:29:46.961 -> abort() was called at PC 0x4201422b on core 1
09:29:46.961 -> 
09:29:46.961 -> 
09:29:46.961 -> Backtrace:0x403777be:0x3fce2ed00x4037a5ad:0x3fce2ef0 0x4037fba5:0x3fce2f10 0x4201422b:0x3fce2f90 0x42001fa9:0x3fce2fb0 0x4200abf1:0x3fce3030 
09:29:46.961 -> 
09:29:46.961 -> 
09:29:46.961 -> 
09:29:46.961 -> 
09:29:46.961 -> ELF file SHA256: 0000000000000000
09:29:46.961 -> 
09:29:46.961 -> Rebooting...
09:29:46.961 -> �ESP-ROM:esp32s3-20210327
09:29:47.012 -> Build:Mar 27 2021
09:29:47.012 -> rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
09:29:47.012 -> Saved PC:0x4202b7b2
09:29:47.012 -> SPIWP:0xee
09:29:47.012 -> mode:DIO, clock div:1
09:29:47.012 -> load:0x3fce3808,len:0x43c
09:29:47.012 -> load:0x403c9700,len:0xbec
09:29:47.012 -> load:0x403cc700,len:0x2a3c
09:29:47.012 -> SHA-256 comparison failed:
09:29:47.012 -> Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
09:29:47.012 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
09:29:47.039 -> Attempting to boot anyway...
09:29:52.113 -> 
09:29:52.113 -> Setup
09:29:52.113 -> Initalise the radio
09:29:52.113 -> RLB_DBG: RadioLib Debug Info
09:29:52.113 -> RLB_DBG: Version:  6.5.0.0
09:29:52.113 -> RLB_DBG: Platform: ESP32
09:29:52.113 -> RLB_DBG: Compiled: Mar 29 2024 18:40:51
09:29:52.113 -> 
09:29:52.113 -> RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
09:29:52.161 -> RLB_DBG: 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
09:29:52.161 -> RLB_DBG: 
09:29:52.161 -> RLB_DBG: M	SX126x
09:29:52.192 -> Join ('login') to the LoRaWAN Network
09:29:52.192 -> RLB_PRO: Configuration mismatch (checksum: 0, mode: 0, class: 0, plan: 1)
09:29:52.192 -> RLB_PRO: Nonces buffer:
09:29:52.192 -> RLB_PRO: 0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
09:29:52.192 -> RLB_PRO: Clearing buffer and starting fresh
09:29:52.192 -> RLB_PRO: Setting up dynamic channels
09:29:52.192 -> RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
09:29:52.224 -> RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
09:29:52.224 -> RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
09:29:52.224 -> RLB_PRO: [MAC] 0x03
09:29:52.224 -> RLB_PRO: 0000000 20 00 00 80                                     |  ...              
09:29:52.224 -> RLB_PRO: LinkADRReq: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 0, nbTrans = 0
09:29:52.224 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
09:29:52.256 -> RLB_PRO: PHY: TX = 16 dBm
09:29:52.256 -> RLB_PRO: LinkADRAns: status = 0x07
09:29:52.256 -> RLB_PRO: [MAC] 0x04
09:29:52.256 -> RLB_PRO: 0000000 07                                              | .                 
[... 121 lines cut ...]
09:30:04.790 -> [LoRaWAN] Frequency error:	1913.28 Hz
09:30:04.790 -> [LoRaWAN] Event information:
09:30:04.790 -> [LoRaWAN] Confirmed:	0
09:30:04.790 -> [LoRaWAN] Confirming:	1
09:30:04.790 -> [LoRaWAN] Datarate:	5
09:30:04.790 -> [LoRaWAN] Frequency:	867.100 MHz
09:30:04.790 -> [LoRaWAN] Output power:	16 dBm
09:30:04.790 -> [LoRaWAN] Frame count:	0
09:30:04.790 -> [LoRaWAN] Port:		0
09:30:04.790 -> [LoRaWAN] LinkCheck margin:	25
09:30:04.830 -> [LoRaWAN] LinkCheck count:	1
09:30:04.830 -> [LoRaWAN] DeviceTime Unix:	1711787399
09:30:04.830 -> [LoRaWAN] DeviceTime second:	1/51
09:30:04.830 -> [LoRaWAN] Next uplink in 1194s

```

**To Reproduce**
use for test the "LoRaWAN_Reference.ino" with the only modification node.setADR(true);

**Additional info (please complete):**
 - MCU: ESP32S3 - Heltes WirelessStick lite V3
 - Wireless module type SX1262
 - Arduino IDE version  2.2.1
 - Library version 6.5.0

HeadBoffin, 2024-04-02:
Thanks for this - can you show the log BEFORE the crash, what happens after the reboot is useful but not quite as important as what happens leading up to the stack smash.

michapr, 2024-04-02:
**before** is the same as in the end of the log - next will be the same reboot (after the delay) ... No more information available...
```
09:30:04.830 -> [LoRaWAN] LinkCheck count:	1
09:30:04.830 -> [LoRaWAN] DeviceTime Unix:	1711787399
09:30:04.830 -> [LoRaWAN] DeviceTime second:	1/51
09:30:04.830 -> [LoRaWAN] Next uplink in 1194s
```

HeadBoffin, 2024-04-02:
> No more information available...

So it's not happening anymore to recreate the log?

Version numbering at library level is making tracing the test logs rather messy - so we'll improve our process to help with that - I'll create a clean development folder to look at this one - I wouldn't normally do a LinkCheck just after a join as it's a bit redundant and means the gateway has gone deaf twice in short order. Consequently the reference shows the API calls but isn't really best practise. Watch this space -> [  ]

michapr, 2024-04-02:
I had to reflash my test device - it was running with your sleeping example (btw: running fine and stable .. ;) ...)
> I wouldn't normally do a LinkCheck just after a join

yes, right, of course. But time request make sense...

Here is a "fresh" log (after reset) with the requested detail:
```
12:22:38.440 -> ESP-ROM:esp32s3-20210327
12:22:38.440 -> Build:Mar 27 2021
12:22:38.462 -> rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
12:22:38.462 -> SPIWP:0xee
12:22:38.462 -> mode:DIO, clock div:1
12:22:38.462 -> load:0x3fce3808,len:0x43c
12:22:38.462 -> load:0x403c9700,len:0xbec
12:22:38.462 -> load:0x403cc700,len:0x2a3c
12:22:38.462 -> SHA-256 comparison failed:
12:22:38.462 -> Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
12:22:38.462 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
12:22:38.462 -> Attempting to boot anyway...
12:22:38.462 -> entry 0x403c98d8
12:22:43.497 -> 
12:22:43.497 -> Setup
12:22:43.497 -> Initalise the radio
12:22:43.497 -> RLB_DBG: RadioLib Debug Info
12:22:43.497 -> RLB_DBG: Version:  6.5.0.0
12:22:43.497 -> RLB_DBG: Platform: ESP32
12:22:43.497 -> RLB_DBG: Compiled: Apr  2 2024 12:21:45
12:22:43.543 -> 
12:22:43.543 -> RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
12:22:43.543 -> RLB_DBG: 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
12:22:43.543 -> RLB_DBG: 
12:22:43.543 -> RLB_DBG: M	SX126x
12:22:43.575 -> Join ('login') to the LoRaWAN Network
12:22:43.575 -> RLB_PRO: Configuration mismatch (checksum: 0, mode: 0, class: 0, plan: 1)
12:22:43.575 -> RLB_PRO: Nonces buffer:
12:22:43.575 -> RLB_PRO: 0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
12:22:43.607 -> RLB_PRO: Clearing buffer and starting fresh
12:22:43.607 -> RLB_PRO: Setting up dynamic channels
12:22:43.607 -> RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
12:22:43.607 -> RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
12:22:43.607 -> RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
12:22:43.607 -> RLB_PRO: [MAC] 0x03
12:22:43.607 -> RLB_PRO: 0000000 20 00 00 80                                     |  ...              
12:22:43.639 -> RLB_PRO: LinkADRReq: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 0, nbTrans = 0
12:22:43.639 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
12:22:43.639 -> RLB_PRO: PHY: TX = 16 dBm
12:22:43.639 -> RLB_PRO: LinkADRAns: status = 0x07
12:22:43.639 -> RLB_PRO: [MAC] 0x04
12:22:43.639 -> RLB_PRO: 0000000 07                                              | .                 
12:22:43.640 -> RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
12:22:43.671 -> RLB_PRO: [MAC] 0x05
12:22:43.671 -> RLB_PRO: 0000000 00 d2 ad 84                                     | ....              
12:22:43.671 -> RLB_PRO: RXParamSetupReq: rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525024
12:22:43.671 -> RLB_PRO: RXParamSetupAns: status = 0x07
12:22:43.671 -> RLB_PRO: [MAC] 0x08
12:22:43.671 -> RLB_PRO: 0000000 01                                              | .                 
12:22:43.671 -> RLB_PRO: RXTimingSetupReq: delay = 1 sec
12:22:43.704 -> RLB_PRO: [MAC] 0x09
12:22:43.704 -> RLB_PRO: 0000000 05                                              | .                 
12:22:43.704 -> RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 0, maxEirp = 16 dBm
12:22:43.704 -> RLB_PRO: [MAC] 0x0C
12:22:43.704 -> RLB_PRO: 0000000 65                                              | e                 
12:22:43.704 -> RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
12:22:43.740 -> RLB_PRO: [MAC] 0x0F
12:22:43.740 -> RLB_PRO: 0000000 fa                                              | .                 
12:22:43.740 -> RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
12:22:43.740 -> RLB_PRO: RejoinParamSetupAns: status = 0x03
12:22:43.806 -> RLB_PRO: 
12:22:43.806 -> RLB_PRO: PHY: Frequency UL = 868.300 MHz
12:22:43.806 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
[... 301 lines cut ...]
12:43:08.389 -> Downlink data: 
12:43:08.389 -> 9B36AEDB18EEA935217484683834110000
12:43:08.389 -> [LoRaWAN] RSSI:		-47.00 dBm
12:43:08.421 -> [LoRaWAN] SNR:		7.50 dB
12:43:08.421 -> [LoRaWAN] Frequency error:	1845.47 Hz
12:43:08.421 -> [LoRaWAN] Event information:
12:43:08.421 -> [LoRaWAN] Confirmed:	0
12:43:08.421 -> [LoRaWAN] Confirming:	1
12:43:08.421 -> [LoRaWAN] Datarate:	5
12:43:08.421 -> [LoRaWAN] Frequency:	867.900 MHz
12:43:08.421 -> [LoRaWAN] Output power:	14 dBm
12:43:08.421 -> [LoRaWAN] Frame count:	0
12:43:08.421 -> [LoRaWAN] Port:		0
12:43:08.421 -> [LoRaWAN] LinkCheck margin:	27
12:43:08.421 -> [LoRaWAN] LinkCheck count:	1
12:43:08.421 -> [LoRaWAN] DeviceTime Unix:	1712054583
12:43:08.460 -> [LoRaWAN] DeviceTime second:	1/3
12:43:08.460 -> [LoRaWAN] Next uplink in 1194s
13:03:02.844 -> 
13:03:02.844 -> Stack smashing protect failure!
13:03:02.877 -> 
13:03:02.877 -> 
13:03:02.877 -> abort() was called at PC 0x420142bf on core 1
...
```

michapr, 2024-04-02:
This is complete log with `sendMacCommandReq(RADIOLIB_LORAWAN_MAC_DEVICE_TIME) ` only...
```
13:09:05.203 -> ESP-ROM:esp32s3-20210327
13:09:05.203 -> Build:Mar 27 2021
13:09:05.203 -> rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
13:09:05.203 -> SPIWP:0xee
13:09:05.203 -> mode:DIO, clock div:1
13:09:05.203 -> load:0x3fce3808,len:0x43c
13:09:05.203 -> load:0x403c9700,len:0xbec
13:09:05.203 -> load:0x403cc700,len:0x2a3c
13:09:05.203 -> SHA-256 comparison failed:
13:09:05.203 -> Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
13:09:05.203 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
13:09:05.249 -> Attempting to boot anyway...
13:09:05.249 -> entry 0x403c98d8
13:09:10.322 -> 
13:09:10.322 -> Setup
13:09:10.322 -> Initalise the radio
13:09:10.322 -> RLB_DBG: RadioLib Debug Info
13:09:10.322 -> RLB_DBG: Version:  6.5.0.0
13:09:10.322 -> RLB_DBG: Platform: ESP32
13:09:10.322 -> RLB_DBG: Compiled: Apr  2 2024 12:21:45
13:09:10.322 -> 
13:09:10.368 -> RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
13:09:10.368 -> RLB_DBG: 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
13:09:10.368 -> RLB_DBG: 
13:09:10.368 -> RLB_DBG: M	SX126x
13:09:10.391 -> Join ('login') to the LoRaWAN Network
13:09:10.391 -> RLB_PRO: Configuration mismatch (checksum: 0, mode: 0, class: 0, plan: 1)
13:09:10.391 -> RLB_PRO: Nonces buffer:
13:09:10.391 -> RLB_PRO: 0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
13:09:10.423 -> RLB_PRO: Clearing buffer and starting fresh
13:09:10.423 -> RLB_PRO: Setting up dynamic channels
13:09:10.423 -> RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
13:09:10.423 -> RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
13:09:10.423 -> RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
13:09:10.423 -> RLB_PRO: [MAC] 0x03
13:09:10.423 -> RLB_PRO: 0000000 20 00 00 80                                     |  ...              
13:09:10.455 -> RLB_PRO: LinkADRReq: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 0, nbTrans = 0
13:09:10.455 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
13:09:10.455 -> RLB_PRO: PHY: TX = 16 dBm
13:09:10.455 -> RLB_PRO: LinkADRAns: status = 0x07
13:09:10.455 -> RLB_PRO: [MAC] 0x04
13:09:10.455 -> RLB_PRO: 0000000 07                                              | .                 
13:09:10.455 -> RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
13:09:10.487 -> RLB_PRO: [MAC] 0x05
13:09:10.487 -> RLB_PRO: 0000000 00 d2 ad 84                                     | ....              
13:09:10.487 -> RLB_PRO: RXParamSetupReq: rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525024
13:09:10.487 -> RLB_PRO: RXParamSetupAns: status = 0x07
13:09:10.487 -> RLB_PRO: [MAC] 0x08
13:09:10.487 -> RLB_PRO: 0000000 01                                              | .                 
13:09:10.487 -> RLB_PRO: RXTimingSetupReq: delay = 1 sec
13:09:10.519 -> RLB_PRO: [MAC] 0x09
13:09:10.519 -> RLB_PRO: 0000000 05                                              | .                 
13:09:10.519 -> RLB_PRO: TxParamSetupReq: dlDwell = 0, ulDwell = 0, maxEirp = 16 dBm
13:09:10.519 -> RLB_PRO: [MAC] 0x0C
13:09:10.519 -> RLB_PRO: 0000000 65                                              | e                 
13:09:10.519 -> RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
13:09:10.519 -> RLB_PRO: [MAC] 0x0F
13:09:10.562 -> RLB_PRO: 0000000 fa                                              | .                 
13:09:10.562 -> RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
13:09:10.562 -> RLB_PRO: RejoinParamSetupAns: status = 0x03
13:09:10.640 -> RLB_PRO: 
13:09:10.640 -> RLB_PRO: PHY: Frequency UL = 868.300 MHz
13:09:10.640 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
13:09:10.640 -> RLB_DBG: Timeout in 1853 ms
13:09:11.033 -> RLB_PRO: Join-request sent <-- Rx Delay start
13:09:11.033 -> RLB_PRO: 
13:09:11.033 -> RLB_PRO: PHY: Frequency DL = 868.300 MHz
13:09:11.033 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
[... 177 lines cut ...]
13:34:23.076 -> RLB_PRO: PHY: SF = 9, BW = 125.000 kHz, CR = 4/5
13:34:24.033 -> RLB_PRO: Opening Rx2 window (123424 us timeout)... <-- Rx Delay end 
13:34:24.160 -> RLB_PRO: Closing Rx2 window
13:34:24.160 -> [LoRaWAN] Next uplink in 180s
13:37:24.211 -> RLB_PRO: 
13:37:24.211 -> RLB_PRO: PHY: Frequency UL = 867.500 MHz
13:37:24.211 -> RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
13:37:24.211 -> RLB_PRO: Uplink (FcntUp = 3) decoded:
13:37:24.252 -> RLB_PRO: 0000000 c0 00 78 56 88 3e ce 3f 13 00 00 00 34 12 ba ab | ..xV.>.?....4...
13:37:24.252 -> RLB_PRO: 0000010 40 85 7e 0b 26 80 03 00 0a 08 3c b0 7b af 00 00 | @.~.&.....<.{...
13:37:24.351 -> RLB_DBG: Timeout in 257 ms
13:37:24.419 -> RLB_PRO: Uplink sent <-- Rx Delay start
13:37:24.419 -> RLB_PRO: 
13:37:24.419 -> RLB_PRO: PHY: Frequency DL = 867.500 MHz
13:37:24.419 -> RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
13:37:29.418 -> RLB_PRO: Opening Rx1 window (45856 us timeout)... <-- Rx Delay end 
13:37:29.496 -> RLB_PRO: Closing Rx1 window
13:37:29.496 -> RLB_PRO: PHY: Frequency DL = 869.525 MHz
13:37:29.496 -> RLB_PRO: PHY: SF = 9, BW = 125.000 kHz, CR = 4/5
13:37:30.431 -> RLB_PRO: Opening Rx2 window (123424 us timeout)... <-- Rx Delay end 
13:37:30.549 -> RLB_PRO: Closing Rx2 window
13:37:30.549 -> [LoRaWAN] Next uplink in 180s
```

Looks like it works without problem (reboot) - will not touch it for some time ... ;)

StevenCellist, 2024-04-02:
Thanks for the troubleshooting! Behaviour replicated, got some logs - I like big logs, I cannot lie. To be continued.

StevenCellist, 2024-04-02:
This issue is fixed with the referenced commit. Thank you for reporting! If you would want to use DeviceTimeReq or LinkCheckReq, make sure to use the latest version on GitHub; be aware that `setCSMA()` just got disabled, so account for that if you were to update.

closed 2024-04-02 by StevenCellist

michapr, 2024-04-06:
Sorry to say,... but I have got again a reboot after some time...
Same sketch with updated files and disabled `setCSMA()` 
For better tracking I have modified teh payload only
```
  uint8_t uplinkPayload[4];
  uplinkPayload[0] = highByte(fcntUp);
  uplinkPayload[1] = lowByte(fcntUp);   // See notes for high/lowByte functions
  uplinkPayload[2] = highByte(fcntDn);
  uplinkPayload[3] = lowByte(fcntDn);
```

Hewre is the "long debug" - maybe you can see here anything or have an idea...

```
[05/04/24 - 20:06:43.377] RLB_PRO: 
[05/04/24 - 20:06:43.379] RLB_PRO: PHY: Frequency UL = 867.100 MHz
[05/04/24 - 20:06:43.385] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 20:06:43.389] RLB_PRO: Uplink (FcntUp = 504) decoded:
[05/04/24 - 20:06:43.393] RLB_PRO: 0000000 c0 00 78 56 88 3e ce 3f 13 00 00 00 34 12 ba ab | ..xV.>.?....4...
[05/04/24 - 20:06:43.401] RLB_PRO: 0000010 40 29 f1 0b 26 80 f8 01 0a f1 ae 15 8a af 00 00 | @)..&...........
[05/04/24 - 20:06:43.409] RLB_PRO: 0000020 14                                              | .                 
[05/04/24 - 20:06:43.415] RLB_DBG: Timeout in 257 ms
[05/04/24 - 20:06:43.468] RLB_PRO: Uplink sent <-- Rx Delay start
[05/04/24 - 20:06:43.472] RLB_PRO: 
[05/04/24 - 20:06:43.473] RLB_PRO: PHY: Frequency DL = 867.100 MHz
[05/04/24 - 20:06:43.476] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 20:06:48.465] RLB_PRO: Opening Rx1 window (45856 us timeout)... <-- Rx Delay end 
[05/04/24 - 20:06:48.515] RLB_PRO: Closing Rx1 window
[05/04/24 - 20:06:48.517] RLB_PRO: PHY: Frequency DL = 869.525 MHz
[05/04/24 - 20:06:48.522] RLB_PRO: PHY: SF = 9, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 20:06:49.464] RLB_PRO: Opening Rx2 window (123424 us timeout)... <-- Rx Delay end 
[05/04/24 - 20:06:49.593] RLB_PRO: Closing Rx2 window
[05/04/24 - 20:06:49.595] [LoRaWAN] Next uplink in 180s
[05/04/24 - 20:06:49.598] 
[05/04/24 - 20:09:49.665] RLB_PRO: 
[05/04/24 - 20:09:49.666] RLB_PRO: PHY: Frequency UL = 867.100 MHz
[05/04/24 - 20:09:49.672] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 20:09:49.676] RLB_PRO: Uplink (FcntUp = 505) decoded:
[05/04/24 - 20:09:49.681] RLB_PRO: 0000000 c0 00 78 56 88 3e ce 3f 13 00 00 00 34 12 ba ab | ..xV.>.?....4...
[05/04/24 - 20:09:49.686] RLB_PRO: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[05/04/24 - 20:09:49.695] 
[05/04/24 - 20:09:49.695] 
[05/04/24 - 20:09:49.695] 
[05/04/24 - 20:09:49.696] Core  1 register dump:
[05/04/24 - 20:09:49.697] 
[05/04/24 - 20:09:49.697] PC      : 0x400556b4  PS      : 0x00060c30  A0      : 0x8201f9f8  A1      : 0x3fce29f0  
[05/04/24 - 20:09:49.709] 
[05/04/24 - 20:09:49.709] A2      : 0xa70fe3f9  A3      : 0xa70fe3f5  A4      : 0x000000ff  A5      : 0x0000ff00  
[05/04/24 - 20:09:49.713] 
[05/04/24 - 20:09:49.713] A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x6b6e5500  A9      : 0x3fce2d30  
[05/04/24 - 20:09:49.721] 
[05/04/24 - 20:09:49.721] A10     : 0x00060c23  A11     : 0x00000000  A12     : 0x00060c20  A13     : 0x00000000  
[05/04/24 - 20:09:49.729] 
[05/04/24 - 20:09:49.729] A14     : 0x02ce338c  A15     : 0x00ffffff  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
[05/04/24 - 20:09:49.738] 
[05/04/24 - 20:09:49.738] EXCVADDR: 0xa70fe3f9  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
[05/04/24 - 20:09:49.744] 
[05/04/24 - 20:09:49.744] 
[05/04/24 - 20:09:49.744] 
[05/04/24 - 20:09:49.744] 
[05/04/24 - 20:09:49.744] 
[05/04/24 - 20:09:49.745] Backtrace:0x400556b1:0x3fce29f0 |<-CORRUPTED
[05/04/24 - 20:09:49.748] 
[05/04/24 - 20:09:49.749] 
[05/04/24 - 20:09:49.749] 
[05/04/24 - 20:09:49.749] 
[05/04/24 - 20:09:49.749] 
[05/04/24 - 20:09:49.749] 
[05/04/24 - 20:09:49.749] 
[05/04/24 - 20:09:49.749] 
[... 394 lines cut ...]
[05/04/24 - 21:20:08.469] [LoRaWAN] SNR:		13.50 dB
[05/04/24 - 21:20:08.471] 
[05/04/24 - 21:20:08.471] [LoRaWAN] Frequency error:	1902.63 Hz
[05/04/24 - 21:20:08.479] 
[05/04/24 - 21:20:08.479] [LoRaWAN] Event information:
[05/04/24 - 21:20:08.480] 
[05/04/24 - 21:20:08.480] [LoRaWAN] Confirmed:	0
[05/04/24 - 21:20:08.481] 
[05/04/24 - 21:20:08.481] [LoRaWAN] Confirming:	1
[05/04/24 - 21:20:08.481] 
[05/04/24 - 21:20:08.481] [LoRaWAN] Datarate:	2
[05/04/24 - 21:20:08.483] 
[05/04/24 - 21:20:08.483] [LoRaWAN] Frequency:	867.900 MHz
[05/04/24 - 21:20:08.486] 
[05/04/24 - 21:20:08.486] [LoRaWAN] Output power:	14 dBm
[05/04/24 - 21:20:08.490] 
[05/04/24 - 21:20:08.490] [LoRaWAN] Frame count:	0
[05/04/24 - 21:20:08.491] 
[05/04/24 - 21:20:08.491] [LoRaWAN] Port:		0
[05/04/24 - 21:20:08.493] 
[05/04/24 - 21:20:08.493] [LoRaWAN] Next uplink in 294s
[05/04/24 - 21:20:08.495] 
[05/04/24 - 21:25:03.348] RLB_PRO: 

```

StevenCellist, 2024-04-06:
@michapr I'm not sure what the issue could be; the crash happens during a hexdump of the uplink message which does not appear related to something particular in the LW stack. However it could be the case that a memory leak occurs somewhere. That is however difficult to investigate and definitely no priority for me. You could help by printing the free memory before and after each up/downlink / sendReceive to see if anything strikes the eye, which could help. I will only have time in two or three weeks to investigate further. Lots of other deadlines upcoming...

michapr, 2024-04-07:
>Lots of other deadlines upcoming...

Yes, of course. This has no priority, I think so.
Will check the memory, thanks for the hint.

HeadBoffin, 2024-04-09:
> > the debug log leading up to the crash is far more important than after as after the crash it is a clean slate.
> yes, that's why I have added 2 cycles before reboot, I thought it is ok. 

To me it looks like a partial log starting at 20:06:43.377 and a crash at 20:09:49.686 followed by a pile of uplinks. Is there another one to see?

michapr, 2024-04-09:
>To me it looks like a partial log starting at 20:06:43.377 and a crash at 20:09:49.686 followed by a pile of uplinks. Is there another one to see?

Right, was only ONE full cycle before, sorry.

Can give you more lines before, but I'm not sure, will you see more here...

```
[05/04/24 - 19:48:05.657] RLB_PRO: 
[05/04/24 - 19:48:05.658] RLB_PRO: PHY: Frequency UL = 868.100 MHz
[05/04/24 - 19:48:05.664] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:48:05.669] RLB_PRO: Uplink (FcntUp = 498) decoded:
[05/04/24 - 19:48:05.672] RLB_PRO: 0000000 c0 00 78 56 88 3e ce 3f 13 00 00 00 34 12 ba ab | ..xV.>.?....4...
[05/04/24 - 19:48:05.680] RLB_PRO: 0000010 40 29 f1 0b 26 80 f2 01 0a bc 5a c3 6f af 00 00 | @)..&.....Z.o...
[05/04/24 - 19:48:05.690] RLB_PRO: 0000020 14                                              | .                 
[05/04/24 - 19:48:05.694] RLB_DBG: Timeout in 257 ms
[05/04/24 - 19:48:05.747] RLB_PRO: Uplink sent <-- Rx Delay start
[05/04/24 - 19:48:05.751] RLB_PRO: 
[05/04/24 - 19:48:05.752] RLB_PRO: PHY: Frequency DL = 868.100 MHz
[05/04/24 - 19:48:05.755] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:48:10.744] RLB_PRO: Opening Rx1 window (45856 us timeout)... <-- Rx Delay end 
[05/04/24 - 19:48:10.797] RLB_PRO: Closing Rx1 window
[05/04/24 - 19:48:10.800] RLB_PRO: PHY: Frequency DL = 869.525 MHz
[05/04/24 - 19:48:10.801] RLB_PRO: PHY: SF = 9, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:48:11.744] RLB_PRO: Opening Rx2 window (123424 us timeout)... <-- Rx Delay end 
[05/04/24 - 19:48:11.871] RLB_PRO: Closing Rx2 window
[05/04/24 - 19:48:11.875] [LoRaWAN] Next uplink in 180s
[05/04/24 - 19:48:11.877] 
[05/04/24 - 19:51:11.944] RLB_PRO: 
[05/04/24 - 19:51:11.946] RLB_PRO: PHY: Frequency UL = 868.100 MHz
[05/04/24 - 19:51:11.951] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:51:11.956] RLB_PRO: Uplink (FcntUp = 499) decoded:
[05/04/24 - 19:51:11.960] RLB_PRO: 0000000 c0 00 78 56 88 3e ce 3f 13 00 00 00 34 12 ba ab | ..xV.>.?....4...
[05/04/24 - 19:51:11.972] RLB_PRO: 0000010 40 29 f1 0b 26 80 f3 01 0a 6c 70 6c ce af 00 00 | @)..&....lpl....
[05/04/24 - 19:51:11.975] RLB_PRO: 0000020 14                                              | .                 
[05/04/24 - 19:51:11.981] RLB_DBG: Timeout in 257 ms
[05/04/24 - 19:51:12.033] RLB_PRO: Uplink sent <-- Rx Delay start
[05/04/24 - 19:51:12.039] RLB_PRO: 
[05/04/24 - 19:51:12.039] RLB_PRO: PHY: Frequency DL = 868.100 MHz
[05/04/24 - 19:51:12.042] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:51:17.031] RLB_PRO: Opening Rx1 window (45856 us timeout)... <-- Rx Delay end 
[05/04/24 - 19:51:17.081] RLB_PRO: Closing Rx1 window
[05/04/24 - 19:51:17.083] RLB_PRO: PHY: Frequency DL = 869.525 MHz
[05/04/24 - 19:51:17.088] RLB_PRO: PHY: SF = 9, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:51:18.031] RLB_PRO: Opening Rx2 window (123424 us timeout)... <-- Rx Delay end 
[05/04/24 - 19:51:18.159] RLB_PRO: Closing Rx2 window
[05/04/24 - 19:51:18.162] [LoRaWAN] Next uplink in 180s
[05/04/24 - 19:51:18.164] 
[05/04/24 - 19:54:18.231] RLB_PRO: 
[05/04/24 - 19:54:18.233] RLB_PRO: PHY: Frequency UL = 868.500 MHz
[05/04/24 - 19:54:18.238] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:54:18.242] RLB_PRO: Uplink (FcntUp = 500) decoded:
[05/04/24 - 19:54:18.246] RLB_PRO: 0000000 c0 00 78 56 88 3e ce 3f 13 00 00 00 34 12 ba ab | ..xV.>.?....4...
[05/04/24 - 19:54:18.253] RLB_PRO: 0000010 40 29 f1 0b 26 80 f4 01 0a b9 6e 51 08 af 00 00 | @)..&.....nQ....
[05/04/24 - 19:54:18.261] RLB_PRO: 0000020 14                                              | .                 
[05/04/24 - 19:54:18.268] RLB_DBG: Timeout in 257 ms
[05/04/24 - 19:54:18.320] RLB_PRO: Uplink sent <-- Rx Delay start
[05/04/24 - 19:54:18.324] RLB_PRO: 
[05/04/24 - 19:54:18.325] RLB_PRO: PHY: Frequency DL = 868.500 MHz
[05/04/24 - 19:54:18.329] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:54:23.318] RLB_PRO: Opening Rx1 window (45856 us timeout)... <-- Rx Delay end 
[05/04/24 - 19:54:23.368] RLB_PRO: Closing Rx1 window
[05/04/24 - 19:54:23.374] RLB_PRO: PHY: Frequency DL = 869.525 MHz
[05/04/24 - 19:54:23.375] RLB_PRO: PHY: SF = 9, BW = 125.000 kHz, CR = 4/5
[05/04/24 - 19:54:24.318] RLB_PRO: Opening Rx2 window (123424 us timeout)... <-- Rx Delay end 
[05/04/24 - 19:54:24.446] RLB_PRO: Closing Rx2 window
[05/04/24 - 19:54:24.448] [LoRaWAN] Next uplink in 180s
[05/04/24 - 19:54:24.451] 
[05/04/24 - 19:57:24.517] RLB_PRO: 
[05/04/24 - 19:57:24.519] RLB_PRO: PHY: Frequency UL = 868.300 MHz
[05/04/24 - 19:57:24.524] RLB_PRO: PHY: SF = 7, BW = 125.000 kHz, CR = 4/5
[... 143 lines cut ...]
[05/04/24 - 20:09:49.800] 
[05/04/24 - 20:09:49.800] Attempting to boot anyway...
[05/04/24 - 20:09:49.800] 
[05/04/24 - 20:09:49.800] entry 0x403c98d8
[05/04/24 - 20:09:49.801] 
[05/04/24 - 20:09:54.887] 
[05/04/24 - 20:09:54.887] Setup
[05/04/24 - 20:09:54.888] 
[05/04/24 - 20:09:54.888] Initalise the radio
[05/04/24 - 20:09:54.889] 
[05/04/24 - 20:09:54.889] RLB_DBG: RadioLib Debug Info
[05/04/24 - 20:09:54.891] RLB_DBG: Version:  6.5.0.0
[05/04/24 - 20:09:54.894] RLB_DBG: Platform: ESP32
[05/04/24 - 20:09:54.897] RLB_DBG: Compiled: Apr  3 2024 17:35:11
[05/04/24 - 20:09:54.900] 
[05/04/24 - 20:09:54.911] RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
[05/04/24 - 20:09:54.916] RLB_DBG: 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
[05/04/24 - 20:09:54.923] RLB_DBG: 
[05/04/24 - 20:09:54.924] RLB_DBG: M	SX126x
[05/04/24 - 20:09:54.960] Join ('login') to the LoRaWAN Network
...
```

BTW: my FcntUp is now at 965 (without any crash) and memory have still the same values...
WSL stick is powered by LiPo and USB, so don't think about a power issue - so have no idea...