jgromes/RadioLib #1056: TTGO LoRa32 (T3S3 v1.2) Error -16

ThePioS, 2024-04-05:
I have been trying to get the LoRaWAN_Starter example to work. The board i am using is a LILYGO T3S3 V1.2 ESP32-S3 LoRa SX1276 Development Board.

To run the code I mapped the pins of my module to the config.h file and added the JoinEUI, DevEUI, AppKey and nwkKey

```

SX1276 radio = new Module(7, 9, 8, 33);

```

In LoRaWAN_Starter.ino I added the following line to void setup:

```

SPI.begin(SCLK_PIN, MISO_PIN, MOSI_PIN);

```

After running the code my device connects to TTN and gets a join accept message but when trying to perform the payload uplink I get the  error: Error in sendReceive(-16)

This is the debug output:

```

00:34:12.553 -> [Radio] Initializing ... 
00:34:12.553 -> Setup ... 
00:34:12.553 -> Initalise the radio
00:34:12.553 -> RLB_DBG: RadioLib Debug Info
00:34:12.553 -> RLB_DBG: Version:  6.5.0.0
00:34:12.553 -> RLB_DBG: Platform: ESP32
00:34:12.553 -> RLB_DBG: Compiled: Apr  5 2024 23:42:39
00:34:12.553 -> 
00:34:12.553 -> RLB_SPI: R	42	12	
00:34:12.553 -> RLB_DBG: M	SX127x
00:34:12.553 -> RLB_SPI: R	1	F	
00:34:12.553 -> RLB_SPI: R	1	9	
00:34:12.553 -> RLB_SPI: W	1	9	
00:34:12.553 -> RLB_SPI: R	1	9	
00:34:12.553 -> RLB_SPI: R	24	5	
00:34:12.553 -> RLB_SPI: W	24	0	

/******************************************/

00:34:12.586 -> RLB_SPI: R	1	89	
00:34:12.586 -> RLB_SPI: R	1E	90	
00:34:12.586 -> RLB_SPI: W	1E	94	
00:34:12.586 -> RLB_SPI: R	1E	94	
00:34:12.586 -> Join ('login') to the LoRaWAN Network
00:34:12.586 -> RLB_PRO: Configuration mismatch (checksum: 0, mode: 0, class: 0, plan: 1)
00:34:12.586 -> RLB_PRO: Nonces buffer:
00:34:12.586 -> RLB_PRO: 0000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
00:34:12.586 -> RLB_PRO: Clearing buffer and starting fresh
00:34:12.586 -> RLB_PRO: Setting up dynamic channels
00:34:12.586 -> RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
00:34:12.586 -> RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
00:34:12.586 -> RLB_PRO: UL:   2 1 868.500 (0 - 5) | DL:   2 1 868.500 (0 - 5)
00:34:12.586 -> RLB_PRO: [MAC] 0x03
00:34:12.586 -> RLB_PRO: 0000000 20 00 00 80                                     |  ...              
00:34:12.586 -> RLB_PRO: LinkADRReq: dataRate = 2, txPower = 0, chMask = 0x0000, chMaskCntl = 0, nbTrans = 0
00:34:12.586 -> RLB_PRO: PHY: SF = 10, BW = 125.000 kHz, CR = 4/5
00:34:12.586 -> RLB_SPI: R	1	89	
00:34:12.586 -> RLB_PRO: PHY: TX = 16 dBm
00:34:12.586 -> RLB_SPI: R	1	89	
00:34:12.586 -> RLB_SPI: R	1	89	
00:34:12.586 -> RLB_SPI: W	1	89	
00:34:12.586 -> RLB_SPI: R	1	89	
00:34:12.586 -> RLB_SPI: R	9	F8	
00:34:12.586 -> RLB_SPI: W	9	F8	
00:34:12.586 -> RLB_SPI: R	9	F8	
00:34:12.586 -> RLB_SPI: R	9	F8	
[... 452 lines cut ...]
00:34:18.577 -> RLB_SPI: mask:		0xFF
00:34:18.577 -> RLB_SPI: new:		0x6
00:34:18.577 -> RLB_SPI: read:		0x0
00:34:18.577 -> RLB_SPI: R	8	0	
00:34:18.577 -> RLB_SPI: W	8	66	
00:34:18.577 -> RLB_SPI: R	8	0	

/******************************************/

00:34:18.577 -> RLB_SPI: R	8	0	
00:34:18.577 -> RLB_SPI: 
00:34:18.577 -> RLB_SPI: address:	0x8
00:34:18.577 -> RLB_SPI: bits:		7 0
00:34:18.577 -> RLB_SPI: value:		0x66
00:34:18.577 -> RLB_SPI: current:	0x0
00:34:18.577 -> RLB_SPI: mask:		0xFF
00:34:18.577 -> RLB_SPI: new:		0x66
00:34:18.577 -> RLB_SPI: read:		0x0
00:34:18.577 -> Error in sendReceive(-16)
00:34:18.577 -> Uplink complete, next in 300 seconds

```

I am new to all of this so i am unsure what exactly is the issue.
If any additional information is needed tell me.

jgromes, 2024-04-06:
Could you provide the complete log? Seems like parts of it are missing. At some point, the radio completely stops responding (as it only returns 0s on the SPI bus) but because you seem to have removed parts of the log, it's impossible to tell where exactly that happens.

Are you sure you have the SX1276 and not the SX1278? The failure seems to be happen somewhere around the frequency configuration.

ThePioS, 2024-04-06:
The seller provided information states that i have the SX1276, since I want to send information over the 868MHz spectrum I doubt the SX1278 will be useful.
I have no way of actually verifying the information but on the module itself is written 868/915MHz.

Here is the full log:
```
14:11:39.746 -> [Radio] Initializing ... 
14:11:39.746 -> Setup ... 
14:11:39.746 -> Initalise the radio
14:11:39.785 -> RLB_DBG: RadioLib Debug Info
14:11:39.785 -> RLB_DBG: Version:  6.5.0.0
14:11:39.785 -> RLB_DBG: Platform: ESP32
14:11:39.785 -> RLB_DBG: Compiled: Apr  5 2024 23:42:39
14:11:39.785 -> 
14:11:39.785 -> RLB_SPI: R	42	12	
14:11:39.785 -> RLB_DBG: M	SX127x
14:11:39.785 -> RLB_SPI: R	1	F	
14:11:39.785 -> RLB_SPI: R	1	9	
14:11:39.785 -> RLB_SPI: W	1	9	
14:11:39.785 -> RLB_SPI: R	1	9	
14:11:39.785 -> RLB_SPI: R	24	5	
14:11:39.785 -> RLB_SPI: W	24	0	
14:11:39.785 -> RLB_SPI: R	24	0	
14:11:39.785 -> RLB_SPI: R	1	9	
14:11:39.785 -> RLB_SPI: R	1	9	
14:11:39.785 -> RLB_SPI: R	1	9	
14:11:39.785 -> RLB_SPI: W	1	8	
14:11:39.785 -> RLB_SPI: R	1	8	
14:11:39.785 -> RLB_SPI: R	1	8	
14:11:39.785 -> RLB_SPI: W	1	88	
14:11:39.785 -> RLB_SPI: R	1	88	
14:11:39.785 -> RLB_SPI: R	1	88	
14:11:39.785 -> RLB_SPI: R	1	88	
14:11:39.785 -> RLB_SPI: W	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: W	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	39	12	
14:11:39.785 -> RLB_SPI: W	39	12	
14:11:39.785 -> RLB_SPI: R	39	12	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: W	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	B	2B	
14:11:39.785 -> RLB_SPI: W	B	23	
14:11:39.785 -> RLB_SPI: R	B	23	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: W	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	20	0	
14:11:39.785 -> RLB_SPI: W	20	0	
14:11:39.785 -> RLB_SPI: R	20	0	
14:11:39.785 -> RLB_SPI: R	21	8	
14:11:39.785 -> RLB_SPI: W	21	8	
14:11:39.785 -> RLB_SPI: R	21	8	
14:11:39.785 -> RLB_SPI: R	1	89	
14:11:39.785 -> RLB_SPI: R	33	27	
14:11:39.785 -> RLB_SPI: W	33	27	
14:11:39.785 -> RLB_SPI: R	33	27	
14:11:39.785 -> RLB_SPI: R	33	27	
14:11:39.785 -> RLB_SPI: W	33	27	
14:11:39.785 -> RLB_SPI: R	33	27	
14:11:39.785 -> RLB_SPI: R	3B	1D	
14:11:39.785 -> RLB_SPI: W	3B	1D	
14:11:39.785 -> RLB_SPI: R	3B	1D	
[... 1371 lines cut ...]
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: R	8	0	
14:11:45.801 -> RLB_SPI: 
14:11:45.801 -> RLB_SPI: address:	0x8
14:11:45.801 -> RLB_SPI: bits:		7 0
14:11:45.801 -> RLB_SPI: value:		0x66
14:11:45.801 -> RLB_SPI: current:	0x0
14:11:45.801 -> RLB_SPI: mask:		0xFF
14:11:45.801 -> RLB_SPI: new:		0x66
14:11:45.801 -> RLB_SPI: read:		0x0
14:11:45.801 -> Error in sendReceive(-16)
14:11:45.801 -> Uplink complete, next in 300  seconds
```

jgromes, 2024-04-06:
@ThePioS

This is the exact point where the radio stops responding:

```
14:11:45.694 -> RLB_SPI: W	7	13	
14:11:45.694 -> RLB_SPI: R	7	13	
14:11:45.694 -> RLB_SPI: R	8	9A	
14:11:45.694 -> RLB_SPI: W	8	33	
14:11:45.694 -> RLB_SPI: R	8	33	
14:11:45.694 -> RLB_PRO: NewChannelReq:
14:11:45.694 -> RLB_PRO: UL:   7 1 867.900 (0 - 5) | DL:   7 1 867.900 (0 - 5)
14:11:45.694 -> RLB_PRO: NewChannelAns: status = 0x03
14:11:45.694 -> Ready!
14:11:45.694 -> 
14:11:45.694 -> Sending uplink
14:11:45.694 -> RLB_SPI: R	1	0	
14:11:45.694 -> RLB_SPI: R	1	0	
14:11:45.694 -> RLB_SPI: R	1	0	
14:11:45.694 -> RLB_SPI: W	1	5	
14:11:45.694 -> RLB_SPI: R	1	0	
```

Before the string "Sending uplink", everything is fine. After that, the radio is always returnign zero. This is what that section looks like in the example - I'm assuming you have not edited it, since you have not posted your code.

```c++
  Serial.println(F("Sending uplink"));

  // Read some inputs
  uint8_t Digital1 = digitalRead(2);
  uint16_t Analog1 = analogRead(3);

  // Build payload byte array
```

Right after that string, there are some digital/analog read. I'm guessing that for some reason this breaks the  SPI communication to the radio. Can you try to remove those `digitalRead`/`analogRead` calls and see if the problem persists?

ThePioS, 2024-04-06:
Thank you @jgromes once I removed the `digitalRead`/ `analogRead` calls the code runs with no errors

This is the new output:

```
15:28:31.044 -> RLB_SPI: R	13	12	
15:28:31.044 -> RLB_SPI: R	12	50	
15:28:31.044 -> RLB_SPI: R	1C	0	
15:28:31.044 -> RLB_SPI: R	0	60	3	1	8	26	86	1	0	3	51	FF	0	1	6	7D	92	14	19	
15:28:31.044 -> RLB_SPI: R	1	89	
15:28:31.044 -> RLB_SPI: W	12	FF	
15:28:31.044 -> RLB_PRO: Downlink (AFcntDown = 1) encoded:
15:28:31.044 -> RLB_PRO: 0000000 49 00 00 00 00 01 03 01 08 26 01 00 00 00 00 0e | I........&......
15:28:31.044 -> RLB_PRO: 0000010 60 03 01 08 26 86 01 00 03 51 ff 00 01 06 7d 92 | `...&....Q....}.
15:28:31.044 -> RLB_PRO: 0000020 14 19                                           | ..                
15:28:31.044 -> RLB_PRO: [MAC] 0x2E
15:28:31.044 -> RLB_PRO: [MAC] 0xEE
15:28:31.044 -> RLB_PRO: [MAC] 0x63
15:28:31.044 -> RLB_PRO: [MAC] 0x90
15:28:31.044 -> RLB_PRO: [MAC] 0x6C
15:28:31.044 -> RLB_PRO: [MAC] 0x56
15:28:31.044 -> Uplink complete, next in 300  seconds

```

jgromes, 2024-04-06:
@ThePioS thanks for checking it, then I guess we will either have to remove the reading pins from the examples or make sure that they don't clash with SPI. I'm more in favor of the former, replacing them with a random number, since the latter could be prone to breaking in the future again.

StevenCellist, 2024-04-06:
Pin 3 is used in LoRa SPI on the T3S3, yup..

HeadBoffin, 2024-04-06:
Meh, on it.

HeadBoffin, 2024-04-08:
Not really a bug and a situation anyone could create, particularly with the automagic pinmaps. And the template info is still MIA ...

I can find some pins not being used for anything else on the automagic pinmaps but we should perhaps have an item on this in the wiki!

jgromes, 2024-04-09:
@HeadBoffin 

> I can find some pins not being used for anything else on the automagic pinmaps but we should perhaps have an item on this in the wiki!

I don't think we will be able to keep that up for very long, as new pinmaps can be added (see the currently open #1049). With that in mind I pushed the change from analog/digital read to random number, let's hope nobody else trips up on this until the next update.

closed 2024-04-09 by jgromes