jgromes/RadioLib #1073: Move Raspberry Pi example to another GPIO library

Elizafox, 2024-04-23:
So the Raspberry Pi example uses pigpio, which hasn't had a commit in 3 years and doesn't work on the Pi 5 or later Pi's.

The author's recommended replacement for such boards is lgpio, by the same author. It's also all-around more modern, supporting /dev/gpiomem and therefore not requiring root privileges.

I've tried my hands at a [preliminary port](/Elizafox/RadioLib). Note this wouldn't be suitable for upstreaming without changes; this is designed for my configuration. But so far it fails with:

```
RLB_DBG:
RadioLib Info
Version:  "6.5.0.0"
Platform: "Generic"
Compiled: "Apr 23 2024" "02:40:57"
RLB_SPI: R	42	12
RLB_DBG: M	SX127x
RLB_SPI: R	1	9
RLB_SPI: R	1	9
RLB_SPI: W	1	9
RLB_SPI:
RLB_SPI: address:	0x1
RLB_SPI: bits:		2 0
RLB_SPI: value:		0x1
RLB_SPI: current:	0x9
RLB_SPI: mask:		0x7
RLB_SPI: new:		0x9
RLB_SPI: read:		0x0
[SX1261] Initializing ... failed, code -16
```

jgromes, 2024-04-23:
Thanks for pointing this out! So far I've been using pigpio without problems, though the need to run it as root has been a pain, so if we can switch to something more modern that would be great.

Regarding the error you are seeing, it seems that RadioLib is having trouble reading out the value it has just written into the SPI register, so the first place I would be looking at is the SPI configuration (number of bits, clock polarity/phase etc).

Elizafox, 2024-04-23:
Clock polarity and phase should be the same. lgpio uses 8 bits. I'm really at a loss here.

jgromes, 2024-04-24:
I tried your lgpio HAL with RPi 3B and SX1276 and I'm getting back pretty much the same result. I did do one change, in that I used GPIO 4 as the chip select instead of GPIO8/CE0 (since that was always returning "GPIO busy" error).

```
RLB_SPI: R    42      12
RLB_SPI: R      1       9
RLB_SPI: R      1       9
RLB_SPI: W      1       9
RLB_SPI: 
RLB_SPI: address:       0x1
RLB_SPI: bits:          2 0
RLB_SPI: value:         0x1
RLB_SPI: current:       0x9
RLB_SPI: mask:          0x7
RLB_SPI: new:           0x9
RLB_SPI: read:          0x0
failed, code -16
```

I also tried running this with SX1261 (because it has different SPI interface), and got back something slightly different:

```
Initializing ... RLB_DBG: 
RadioLib Info
Version:  "6.5.0.0"
Platform: "Generic"
Compiled: "Apr 24 2024" "05:50:22"
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     AA      AA
RLB_SPI: CMDR   1D      3       20
RLB_SPI: SI                             0       0       0       0       0       0       0       0       0       0       0       0       0       0       000
RLB_SPI: SO     A2      A2      A2      A2      53      58      31      32      36      31      20      56      32      44      20      32      44      3032      0
RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
RLB_DBG: 0000320 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00 | SX1261 V2D 2D02.
RLB_DBG: 
RLB_DBG: M      SX126x
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDW   80
RLB_SPI: SI             0
RLB_SPI: SO     A2      A2
RLB_SPI: CMDR   17
RLB_SPI: SI             0       0       0
RLB_SPI: SO     A2      A2      0       20
RLB_SPI: CMDW   7
RLB_SPI: SI             0       0
RLB_SPI: SO     A2      A2      A2
RLB_SPI: CMDW   97
RLB_SPI: SI             0       0       1       40
RLB_SPI: SO     A2      A2      A2      A2      A2
RLB_SPI: CMDW   8F
RLB_SPI: SI             0       0
RLB_SPI: SO     A2      A2      A2
RLB_SPI: CMDW   8A
RLB_SPI: SI             1
RLB_SPI: SO     A2      A2
RLB_SPI: CMDW   93
RLB_SPI: SI             20
RLB_SPI: SO     A2      A2
RLB_SPI: CMDW   88
RLB_SPI: SI             3       16      A       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      A2      A2      A2      A2
RLB_SPI: CMDW   2
RLB_SPI: SI             43      FF
RLB_SPI: SO     A2      A2      A2
RLB_SPI: CMDW   8
RLB_SPI: SI             0       0       0       0       0       0       0       0
RLB_SPI: SO     A2      A2      A2      A2      A2      A2      A2      A2      A2
RLB_DBG: GPIO post-transfer timeout, is it connected?
failed, code -705
```

Looks like the SPI can interface with the radio fine. Suspiciously enough, there seems to be no actual timeout, which should be one second, but the program returns immediately. This could also be the cause for the -16, since on SX127x, modification of the operation mode register (which is the thing that is failing) takes a couple ms to propagate and RadioLib waits for that.

After that, I realized that the HAL uses `unsigned long` for its `millis`/`micros` methods, while internally `uint32_t` is mostly used, which is an oversight on my part. Doing a dirty cast to to `uint32_t` resolves the -16 error. There might be some other timing problems with that though, so maybe a better workaround would be to subtract part of the Unix timestamp (E.g. Januray 1st 2020) in the HAL so that it's not overflowing the 32-bit numbers. 

```c++
    unsigned long millis() override {
      uint32_t time = lguTimestamp() / 1000000UL;
      return time;
    }

    unsigned long micros() override {
      uint32_t time = lguTimestamp() / 1000UL;
      return time;
    }
```

Elizafox, 2024-04-24:
Huh! I swore that was fine... but that makes sense! I will give it a spin locally and report back.

Elizafox, 2024-04-24:
That worked! Wow, thanks, I had been banging my head against this for days. This was such a non-obvious error.

However, it's probably better to move the code that interacts with millis/micros to unsigned long. That would probably be more ideal.

jgromes, 2024-04-24:
> I'm also giving out a try porting it to gpiod and spidev directly

That would be extremely interesting!

> it's probably better to move the code that interacts with millis/micros to unsigned long

Definitely, that's the actual fix on RadioLib side. I can push that in the next couple of days (or you know - PRs are welcome).

Elizafox, 2024-04-24:
> That would be extremely interesting!
>
>
> Definitely, that's the actual fix on RadioLib side. I can push that in the next couple of days (or you know - PRs are welcome ;) ).

This is done in #1075.

Elizafox, 2024-04-25:
I think lgpio might be the way to go here. Event management with gpiod is a pain in this mode... we'd have to spawn a thread, lock/unlock data structures, etc... I don't think it's worth the trouble, especially given lgpio works just fine and also uses spidev (and the kernel GPIO interface directly). Once #1075 lands I will get the HAL into shape for merging.

jgromes, 2024-06-01:
@Elizafox I've been using your lgpio HAL in my development for a while now and it seems to be working just fine (with a minor addition of `tone`/`noTone` implementation) - if you could open up a PR we could merge this. Thanks!

jgromes, 2024-06-14:
@Elizafox I've changed the example and the CI to use lgpio in bc36c1e98afdc5f05402d129a2dc54184b7db453 - it's been working great, so I don't think there's any reason not to do this change. Thanks for pointing me in this direction!

closed 2024-06-14 by jgromes