jgromes/RadioLib #1084: importing Radiolib into VSCODE + Platformio with NRF52840 board does not compile

richonguzman, 2024-05-02:
hi !!!

(I'm trying to code the TTGO-T-ECHO board = NRF52840 + SX1262 board)

so In VSCODE + Platformio I add to "platformio.ini" the following:

`[platformio]
default_envs = t_echo

[env]
platform = nordicnrf52
framework = arduino
monitor_speed = 115200
upload_protocol = nrfutil

	
[env:t_echo]
board = nrf52840_dk_adafruit
build_flags = -Werror -Wall -DT_ECHO
lib_deps =
	mikalhart/TinyGPSPlus @ 1.0.3
	jgromes/RadioLib @ 6.1.0`

and as soon as Radiolib is imported , when trying to compile the following error appears:

`.pio/libdeps/t_echo/RadioLib/src/Module.cpp: In member function 'void Module::SPIreadRegisterBurst(uint16_t, size_t, uint8_t*)':
.pio/libdeps/t_echo/RadioLib/src/Module.cpp:163:11: error: argument 2 null where non-null expected [-Werror=nonnull]
     memcpy(buffOutPtr, dataOut, numBytes);
     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/stdlib.h:11:0,
                 from /Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/c++/7.2.1/cstdlib:75,
                 from /Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/c++/7.2.1/stdlib.h:36,
                 from /Users/ricardoguzman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Arduino.h:6,
                 from .pio/libdeps/t_echo/RadioLib/src/BuildOpt.h:6,
                 from .pio/libdeps/t_echo/RadioLib/src/TypeDef.h:6,
                 from .pio/libdeps/t_echo/RadioLib/src/Module.h:4,
                 from .pio/libdeps/t_echo/RadioLib/src/Module.cpp:1:
/Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/string.h:31:8: note: in a call to function 'void* memcpy(void*, const void*, size_t)' declared here
 _PTR   _EXFUN(memcpy,(_PTR __restrict, const _PTR __restrict, size_t));
        ^
.pio/libdeps/t_echo/RadioLib/src/Module.cpp: In member function 'void Module::SPIwriteRegisterBurst(uint16_t, uint8_t*, size_t)':
.pio/libdeps/t_echo/RadioLib/src/Module.cpp:177:11: error: argument 1 null where non-null expected [-Werror=nonnull]
     memcpy(dataIn, &buffIn[this->SPIaddrWidth/8], numBytes);
     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/stdlib.h:11:0,
                 from /Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/c++/7.2.1/cstdlib:75,
                 from /Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/c++/7.2.1/stdlib.h:36,
                 from /Users/ricardoguzman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Arduino.h:6,
                 from .pio/libdeps/t_echo/RadioLib/src/BuildOpt.h:6,
                 from .pio/libdeps/t_echo/RadioLib/src/TypeDef.h:6,
                 from .pio/libdeps/t_echo/RadioLib/src/Module.h:4,
                 from .pio/libdeps/t_echo/RadioLib/src/Module.cpp:1:
/Users/ricardoguzman/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/arm-none-eabi/include/string.h:31:8: note: in a call to function 'void* memcpy(void*, const void*, size_t)' declared here
 _PTR   _EXFUN(memcpy,(_PTR __restrict, const _PTR __restrict, size_t));`
 
 I guess it has something to do with NRF52840 module, but the error appears just as importing the library and It has nothing to do with the coding in VSCODE

jgromes, 2024-05-02:
You have an outdated version of RadioLib (6.1.0) which is nearly a year old at this point, so the first thing to try is update to the latest one (6.5.0). I wouldn't be surprised if the error goes away as that particular part of the library has been reworked. If it doesn't, please post an update.

richonguzman, 2024-05-02:
6.5.0 generates the same error (only changes line from "/RadioLib/src/Module.cpp:163:11" to "/RadioLib/src/Module.cpp:167:11"

I disabled all radiolib and initiations and all..... so when adding it as platformio library only it still produces the error

jgromes, 2024-05-02:
Can you also try with the current master?

> I disabled all radiolib and initiations and all..... so when adding it as platformio library only it still produces the error

It's a compilation error, removing initializations will have no effect on it.

HeadBoffin, 2024-05-02:
I have some ideas but having the issues template filled in, formatted, with an un-redacted compile log would help immensely. Otherwise my advanced guessing skills based on information supplied is that there is an assert warning earlier that will clearly state the problem.

richonguzman, 2024-05-02:
> I have some ideas but having the issues template filled in, formatted, with an un-redacted compile log would help immensely. Otherwise my advanced guessing skills based on information supplied is that there is an assert warning earlier that will clearly state the problem.

sorry, but how can I generate the log to post?

> Can you also try with the current master?
> 
> > I disabled all radiolib and initiations and all..... so when adding it as platformio library only it still produces the error
> 
> It's a compilation error, removing initializations will have no effect on it.

I replace :
jgromes/RadioLib @ 6.5.0
for:
https://github.com/jgromes/RadioLib.git

which downloads few more stuff and processed to compile and got this:

`.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp: In member function 'int16_t LoRaWANNode::processCFList(uint8_t*)':
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2393:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].enabled = true;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2394:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].idx   = chIndex;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2395:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].freq  = freq;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2396:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].drMin = minDr;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2397:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].drMax = maxDr;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2400:149: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_DOWNLINK][chIndex] = this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex];
                                                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2400:79: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_DOWNLINK][chIndex] = this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex];`

HeadBoffin, 2024-05-02:
>  but how can I generate the log to post?

It's just the compile output / log / messages - you've done it twice now but each time you've just provided the bits you think are important rather than all of it.

As I can compile for nRF using the Nordic toolchain just fine, we need to find out what the differences are for you.

richonguzman, 2024-05-02:
> It's just the compile output / log / messages - you've done it twice now but each time you've just provided the bits you think are important rather than all of it.
>
> As I can compile for nRF using [...]

should I paste the complete output of compilation from VSCODE as a full log?

HeadBoffin, 2024-05-02:
> should I paste the complete output of compilation from VSCODE as a full log?

As above: "each time you've just provided the bits you think are important rather than **all of it**"

So yes, the full log as it will allow use to find out what the differences are for you.

richonguzman, 2024-05-02:
Processing t_echo (board: nrf52840_dk_adafruit; platform: nordicnrf52; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52840_dk_adafruit.html
PLATFORM: Nordic nRF52 (10.4.0) > Nordic nRF52840-DK (Adafruit BSP)
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (jlink) On-board (jlink) External (blackmagic, cmsis-dap, stlink)
PACKAGES: 
 - framework-arduinoadafruitnrf52 @ 1.10600.0 (1.6.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TinyGPSPlus @ 1.0.3
|-- RadioLib @ 6.5.0+sha.7209690
|-- SPI @ 1.0
Building in release mode
Compiling .pio/build/t_echo/lib286/RadioLib/protocols/LoRaWAN/LoRaWAN.cpp.o
Compiling .pio/build/t_echo/FrameworkArduino/wiring_analog_nRF52.c.o
Compiling .pio/build/t_echo/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/t_echo/FrameworkArduino/wiring_private.c.o
Compiling .pio/build/t_echo/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/t_echo/libFrameworkArduino.a
Indexing .pio/build/t_echo/libFrameworkArduino.a
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp: In member function 'int16_t LoRaWANNode::processCFList(uint8_t*)':
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2393:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].enabled = true;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2394:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].idx   = chIndex;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2395:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].freq  = freq;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2396:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].drMin = minDr;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2397:75: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex].drMax = maxDr;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2400:149: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_DOWNLINK][chIndex] = this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex];
                                                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
.pio/libdeps/t_echo/RadioLib/src/protocols/LoRaWAN/LoRaWAN.cpp:2400:79: error: array subscript is above array bounds [-Werror=array-bounds]
       this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_DOWNLINK][chIndex] = this->availableChannels[RADIOLIB_LORAWAN_CHANNEL_DIR_UPLINK][chIndex];
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
*** [.pio/build/t_echo/lib286/RadioLib/protocols/LoRaWAN/LoRaWAN.cpp.o] Error 1
=========================================================== [FAILED] Took 3.86 seconds ===========================================================

 *  The terminal process "platformio 'run', '--environment', 't_echo'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

HeadBoffin, 2024-05-02:
The BSP's for SAMD vs nRF52 are rather odd. The nRF52 has a more recent CMSIS but a far older compiler. The SAMD (aka Adafruit Feather M0) compiles with no errors and yes, now I can see errors for the nRF52 with the same setup as you. But it will end up compiling if you remove -Werror. As I don't own a T-ECHO I can't test it, but you can if you remove the compile flag, you may get lucky.

richonguzman, 2024-05-02:
> The BSP's for SAMD vs nRF52 are rather odd. The nRF52 has a more recent CMSIS but a far older compiler. The SAMD (aka Adafruit Feather M0) compiles with no errors and yes, now I can see errors for [...]

I deleted -Werror -Wall and will test how it goes as it clearly compiles without any warning now :P

richonguzman, 2024-05-03:
so, maybe I'm just lucky or you are making magic code :P 

but with both library importing (without -Werror -Wall) : 
1) jgromes/RadioLib @ 6.5.0
or
2) https://github.com/jgromes/RadioLib.git

although it still make some warnings they both make T-Echo with SX1262 and NRF52840 runs well 

thanks!!!

closed 2024-05-03 by richonguzman

HeadBoffin, 2024-05-03:
Thanks for the feedback that it runs OK - I've no magic code, I was getting 'odd' warnings with the nRF toolchain that matched what you were seeing but had no way to test.