jgromes/RadioLib #1085: SX1280 startTransmit after CAD not working

jonas-koeritz, 2024-05-03:
### What I want to achieve
I am trying to implement a listen before talk (LBT) scheme. 

### What I've tried
There is no example for interrupt driven CAD for the SX1280, I've tried to implement this myself using the `RADIOLIB_LOW_LEVEL` flag to manipulate state directly. After this didn't work as expected, I resorted to use `scanChannel()`. This returns correct results but a subsequent `startTransmit()` just returns 0 and doesn't send a packet.

I've followed the example code for blocking CAD and can successfully detect an occupied channel using `scanChannel()`. However, if I try to `startTransmit()` when I've detected that the channel is free. It returns state 0 (no error) but does not send the LoRa packet.

If I remove the CAD code from the same codebase, the packets get sent at the anticipated interval and can be received by a matchingly configured receiver.

### Guesses
Maybe something that `scanChannel()` does to the modules state results in it not being configured correctly for transmission afterwards. Receiving data works, transmission doesn't.

HeadBoffin, 2024-05-04:
By implication CAD is doing a receive - in this case listening across spreading factors. So the radio isn't really setup for a transmit on a particularly SF. Maybe set the SF you want to use before transmitting?

jgromes, 2024-05-05:
@jonas-koeritz The fact that SX128x lacks the interface for interrupt-driven CAD is an oversight on my part and will be rectified - thanks! However, I was unable to replicate the primary issue which is that `startTransmit` after CAD does not send a packet. Could you please fill in the infromation requested by the issue template, mainly the code you are running?

jonas-koeritz, 2024-05-05:
Thanks for getting back on the interrupt driven CAD stuff. I fixed my other issue in the meantime. It was caused by an integration error, while debugging I had to detach a component, which caused the main loop to run slowly (waiting for that component to respond). As a result, the interrupt flags where not checked/cleared correctly.

closed 2024-05-05 by jonas-koeritz

jgromes, 2024-05-05:
@jonas-koeritz glad I could help! Do let me know if you get a chance to try the interrupt-driven version of CAD (I only found one SX1280 device so I couldn't test it fully).