jgromes/RadioLib #1091: Move examples to a separate repository

caveman99, 2024-05-11:
Ever since the libtock example was included, our CI runs take a really long time and even return 'no space on device' with certain locally hosted runners. Turns out the submodule for libtock-c is massive and pulls in a truckload of other submodules. While the examples are really valuable they have grown to a size where it would be sensible to move them to their own github repository.

caveman99, 2024-05-11:
Hm.

`Submodule 'examples/lora/RadioLib' (https://github.com/jgromes/RadioLib.git) registered for path 'examples/NonArduino/Tock/libtock-c/examples/lora/RadioLib'`

Not a good idea... 

`Submodule 'lvgl/lvgl' (https://github.com/littlevgl/lvgl.git) registered for path 'examples/NonArduino/Tock/libtock-c/examples/lora/RadioLib/examples/NonArduino/Tock/libtock-c/lvgl/lvgl'`

Circular reference.

StevenCellist, 2024-05-11:
Honestly, I completely agree that Tock, LVGL etc are massive and make me get a headache every time I have to clone or fire up another environment or whatever... I do think that examples should be here because that is where people look, but just the source code. Cloning RadioLib (single branch) without its dependencies is plain 6MB in size, which is what this should be. Not getting all the extras along would be a great improvement.

caveman99, 2024-05-11:
Problem is, if you include it in Platformio as a GIT dependency you can't specify to not check out submodules or make a flat checkout.

HeadBoffin, 2024-05-11:
"Ever since the libtock example was included, our CI runs take a really long time and even return 'no space on device' with certain locally hosted runners" - who's CI are you referring to @caveman99?

Obviously @StevenCellist needs to sort out his Git foo ;-) and it very occasionally catches me out, but I can see how that will rack up the minutes on CI as well as space.

Perhaps move it to the RadioLib-Org.

jgromes, 2024-05-11:
> it would be sensible to move them to their own github repository

In general that is not an option, as a lot of users still use this repository as an Arduino library, which enforces a specific directory structure, including the location of examples. Platformio has inherited that, so moving all examples to some other repo would effectively remove examples from the POV of majority of the users. Using a submodule for the examples is also not possible (see https://github.com/arduino/library-registry/blob/main/FAQ.md#are-git-submodules-supported).

What we can do is either: 

a) Move the Tock example into a dedicated examples repository. It's not interesting to most users anyway.
b) Keep the Tock example here provided that it can be made less cumbersome to use (and resolves that circular dependency - thanks for pointing that out @caveman99).

cc @alistair23 (since you were the one who originally added the Tock example)

caveman99, 2024-05-11:
> "Ever since the libtock example was included, our CI runs take a really long time and even return 'no space on device' with certain locally hosted runners" - who's CI are you referring to @caveman99?

In this case it's Meshtastic, but i can assure you this will bite everyone trying to include Radiolib in a pio based ci job.

caveman99, 2024-05-11:
Heads up: A quick workaround is to include the ZIP download link for master instead of the git link.

HeadBoffin, 2024-05-11:
> but i can assure you this will bite everyone trying to include Radiolib in a pio based ci job.

I was not aware of that*. I was actually enquiring as to your interest in this as that speaks to the severity of the impact - in this case I can appreciate that it's quite high.

* ref Waynes World, schwing

alistair23, 2024-05-12:
> a) Move the Tock example into a dedicated examples repository. It's not interesting to most users anyway.
> b) Keep the Tock example here provided that it can be made less cumbersome to use (and resolves that circular dependency - thanks for pointing that out @caveman99).

What about removing the submodule and having the RadioLib build process just manually clone the repo?

Velocet, 2024-05-12:
Keep it simple: axe libtock or the example and think of a viable solution afterwards.

It doesn't even matter where the problem is addressed: even if the examples where in a separate repo you'd still have the problem.

I have the same issue bothering me as some of my build settings pull in all those dependencies.

---

*Short note about auto pin assignments for DIY builds: LoRa32 will define a pin standard for most used platforms based on already established assignments or one can define a common lib (`#define esphome`) from which the assignments will be used. Most libs already have pin definitions. Those will be copied over.*

closed 2024-05-13 by jgromes

jgromes, 2024-05-13:
@caveman99 #1098 was merged (thanks @alistair23!) removing the tock submodule, in 23dcc4b8b7e29344c3692c4bab8b5db579bdf4cb I also dropped the reference from gitmodules, so there are now no additional submodules required by RadioLib.

Let me know if that resolved the issue you were facing, thanks!

caveman99, 2024-05-13:
@jgromes @alistair23 thanks for the quick action. i updated the reference and the repository is building fine again!

Velocet, 2024-05-13:
Works for me! Thanks! 🙏🏻