jgromes/RadioLib #1124: Examples not running on seed studio XIAO esp32C3

patrick-kiwi, 2024-06-14:
Examples for SX127x not running on XIAO esp32-C3.  Compiles okay but there's no serial output.  Sanity check using an esp32-Devkit V1 and the code works fine.    

**Additional info:**
 - MCU: xiao esp32C3
 - Wireless module type SX1278
 - Arduino IDE version  1.8.19
 - Library version 6.6.0

StevenCellist, 2024-06-14:
What steps did you take to make sure the problem is not with the ESP32-C3 device?

patrick-kiwi, 2024-06-14:
It's new straight out of the packet.  Also, I just uploaded a "hello world" sketch to the esp32-c3 and it works fine.

StevenCellist, 2024-06-15:
Please share your complete sketch, and try running the sketch with the build flag `RADIOLIB_DEBUG_BASIC` enabled (e.g. in `BuildOptUser.h`) and share the output, if any.

HeadBoffin, 2024-06-20:
> I also wired up MISO MOSI and SCK just in case the module needs that?

Not so much as Just In Case as Absolutely Required because the radio is operated over SPI - and which ones did you use as the ESP32 comes with two!

jgromes, 2024-06-28:
Closed due to inactivity

closed 2024-06-28 by jgromes

patrick-kiwi, 2024-06-28:
Had a go again today with the esp32C3, wired up everything correctly, but no serial output?  I gave up on it.  Got it working fine with an esp32-devkit-V1 no problem.  

#include <RadioLib.h>
#include <SPI.h>

//esp32C3 settings
#define l_NSS 3
#define l_DIO0 5
#define l_RESET 4
#define l_DIO1 20

//Lora settings
#define FREQUENCY 433.0
#define BANDWIDTH 125.0
#define SPREADING_FACTOR 7
#define CODING_RATE 5
#define OUTPUT_POWER 12
#define PREAMBLE_LEN 8
#define GAIN 0
#define SX127X_SYNC_WORD 0x12

#define RADIOLIB_DEBUG
#define RADIOLIB_VERBOSE

SX1278 radio = new Module(l_NSS, l_DIO0, l_RESET, l_DIO1);

void setup() {
  Serial.begin(115200);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin(FREQUENCY, BANDWIDTH, SPREADING_FACTOR, CODING_RATE, SX127X_SYNC_WORD, OUTPUT_POWER, PREAMBLE_LEN, GAIN);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when new packet is received
  radio.setPacketReceivedAction(setFlag);

  // start listening for LoRa packets
  Serial.print(F("[SX1278] Starting to listen ... "));
  state = radio.startReceive();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  // we got a packet, set the flag
  receivedFlag = true;
}

void loop() {
  // check if the flag is set
  if(receivedFlag) {
    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
    String str;
    int state = radio.readData(str);

    // you can also read received data as byte array
    /*
      byte byteArr[8];
      int numBytes = radio.getPacketLength();
      int state = radio.readData(byteArr, numBytes);
    */

    if (state == RADIOLIB_ERR_NONE) {
      // packet was successfully received
      Serial.println(F("[SX1278] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1278] Data:\t\t"));
      Serial.println(str);

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1278] RSSI:\t\t"));
      Serial.print(radio.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1278] SNR:\t\t"));
      Serial.print(radio.getSNR());
      Serial.println(F(" dB"));

      // print frequency error
      Serial.print(F("[SX1278] Frequency error:\t"));
      Serial.print(radio.getFrequencyError());
      Serial.println(F(" Hz"));

    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("[SX1278] CRC error!"));
    } else {
      // some other error occurred
      Serial.print(F("[SX1278] Failed, code "));
      Serial.println(state);
    }
  }
}

thieri, 2024-08-15:
SX1278 MISO is to connect to D9(MISO) which is also bootstrapping pin. this should be the cause of the issue.