jgromes/RadioLib #1131: RadioLib not compile for Arduino Nano ESP32

andreock, 2024-06-21:
Due to new code related to pin naming(https://forum.arduino.cc/t/esp32servo-library-gives-compiling-errors-with-board-nano-esp32/1157548/4) in the Nano ESP32 platfom RadioLib won't compile anymore.

You can try with a minimal code like this:

```cpp

#include <RadioLib.h>

SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(115200);
  SPI.begin(12, 13, 11, 10);
  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK(433.92, 4.8, 5.0, 135.0, 17, 16, false);
  
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

// counter to keep track of transmitted packets
int count = 0;

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  String str = "Hello";
  int state = radio.transmit(str);

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));
  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));
  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));
  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
  }
  delay(1000);
}

```

```bash
Compilation log(trimmed):

In file included from /Users/andreock/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:223,
                 from .pio/libdeps/arduino_nano_esp32/Adafruit GFX Library/Adafruit_GFX.h:5,
                 from src/main.cpp:18:
/Users/andreock/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:48:61: error: 'digitalPinToGPIONumber' is not a type
 #define pinMode(pin, mode)                          pinMode(digitalPinToGPIONumber(pin), mode)
                                                             ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/arduino_nano_esp32/RadioLib/src/Hal.h:68:18: note: in expansion of macro 'pinMode'
     virtual void pinMode(uint32_t pin, uint32_t mode) = 0;
                  ^~~~~~~
/Users/andreock/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:47:66: error: 'digitalPinToGPIONumber' is not a type
 #define digitalWrite(pin, val)                      digitalWrite(digitalPinToGPIONumber(pin), val)
                                                                  ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/arduino_nano_esp32/RadioLib/src/Hal.h:76:18: note: in expansion of macro 'digitalWrite'
     virtual void digitalWrite(uint32_t pin, uint32_t value) = 0;
                  ^~~~~~~~~~~~
/Users/andreock/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:43:65: error: 'digitalPinToGPIONumber' is not a type
 #define digitalRead(pin)                            digitalRead(digitalPinToGPIONumber(pin))
                                                                 ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/arduino_nano_esp32/RadioLib/src/Hal.h:84:22: note: in expansion of macro 'digitalRead'
     virtual uint32_t digitalRead(uint32_t pin) = 0;
                      ^~~~~~~~~~~
/Users/andreock/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:45:69: error: 'digitalPinToGPIONumber' is not a type
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/arduino_nano_esp32/RadioLib/src/Hal.h:93:18: note: in expansion of macro 'attachInterrupt'
     virtual void attachInterrupt(uint32_t interruptNum, void (*interruptCb)(void), uint32_t mode) = 0;
                  ^~~~~~~~~~~~~~~
```

jgromes, 2024-06-22:
Arduino Nano ESP32 is not a supported platform, as can be seen from the list in readme. If it worked before, it was mostly just down to lucky coincidence.

The Nano ESP32 does not seem to conform to Arduino API (a strange fact considering that it is an official Arduino board) and in actuality completely breakes the API by using macros to define things like `digitalWrite`, which is an outstandingly stupid move.

Honestly I'm tempted to just not support this platform.

andreock, 2024-06-22:
Yes it works with ESP32-S3 DevkitC. I'm working on porting RadioLib to Nano ESP32 since I need for job, I forked it and now the begin works, I'm working also on others method. Can I make a PR when I will finish the porting?

jgromes, 2024-06-22:
> Yes it works with ESP32-S3 DevkitC

That is not what I meant. What happens if you use the Arduino ESP32 (i.e., select ESP32 in Arduino IDE or Platformio, whichever you use) and try to use that instead of the Arduino Nano ESP32 core.

> I'm working on porting RadioLib to Nano ESP32

There's not much that needs to be done, essentially the only thing you have to do is `#undef` the offending macros in BuildOpt.h. However, that's not really the point - does RadioLib really need to support a non-compliant, rarely used platform when ESP32 core is available, widely used, Arduino-API compliant and tested?

closed 2024-07-04 by andreock

BugProg, 2025-02-03:
Hi!

Haha, 😂 I started my project a long time ago with my Arduino Nano ESP32, and I am just starting to work on the radio. I'm so upset to see the only Arduino board that is not supported is mine 😭.

I can understand you can't support every Arduino board, specially when it's working differently.
When I bought the Arduino Nano ESP32 and the Lora module, I didn't see the incompatibility issue with it.

I'm not asking to add the support for the Arduino Nano I will try to fix the code myself, but I need some help.

I followed this link : https://forum.arduino.cc/t/esp32servo-library-gives-compiling-errors-with-board-nano-esp32/1157548/4 originally posted in the first post.

I replaced:

```c
void analogWrite( uint8_t APin, uint16_t AValue );
```
to 
```c
void (analogWrite)( uint8_t APin, uint16_t AValue );
```

And I can build the code without any issue or errors.👌 But nothing happened when I uploaded the code to the Arduino. I tried with official supported Arduino, and it's working. Do you have an idea of what I have to do to make the RadioLib working?

andreock, 2025-02-03:
Hi, I make a fork of radiolib that works with Arduino Nano ESP32: https://github.com/andreock/RadioLib