jgromes/RadioLib #1132: RSSI bug in PhysicalLayer? also getFrequencyError() seems missing

onyx99, 2024-06-22:
Hello i seem to have a bug with RSSI using the following code and using latest version of RadioLib

I am using Hal and SX1262

RSSI is always 25.0 with phy

also radio.getFrequencyError() seems missing..

Code to reproduce:

// get pointer to the common layer
PhysicalLayer* phy = (PhysicalLayer*)&radio;

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

void setFlag(void) {
 // we got a packet, set the flag
  receivedFlag = true;
}

void loop() {
	
	if(receivedFlag) {
    // reset flag
    receivedFlag = false;

   
    int numBytes = radio.getPacketLength();
    
	
	uint8_t byteArr[256];
	int state = radio.readData(byteArr, numBytes);	
 
    if (state == RADIOLIB_ERR_NONE) {
		printf("received rssi=%.2f snr=%.2f\n",radio.getRSSI(), radio.getSNR());
		printf("received rssi=%.2f snr=%.2f\n",phy->getRSSI(), phy->getSNR());
	}
	
	// put module back to listen mode
    radio.startReceive();
	
 }
	
}

Result:

received rssi=-19.00 snr=5.75
received rssi=-25.00 snr=5.75

result physical
received rssi=-79.00 snr=-16.75
received rssi=-25.00 snr=-16.75

Kind regards.

jgromes, 2024-06-22:
> Code to reproduce

That is not sufficient to reproduce the issue as critical sections of the code are missing, you have to post the whole thing.

Also, I'm confused which numbers in the result are from directly calling the `radio` instance methods and which ones are via `phy`.

onyx99, 2024-06-22:
`// include the library
#include <RadioLib.h>

// include the hardware abstraction layer
#include "PiHal.h"

SX1262 radio = new Module(hal, 7, 17, 22, RADIOLIB_NC);

// get pointer to the common layer
PhysicalLayer* phy = (PhysicalLayer*)&radio;

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

void setFlag(void) {
 // we got a packet, set the flag
  receivedFlag = true;
}

void loop() {
	
	if(receivedFlag) {
    // reset flag
    receivedFlag = false;

  
    int numBytes = radio.getPacketLength();
    
	
	uint8_t byteArr[256];
	int state = radio.readData(byteArr, numBytes);	
 
    if (state == RADIOLIB_ERR_NONE) {
		printf("received rssi=%.2f snr=%.2f\n",radio.getRSSI(), radio.getSNR());
		printf("received rssi=%.2f snr=%.2f\n",phy->getRSSI(), phy->getSNR());
	}
	
	// put module back to listen mode
    radio.startReceive();
	
 }
	
}

void setup() {
	
  int state = radio.begin();
  if(state == RADIOLIB_ERR_NONE) {
    printf("success1!\n");
  } else {
    printf("failed, code %d\n", state);
	while(true);
  }

  // or we can use the "phy" pointer to access the common layer
  // PhysicalLayer has some common configuration
  printf("[PHY] Changing frequency ... \n");

  // set the function that will be called
  // when new packet is received
  radio.setPacketReceivedAction(setFlag);
  
 // start listening for LoRa packets
  printf("[868] Starting to listen ... \n");
  state = radio.startReceive();
  if (state == RADIOLIB_ERR_NONE) {
	 printf("success2!\n");
  } else {
	printf("failed, code %d\n");
  
  }

}

int main(int argc, char** argv) {
	
  setup();
	
  for(;;) {
	  
	  loop();
	  
  }

 return(0);
	
}`

jgromes, 2024-06-25:
Thank you, with this I was able to reproduce the issue. The -25 return by getRSSI is actually an error code for an unsupported implementation due to a missing override in method SX126x. It is somewhat unfortunate this error value can be interpreted as a real RSSI. Nevertheless, it is fixed in the mentioned commit, thanks for reporting!

Regarding `getFrequencyError`, this method is not available in PhysicalLayer if that's what you meant in the original post. The reason for that is quite simple, not all supported radio modules have that functionality.

closed 2024-06-25 by jgromes