jgromes/RadioLib #1134: DreamLNK LLCC68-915MHz with ESP32/Pico SPI error

LimesKey, 2024-06-23:
### Sketch that is causing the module fail

```c++
/*
   RadioLib SX126x Spectrum Scan Example

   This example shows how to perform a spectrum power scan using SX126x.
   The output is in the form of scan lines, each line has 33 power bins.
   First power bin corresponds to -11 dBm, the second to -15 dBm and so on.
   Higher number of samples in a bin corresponds to more power received
   at that level.

   To show the results in a plot, run the Python script
   RadioLib/extras/SX126x_Spectrum_Scan/SpectrumScan.py

   WARNING: This functionality is experimental and requires a binary patch
   to be uploaded to the SX126x device. There may be some undocumented
   side effects!

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// this file contains binary patch for the SX1262
#include <modules/SX126x/patches/SX126x_patch_scan.h>

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1262 radio = new Module(5, 4, 17, 16);

void setup() {
  Serial.begin(115200);

   // initialize SX1262 FSK modem with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.beginFSK();
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // upload a patch to the SX1262 to enable spectral scan
  // NOTE: this patch is uploaded into volatile memory,
  //       and must be re-uploaded on every power up
  Serial.print(F("[SX1262] Uploading patch ... "));
  state = radio.uploadPatch(sx126x_patch_scan, sizeof(sx126x_patch_scan));
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // configure scan bandwidth to 234.4 kHz
  // and disable the data shaping
  Serial.print(F("[SX1262] Setting scan parameters ... "));
  state = radio.setRxBandwidth(234.3);
  state |= radio.setDataShaping(RADIOLIB_SHAPING_NONE);
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }
}

void loop() {
  Serial.print(F("[SX1262] Starting spectral scan ... "));

  // start spectral scan
  // number of scans in each line is 2048
  // number of samples: 2048 (fewer samples = better temporal resolution)
  int state = radio.spectralScanStart(2048);
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // wait for spectral scan to finish
  while(radio.spectralScanGetStatus() != RADIOLIB_ERR_NONE) {
    delay(10);
  }

  // read the results
  uint16_t results[RADIOLIB_SX126X_SPECTRAL_SCAN_RES_SIZE];
  state = radio.spectralScanGetResult(results);
  if(state == RADIOLIB_ERR_NONE) {
    // we have some results, print it
    Serial.print("SCAN ");
    for(uint8_t i = 0; i < RADIOLIB_SX126X_SPECTRAL_SCAN_RES_SIZE; i++) {
      Serial.print(results[i]);
      Serial.print(',');
    }
    Serial.println(" END");
  }

  // wait a little bit before the next scan
  delay(5);
}
```

### Hardware setup
I have the LORA module attached to a custom PCB like this, 
[image]
I'm using standard Dupont wires and connecting the appropriate pins to the microcontroller,
[image]
[image]

### Debug mode output
Could not mange to get this to work, I got,
```c++
c:\Users\Ryan\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:56:32: note: '#pragma message: 
RadioLib Info
Version:  "6.6.0.0"
Platform: "ESP32"
Compiled: "Jun 23 2024" "17:29:08"'
   56 |   #pragma message(RADIOLIB_INFO)
      |                                ^
```

### Additional info (please complete):
 - MCU: Tried both ESP32-WROOM-D & Raspberry Pi Pico 2W
 - Wireless module:  `DreamLNK DL-LLCC68-S-915` [link](https://www.lcsc.com/product-detail/LoRa-Modules_DreamLNK-DL-LLCC68-S-915_C2894791.html)
 - Arduino IDE version: 2.3.2
 - Library version: 6.6.0

I've tried many many LORA libraries to try to get my module to work but with no success. It's some issue with SPI but I've checked all the wiring and pinouts and it looks good. I've tried many different wire configurations and I've even switched to different SPI pinouts but still no difference.

Without any pins connected, I get, 
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
[SX1262] Initializing ... ets Jul 29 2019 12:21:46
```
With the appropriate pins connected,
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
[SX1262] Initializing ... failed, code -2
``` 
I have never gotten another error code other than `-2`. I have gotten similar errors trying other libraries.

jgromes, 2024-06-24:
> Could not mange to get this to work, I got,

Did you upload the sketch after compiling it with the chagned options?

There's a wiki page called [Troubleshooting guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide), I suggest following it. Especially the part about different SPI interfaces. I would also check the connections with a multi-meter, to ensure there really is an electrical connection. The soldering on the radio module pads looks a bit suspicious. Also, sometimes these jumper cables are very unreliable, if they are loose on the pins there is not a good connection.

TL;DR It looks to be a hardware issue, especially if you also got these errors from different libraries.

LimesKey, 2024-06-24:
> Did you upload the sketch after compiling it with the changed options?

Of course

> There's a wiki page called [Troubleshooting guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide), I suggest following it.

I've already looked very deeply at that and tried changing the SPI interface. On this library, it's VSPI so I wouldn't have to change it but on other libraries, it differed and was HSPI.

>check the connections with a multi-meter

Did you not notice the multimeter in the second photo? I've done a continuity test on both the LORA to PCB and LORA to ESP32 and it's all fine.

>TL;DR It looks to be a hardware issue, especially if you also got these errors from different libraries.

I'll offer to ship you or someone else a LORA module, the PCB, an antenna and wires, for if you'd like to test it yourself. I can ship anywhere in North America.

jgromes, 2024-06-24:
> Of course

Well, what was the output in the serial monitor? Or was there no output? That seems unlikely considering that basic info (the same as shown during build) gets printed at the beginning regardless of any successes or fails in communication with the module, so if that is the case, I would double-check you have actually uploaded the correct thing.

> Did you not notice the multimeter in the second photo

I did not actually, though I did not care much to play Where's Waldo at that point. You did not mention checking it, hence I suggested to do so. It's the most basic thing, yet still quite a few people skip that crucial step.

> I'll offer to ship you or someone else a LORA module, the PCB, an antenna and wires, for if you'd like to test it yourself

I'm going to have to pass on that offer. I'm sure there are some freelancing services available somewhere - not here though.

LimesKey, 2024-06-24:
> Well, what was the output in the serial monitor? 

Ah sorry I thought you were talking about the default Arduino, I also thought that the `note: '#pragma message: ` message was an error code. I have the output.  [debug-mode-SPI-RadioLib.txt](https://github.com/user-attachments/files/15960141/debug-mode-SPI-RadioLib.txt)

jgromes, 2024-06-25:
>  I also thought that the note: '#pragma message:  message was an error code.

No, it is what is says it is - a message. Neither an error, nor a warning.

The debug output shows strange behavior - sometimes MISO returns zeroes, sometimes ones, sometimes what seems to be random garbage. Because it is so unstable, it's probably not a bad SPI configuration, but it strengthens the case for hardware or wiring problem. So I'm afraid I can't help you beyond suggesting to check the wiring again. Another step would be to use an oscilloscope or a logic analyzer if you have access to that.

LimesKey, 2024-06-29:
After a long, troublesum time with a logic analyzer, I found one that the library is using HSPI and not the normal and default VSPI like other libraries. I had definelty though this might be an issue and for other libraries I manually change it to VSPI but I don't think I did this for RadioLIB. Here's a quick screenshot of my logic analyzer for some pins of HSPI, while for VSPI there was nothing the entire time.  [image]

It no longer seems like a SPI communication issue but now I'm recieving, 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
```

LimesKey, 2024-06-29:
Oh shoot the LORA module is burning hot and is starting to smell like burning plastic - that's why the USB connector is saying 4W... How is this possible, the only real thing I changed is the SPI pins???

LimesKey, 2024-06-30:
Sorry, I accidentally swapped VCC with GND and GND with VCC, I will replace the module and try again... :facepalm:

jgromes, 2024-06-30:
> After a long, troublesome time with a logic analyzer, I found that the library uses HSPI and not the normal and default VSPI like other libraries.

It uses whatever your platform defines as the default `SPI`. This is also covered in the [Troubleshooting guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide) and eve the [Basics](https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup) page ...

LimesKey, 2024-07-03:
Not sure what happened there, I read and did that but something went wrong. Anyways though, I made a little more progress today, it appears with the default HSPI pin configuration, when `MISO` is connected to Pin 12, the module spits out a flashing error message
```
Traceback (most recent call last):
  File "/home/ryan/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool.py", line 37, in <module>
    esptool._main()
  File "/home/ryan/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/__init__.py", line 1064, in _main
    main()
  File "/home/ryan/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/__init__.py", line 859, in main
    operation_func(esp, args)
  File "/home/ryan/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/cmds.py", line 466, in write_flash
    flash_end = flash_size_bytes(
  File "/home/ryan/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/util.py", line 37, in flash_size_bytes
    if "MB" in size:
TypeError: argument of type 'NoneType' is not iterable
Failed uploading: uploading error: exit status 1
```
This appears to be because pin 12 is connected to some sort of internal voltage regulator and by pulling the pin high during boot, the wrong voltage is being used causing flashing errors.

Secondly,  connecting the NSS pin from the LLCC68 module to the microcontroller on either Pin 15 (HSPI) or 5 (VSPI) following the ESP32 SPI CS pin guide [here](https://randomnerdtutorials.com/esp32-spi-communication-arduino/), this causes the module to lock up and does not allow for flashing or for a serial output as far as I know.

This is after soldering on a new LORA module straight from LCSC instead of the AliExpress module I was using earlier, I've also replaced the Dupont cables as they were feeling loose. I'm just using this issue now as a personal recollection of the steps I've been trying in case anyone else can get it to work. I'm about ready to give up though, it's terrible doing the same operational steps over and over and without getting much farther than either a generic `Invalid Header` issue or a SPI communication issue.

jgromes, 2024-07-03:
> I later tried to define the SPI config with custom pins but I don't think this work as it would just immediately spit out the -2 spi error instead of a few second delay like it normally does,

Your sketch shows you are using `SX1262` as the radio class, yet you seem to actually be using LLCC68. Those two are not the same, and RadioLib will actively check whether the hardware matches the software. Result of that check is shown in the debug output with `RADIOLIB_DEBUG_BASIC` enabled.