jgromes/RadioLib #1139: [Issue: Module not working]: No more communication after "[SX1262] Initializing ..."

Luddi96, 2024-06-25:
**Sketch that is causing the module fail**

```c++
/*
   RadioLib SX126x Ping-Pong Example

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// uncomment the following only on one
// of the nodes to initiate the pings
//#define INITIATING_NODE

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1262 radio = new Module(8, 3, 5, 4);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

// or using CubeCell
//SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

// save transmission states between loops
int transmissionState = RADIOLIB_ERR_NONE;

// flag to indicate transmission or reception state
bool transmitFlag = false;

// flag to indicate that a packet was sent or received
volatile bool operationDone = false;

// this function is called when a complete packet
// is transmitted or received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  // we sent or received a packet, set the flag
  operationDone = true;
}

void setup() {
  Serial.begin(9600);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when new packet is received
  radio.setDio1Action(setFlag);

  #if defined(INITIATING_NODE)
    // send the first packet on this node
    Serial.print(F("[SX1262] Sending first packet ... "));
    transmissionState = radio.startTransmit("Hello World!");
    transmitFlag = true;
  #else
    // start listening for LoRa packets on this node
    Serial.print(F("[SX1262] Starting to listen ... "));
    state = radio.startReceive();
    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while (true);
    }
  #endif
}

void loop() {
  // check if the previous operation finished
  if(operationDone) {
    // reset flag
    operationDone = false;

    if(transmitFlag) {
      // the previous operation was transmission, listen for response
      // print the result
      if (transmissionState == RADIOLIB_ERR_NONE) {
        // packet was successfully sent
        Serial.println(F("transmission finished!"));

      } else {
        Serial.print(F("failed, code "));
        Serial.println(transmissionState);

      }

      // listen for response
      radio.startReceive();
      transmitFlag = false;

    } else {
      // the previous operation was reception
      // print data and send another packet
      String str;
      int state = radio.readData(str);

      if (state == RADIOLIB_ERR_NONE) {
        // packet was successfully received
        Serial.println(F("[SX1262] Received packet!"));

        // print data of the packet
        Serial.print(F("[SX1262] Data:\t\t"));
        Serial.println(str);

        // print RSSI (Received Signal Strength Indicator)
        Serial.print(F("[SX1262] RSSI:\t\t"));
        Serial.print(radio.getRSSI());
        Serial.println(F(" dBm"));

        // print SNR (Signal-to-Noise Ratio)
        Serial.print(F("[SX1262] SNR:\t\t"));
        Serial.print(radio.getSNR());
        Serial.println(F(" dB"));

      }

      // wait a second before transmitting again
      delay(1000);

      // send another one
      Serial.print(F("[SX1262] Sending another packet ... "));
      transmissionState = radio.startTransmit("Hello World!");
      transmitFlag = true;
    }
  
  }
}
```

**Hardware setup**
Heltec HT-CT62 Module on breakout PCB

**Debug mode output**
Enabled, however no additional output. Only thing I get is
[SX1262] Initializing ...
After that, no more information on the serial port..

**Additional info (please complete):**
Using the HT-CT62 Module. It is a combined ESP32-C3 and SX1262 chip
See internal schematic here: https://resource.heltec.cn/download/HT-CT62/HT-CT62_Schematic_Diagram.pdf
And pinout here: https://heltec.org/project/ht-ct62/

Arduino set to ESP32 C3 Dev Module

jgromes, 2024-06-25:
> Debug mode output
> Enabled, however no additional output. 

You are not enabling the debug mode - the [wiki page](https://github.com/jgromes/RadioLib/wiki/Debug-mode) specifies how to do it, either do it in BuildOptUser.h, or define it in your build system. If you are using Arduino IDE, the former is probably easier for you. My guess is that you are probably not using the correct SPI bus, butr that is something that you will need to clarify with the manufacturer.

StevenCellist, 2024-06-25:
One crucial piece of information would be your wiring: did you connect 18/19 to your serial monitor, or 20/21?

Luddi96, 2024-06-25:
> One crucial piece of information would be your wiring: did you connect 18/19 to your serial monitor, or 20/21?

I am using a USB/UART bridge, connected to pins 20/21

> You are not enabling the debug mode - the [wiki page](https://github.com/jgromes/RadioLib/wiki/Debug-mode) specifies how to do it, either do it in BuildOptUser.h, or define it in your build system. If you are using Arduino IDE, the former is probably easier for you.

Ah sorry, missed that. With the debugging turned on, I get this during compile:
`note: '#pragma message: 
RadioLib Info
Version:  "6.6.0.0"
Platform: "ESP32"
Compiled: "Jun 25 2024" "20:38:26"'
   56 |   #pragma message(RADIOLIB_INFO)`

And serial output for #define RADIOLIB_DEBUG_BASIC:
`[SX1262] Initializing ... RLB_DBG: 
RadioLib Info
Version:  "6.6.0.0"
Platform: "ESP32"
Compiled: "Jun 25 2024" "20:38:29"`

Or for #define RADIOLIB_DEBUG_SPI:
`[SX1262] Initializing ... `

I am pretty sure I have double-checked the pinning according to the datasheet and adjusted NRS, RESET, DIO, BUSY. Is there a way to change the MISO, MOSI, SCK pins as well? Or what are the defaults used for ESP32?

I have found another person with those settings and another library have success.
`#define USE_SX1262
#define LORA_SCK 10
#define LORA_MISO 6
#define LORA_MOSI 7
#define LORA_CS 8
#define LORA_DIO0 RADIOLIB_NC
#define LORA_RESET 5
#define LORA_DIO1 3
#define LORA_DIO2 RADIOLIB_NC
#define LORA_BUSY 4
#define SX126X_CS LORA_CS
#define SX126X_DIO1 LORA_DIO1
#define SX126X_BUSY LORA_BUSY
#define SX126X_RESET LORA_RESET
#define SX126X_DIO2_AS_RF_SWITCH
#define SX126X_DIO3_TCXO_VOLTAGE 1.8`

StevenCellist, 2024-06-25:
Here's the catch: you connected to a different serial port than the usual Serial: the normal Serial is connected to 18/19, while 20/21 is either USBSerial or you need to define USB_MODE=1 and USB_CDC_ON_BOOT=1 or whatever the exact flag is - Google is your friend. 
So either you enable those two flags, or route the DEBUG_PORT to USBSerial in BuildOpt(User).h.
Hoping that I didn't make a mistake... :)

Luddi96, 2024-06-25:
Interesting, I thought it was the other way around.. How come I get some debug information then, from the #define RADIOLIB_DEBUG_BASIC, but not from the #define RADIOLIB_DEBUG_BASIC?

Also, can you give me a hint, where I can find the default MISO/MOSI/SCK pins or where I can change them?

StevenCellist, 2024-06-25:
The output "[SX1262] Initializing ..." is not from debug, but from Serial.print() in the sketch. So none of the debug output is coming through.
I have a sketch already for the CT62 but not at my desk. Will send the missing part for SPI here when possible.

Luddi96, 2024-06-25:
I think I found a fix:

I edited line 95-97 in ArduinoHal.cpp

`void inline ArduinoHal::spiBegin() {
  spi->begin(10,6,7,8);
}`

Now SPI Debug gives:
`Initializing ... RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	AA	AA	
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	A2	A2	
RLB_SPI: CMDR	1D	3	20	
RLB_SPI: SI				0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
RLB_SPI: SO	A2	A2	A2	A2	53	58	31	32	36	31	20	56	32	44	20	32	44	30	32	0	
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	AA	AA	
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
`
[SX1262] Received packet!
[SX1262] Data:		Hello World!
[SX1262] RSSI:		-19.00 dBm
[SX1262] SNR:		10.75 dB
[SX1262] Sending another packet ... transmission finished!

StevenCellist, 2024-06-25:
Good to see your debug output is all good.

& I'm slightly confused by my own sketch, but I assume I've done it right:
```
#include <RadioLib.h>
#include <SPI.h>

SX1262 radio = new Module(8, 3, 5, 4);

void setup() {
  Serial.begin(115200);
  SPI.begin(10, 6, 7);
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
}
```

Let me know if it works this way, but I'd at least expect `(10, 6, 7, 8)` instead of just `(10, 6, 7)`, no clue why I did what I did there.

Luddi96, 2024-06-25:
Ahh yes this works too! Very nice, thank you!

So the solution once again, for anyone trying to use a HT-CT62 with this library:

Set
`SX1262 radio = new Module(8, 3, 5, 4);`
And do
`SPI.begin(10, 6, 7);` before calling `radio.begin();`

The pinout of the HT-CT62:

Pin Name on SX1262 | Pin Name on C3 | Physical Pin on C3 | GPIO Number on C3
-- | -- | -- | --
MISO | MTCK | 12 | 6
MOSI | MTDO | 13 | 7
SCK | GPIO10 | 16 | 10
NSS | GPIO8 | 14 | 8
RST | MTDI | 10 | 5
DIO1 | GPIO3 | 8 | 3
BUSY | MTMS | 9 | 4

closed 2024-06-25 by Luddi96

mbssilva, 2024-12-15:
Hello.

I'm facing the same problem with ping pong example. After "[SX1262] Initializing ... " I don't have more logs in serial.
I'm not experienced in in LoRa (I am studing it), so I hope I'm not questioning something too basic.
@Luddi96 , where do I find this ArduinoHal file to edit?
And @jgromes , is it supposed to work the following snippet:
radio.begin(915.0, 125.0, 9, 7, RADIOLIB_SX126X_SYNC_WORD_PRIVATE, 10, 8, 1.8, false);

With 915.0 MHz?

Thanks in advance.

jgromes, 2024-12-15:
@mbssilva I would suggest to follow the steps in my first post in this issue, that is to start with the debug mode. This behavior is usually caused on microcontrollers that stop printing to serial when they enter an infinite loop (e.g. the ESP32). This was fixed in the latest examples by adding a `delay(10);` into all halting loops, i.e. `while(true) { delay(10); }` instead of just `while(true);`. You have not shared any information about your code or hardware, so I can't tell if this is also the case for you.

> is it supposed to work the following snippet:

Not sure what you're asking about - if it's about the serial output stopping, then most likely not. If it's about changing frequency to 915 MHz then yes, it will do that, among other things. see the API documentation, linked from the readme.

mbssilva, 2024-12-15:
@jgromes I thank you for the reply.
To give you more context: I posted that it was almost 2AM and I had been struggling with the Hello World the entire day. So I was simply exhausted.
Today, it worked and I going to let what I did here to future interested users.

I bought two Heltec ESP LoRa 32 V3. Then, following the getting started guide it didn't work. Futher searches and I discovered that Heltec simply changed a lot of specs from V2 to V3 and doesn't guarantee backward compatibility. In fact, all data available online to V2 (that is not that much and most part provided by forum user, not the oficial support) are like outdated to V3.
Some hours later I eventually came here to RadioLib (after testing every lib or code snippet I found), just to hit the wall: It's quite more advanced than something than a simple hobbist is able to deal with in general. It's not a criticism to your work that is awesome here. Just saying that compared do LoRa.h it's like getting from car to plane and required to fly it. So I was lost.
Firt tried the api reference (https://jgromes.github.io/RadioLib/class_s_x126x.html#a936a40038e05740a528f2b53f8e17010) but some things in the examples were not matching the docs (like the snipped the Luddi96 shared of radio.begin has more parameters than the function documented in the api reference). And I was not understanding what connections that should be inputed in "Module()".
After some research I found the Heltec ESP LoRa 32 V3 pin order to new Module:
`SX1262 radio = new Module(8, 14, 12, 13);`
So magically you examples to SX1262 started to work.

About the previous question, it was about the frequency so thank you for the clarification.

I know that probably it's very elementar to some of you more experienced. Thank you for keeping this forum/issue space. It was quite helpful.

For future interested parties I used:
examples/SX126x/SX126x_Receive_Interrupt/SX126x_Receive_Interrupt.ino
examples/SX126x/SX126x_Transmit_Interrupt/SX126x_Transmit_Interrupt.ino
And then I changed the pins in Module() instancing.

jgromes, 2024-12-17:
@mbssilva 

> some things in the examples were not matching the docs (like the snipped the Luddi96 shared of radio.begin has more parameters than the function documented in the api reference)

Could you point me exactly where this confusion was coming from? The API reference is generated directly from the source code (/the comments in it), so it should always be up to date.

More generally, is there comething we can do to make it easier for new users? I really try to point people to the examples and wiki pages, so just wondering whether there's something that could be done better from the perspective of a new user :)