jgromes/RadioLib #1143: {LR11x0] IRQ is always applied to DIO9. DIO11 is not supported as an event signal yet.

lyusupov, 2024-06-30:
**Describe the bug**

Some new hardware design uses DIO11 signal to deliver the LR11x0 radio interrupt to the MCU. This is allowed by LR1121 'User Mnanual'.

However, `irq2` argument for the RadioLib `setDioIrqParams()` is always zero.

<br>
<br>

[image]

<br>
<br>

**Expected behavior**

When a hardware design uses DIO11 signal to deliver the LR11x0 radio interrupt to the MCU - we can either:
1) apply a copy of `irq1` in every setDioIrqParams(irq1, irq2) call to `irq2` argument, or
2) use an additional `IrqSwap` argument for LR11x0 **begin()** method. When `IrqSwap` is `true` - apply the `irq mask` to the second argument of setDioIrqParams() rather than the first one.

I will be happy to use your own method of the issue resolution other than two my ones listed above.

**Additional info :**

 - MCU: ESP32
 - Library version 6.6.0

HeadBoffin, 2024-07-01:
> Some new hardware design _(due to h/w compatibility reasons)_

Can you clarify if this is your hardware design or some commodity item generally for sale? If the latter, link please.

jgromes, 2024-07-01:
@lyusupov I agree with @HeadBoffin that it would be good to know if we are aiming for a specific hardware (e.g. one that allows to use both DIO9 and DIO11, that would require some thinking). To add some context, I have used SeeedStudi WM1110 as the development platform, hence no DIO11 usage. 

However, at the moment, I don't really see a good use-case for separate functions on DIO pins, so the first proposed solution (to apply the same IRQ configuration to both DIO pins) is OK from my point of view, at least for now. So if you could send that as a PR, that would be great!

lyusupov, 2024-07-01:
The hardware is an RF module under development.
It is pin compatible with RFM95.
Both DIO9 and DIO11 signals  are exposed on it's pads.

**New PCB designs**  with this module on-board can use either DIO9 or DIO11 for the purpose of IRQ signal.
However, when the **module is in use as a replacement of RFM95 on an existing PCB design  - only DIO11 matches the actual (DIO1) IRQ pad of the RFM95**. 
That's why support for DIO11 is requested in this issue report.
 
> So if you could send that as a PR, that would be great!

PR #1144 is available. Please, review and merge.

jgromes, 2024-07-02:
@lyusupov thank you for the context, and for the PR - since #1144 is merged, this can be closed.

closed 2024-07-02 by jgromes