jgromes/RadioLib #1153: DevNonce is too small : BUG

askmurphy, 2024-07-08:
Hello, found a bug with the "LilyGO TTGO T-Beam - LoRa 868MHz - NEO-M8N GPS". If I use radiolib and compile the RadioLib LoRaWAN Starter Example with the correct LoRawan keys and create the relevant LoRa sensor at TTN, it all works fine: the data arrives neatly at TTN. But as soon as I disconnect the T-Beam from the power and connect it again after 5 minutes, I no longer have a connection with TTN: I get this error message on the TTN console: "DevNonce is too small". Also a DevNonce reset on the TTN panel does not help. It only works if I remove the device from the TTN panel and recreate it. This is quite clumsy and uncooperative. Hope this can be resolved soon ;-)

**Additional info:**
 - LilyGO TTGO T-Beam - LoRa 868MHz - NEO-M8N GPS"
 - Wireless module type SX1276
 - Arduino IDE version  2.3.2 (Windows 11)
 - Library version: latest

StevenCellist, 2024-07-08:
Hi @askmurphy, please refer to the [README in the LoRaWAN examples folder](https://github.com/jgromes/RadioLib/tree/master/examples/LoRaWAN#radiolib-persistence) for information on this. There you will find an explanation and links to sketches that will resolve this problem for you.

askmurphy, 2024-07-08:
Ok, will check this tomorrow. Thanks!

askmurphy, 2024-07-09:
Just curious, why are there 2 different version of the radiolib examples?  
One which has the nonces issue and the last one without (not working here I am afraid..) ?
It's confusing...

StevenCellist, 2024-07-09:
> Just curious, why are there 2 different version of the radiolib examples?

That is explained in the README I linked earlier. The implementation is different for all platforms, and thus difficult to keep up to date in RadioLib itself.

> New device created in TTN, first Join ok.

Please provide the logs created by your device upon the first boot, with `RADIOLIB_DEBUG_PROTOCOL` enabled in `BuildOptUser.h`. And then also the debug logs from the second boot. Then I can see if there is a mistake in the examples or in your usage.

askmurphy, 2024-07-09:
Ok, I have tested it again. Double checked correct radiolib version v6.6.0 and example https://github.com/radiolib-org/radiolib-persistence/tree/main/examples/LoRaWAN_ESP32   (maybe I had still the wrong example the previous test??)

RADIOLIB_DEBUG_PROTOCOL enabled in BuildOptUser.h also.

Created new device at TTN, compiled, joined and 1st time: it works great again.
Powered off the device and powered on again after a few minutes, no errors anymore. Seems to be it works ;-)

[firstboot_log.txt](https://github.com/user-attachments/files/16145465/firstboot_log.txt)
[secondboot.txt](https://github.com/user-attachments/files/16145430/secondboot.txt)

So, it seems to be ok ;-)

Maybe an idea to make a TTN sub-folder in the examples-folder from your nice Library?  
It's a bit confusing now for some.

I will rewrite the software of my little Pond-device (https://github.com/askmurphy/Lora-pH-temperature-sensor) using your library now.

Many thanks for your help and keep on the good work, really appreciated !!!!!!!!!!!

StevenCellist, 2024-07-09:
Awesome, looks all good.

I agree that the split in the examples may be a bit confusing, but it has been a topic of discussion before and we settled on the current solution.. not completely perfect but best maintainable.

Veel succes verder ;)

closed 2024-07-09 by StevenCellist

askmurphy, 2024-07-23:
hello again,
Tried today the T-Beam again with the above code, after uploading the software I have the DevNonce trouble again.
Not sure what's happend the last 2 weeks but it doesn't work now.

```
Saving nonces to flash
Join failed: -1116
Boots since unsuccessful join: 1
Retrying join in 60 seconds
Sleeping

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1288
load:0x40078000,len:13872
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3048

```

askmurphy, 2024-07-23:
```
Time
Type
Data preview
Verbose stream

    arrow_upward
    20:30:00
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    arrow_upward
    20:28:22
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    arrow_upward
    20:25:43
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    info
    20:25:21
    Stream reconnected
    The stream connection has been re-established
    info
    20:25:16
    Stream connection closed
    The connection was closed
    arrow_upward
    20:22:35
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    arrow_upward
    20:19:28
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    arrow_upward
    20:16:20
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    arrow_upward
    20:14:12
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    arrow_upward
    20:13:04
    Join-request to cluster-local Join Server failed
    DevNonce is too small
    clear_all
    20:06:23
    Events cleared
    The events list has been cleared
```

askmurphy, 2024-07-23:
Tested this with the Arduino IDE, also with Visual Studio: same troubles.

StevenCellist, 2024-07-23:
If you used the T-Beam for other sketches/programs or erased the flash for whatever reason, the DevNonces are likely lost. In that case you need to reset the used DevNonces on your LoRaWAN Network Server.

askmurphy, 2024-07-23:
Have tried that a few times, didn't work.
So, I have removed the device in the TTN console and created it again: now it works again.
2 tests: first with Arduino IDE, compiled and uploaded: TTN Lora connection succesfull.
2nd test: same Lora keys but different program (Visual Studio code): works too.
Weird...     What is DevNonces and why did they make it this way, seems a but buggy...

askmurphy, 2024-07-25:
Does anyone have an example of a simple radiolib (LoRaWan) code with the following:
- no deep sleep function (it's an adapter powered sensor, which has also a display to show the actual temperature and pH).
- using CayenneLPP to send the temperature and pH value every 5 minutes to TTN

Would be nice!

StevenCellist, 2024-07-27:
@askmurphy this is a question that is completely unrelated to this topic (it isn't even an issue now); please open a Discussion for this.