jgromes/RadioLib #1157: error: missing binary operator before token "("        #if !defined(ESP_IDF_VERSION) || (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5,0,0))

MissingDirection, 2024-07-10:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Troubleshooting Guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide) Wiki page and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

```c++
paste the sketch here, even if it is an unmodified example code
```

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
c:\Users\33119\Documents\Arduino\libraries\RadioLib\src\ArduinoHal.cpp:127:78: error: missing binary operator before token "("
       #if !defined(ESP_IDF_VERSION) || (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5,0,0))
                                                                              ^
c:\Users\33119\Documents\Arduino\libraries\RadioLib\src\ArduinoHal.cpp:132:78: error: missing binary operator before token "("
       #if !defined(ESP_IDF_VERSION) || (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5,0,0))
                                                                              ^
c:\Users\33119\Documents\Arduino\libraries\RadioLib\src\ArduinoHal.cpp:170:76: error: missing binary operator before token "("
     #if !defined(ESP_IDF_VERSION) || (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5,0,0))
                                                                            ^
c:\Users\33119\Documents\Arduino\libraries\RadioLib\src\ArduinoHal.cpp: In member function 'virtual void ArduinoHal::noTone(uint32_t)':
c:\Users\33119\Documents\Arduino\libraries\RadioLib\src\ArduinoHal.cpp:174:19: error: 'ledcDetach' was not declared in this scope
     ledcDetach(pin)

**Additional info (please complete):**
 - MCU: ESP32
 - Wireless module type SX1276
 - Arduino IDE version  2.3.2
 - Library version 6.6.0

jgromes, 2024-07-10:
Which version of the ESP32 Arduino core are you using? Most likely something rather outdated, since it doesn't seem to contain the `ESP_IDF` macros RadioLib uses to determine ESP-IDF version.

jgromes, 2024-07-26:
Closed due to inactivity.

closed 2024-07-26 by jgromes

oisact, 2024-11-26:
Updating the Espressif 32 Platform to version 6.9.0 in Visual Studio Code solved this issue for me.