jgromes/RadioLib #1160: Request for Robot36 and Robot72 in SSTV

Panger95, 2024-07-13:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Troubleshooting Guide](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide) Wiki page and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jgromes, 2024-07-13:
I think the reason I originally did not include these modes is that they use YCbCr color as opposed to RGB, and also because of their asymmetric timing. Especially Robot36, which averages the Cb/Cr components over two lines, which is not really compatible with RadioLib one-line-at-a-time approach.

That being said, I was able to add some support for it and it seems to be working. Fair warning, I'm not doing the above-mentioned averaging, which leads to the colors being slightly off. However, the image is still pretty clear: see below, Robot36 sent via SX1278.

[image]

closed 2024-07-13 by jgromes

Panger95, 2024-07-20:
Do you remember which filter bandwidth, if it was USB, Narrow FM, or LSB.
Also which software did you use to decode it? Did you use QSSTV?

jgromes, 2024-07-20:
I used the SSTV in AFSK mode, so FM - I don't recall whether it was narrow or wide but either one should work. For decoding I used MMSSTV.