jgromes/RadioLib #1174: [LR1121]: Strange SPI behavior

Tipetofficial, 2024-07-30:
I am using the LR1121 on an own PCB with an ESP32-S3-WROOM-1U and I keep getting the code -2 for a bad SPI-Interface when using the [LR11x0_Firmware_Update](https://github.com/jgromes/RadioLib/tree/master/examples/LR11x0/LR11x0_Firmware_Update). I checked my wiring (see schematic) and followed the Non-standard SPI setup as u can see in the code below. 

As displayed in the basic debug output the red value and the expected value are the same, but it still fails.

I would highly appreciate any hints to how I could resolve this issue.
### **Schematic**
[Schematic_black.pdf](https://github.com/user-attachments/files/16423946/Schematic_black.pdf)
### **Code that fails** 
```c
#include <Arduino.h>
#include <SPI.h>
/*
  RadioLib LR11x0 Firmware Update Example

  This example updates the internal LR1110 firmware.
  Newer versions of the firmware introduce fixes
  and possibly even new features, so it is recommended
  to use the latest available firmware version
  when possible.

  Other modules from LR11x0 family can also be used.

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
//#include "LR1121.h"
// select the firmware image you want to upload
// WARNING: Make sure you select the correct firmware
//          for your device! Uploading incorrect firmware
//          (e.g. LR1110 firmware to LR1120 device)
//          may damage your hardware!
//#define RADIOLIB_LR1110_FIRMWARE_0303
//#define RADIOLIB_LR1110_FIRMWARE_0304
//#define RADIOLIB_LR1110_FIRMWARE_0305
//#define RADIOLIB_LR1110_FIRMWARE_0306
//#define RADIOLIB_LR1110_FIRMWARE_0307
//define RADIOLIB_LR1110_FIRMWARE_0401
//#define RADIOLIB_LR1120_FIRMWARE_0101
//#define RADIOLIB_LR1120_FIRMWARE_0102
//#define RADIOLIB_LR1120_FIRMWARE_0201
//#define RADIOLIB_LR1121_FIRMWARE_0102
#define RADIOLIB_LR1121_FIRMWARE_0103
// enable this macro if you want to store the image in host
// MCU RAM instead of Flash.
// NOTE: the firmware images are very large, up to 240 kB!
//#define RADIOLIB_LR1110_FIRMWARE_IN_RAM

// include the firmware image
#include <modules/LR11x0/LR11x0_firmware.h>

#define SPI_MOSI  11
#define SPI_MISO  10
#define SPI_CLK   9 
#define SPI_CLOCK_FREQ 2000000
#define NSS       8

/* 
SPIClass* g_spi;

void spiinit()
{
  g_spi = new SPIClass(HSPI);
  g_spi->begin(SPI_CLK, SPI_MISO, SPI_MOSI, NSS);
  pinMode(g_spi->pinSS(), OUTPUT);  
}

Module* mod;
[... 235 lines cut ...]
RLB_DBG: LR11x0 not found! (1 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (2 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (3 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (4 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (5 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (6 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (7 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (8 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (9 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: LR11x0 not found! (10 of 10 tries) RADIOLIB_LR11X0_CMD_GET_VERSION = 0x03<\n>
RLB_DBG: Expected: 0x03<\n>
RLB_DBG: No LR11x0 found!<\n>
failed, code -2<\r><\n>
```
### **Output Logic Analyser**
[image]

jgromes, 2024-07-30:
This is interesting, since the excepted and read-out values match, the only way for the radio presence check to fail is if the method `LR11x0::getVersion` fails itself. There's probably some subtle difference between LR1110 (which I used to write the LR11x0 driver) and the LR1121 in that regard that is causing this method to fail. Unfortunately I'm away from most of my hardware at the moment, so this will take a couple of days for me to look into.

jgromes, 2024-08-04:
@Tipetofficial I went through the log once more and noticed that a command to read WiFi scanning formware version is being called. For LR1121 that will always fail, since it does not have that functionality. A check to fix this was added in 87fac1797e8e8ef2a8c6cffb722e4212c3050a19, but was not released yet. Could you please try this again with the latest master?

Tipetofficial, 2024-08-07:
@jgromes Thank you for the fast answer, the master fixed this problem but now I get the -707 error. I am using a TCXO and added 

```c
radio.setTCXO(2.4);
```

because the TCXO I use need at least 1.7V rather that the default 1.6V but the error stays the same.

jgromes, 2024-08-07:
You have to pass the TCXO voltage as an argument to the begin method, otherwise it will fail before it has a chance to initialize.

Tipetofficial, 2024-08-12:
Thank you. With your help I was able to fix the issue.

closed 2024-08-12 by Tipetofficial

klabshowroom, 2024-11-03:
@Tipetofficial Hi, I am struggle for LR1121 with Semtek EVK, would you please give me a hand to find my problem.
[image]