jgromes/RadioLib #1181: complete wiring information for SPX1276 LoRa

alehvalelabs, 2024-08-06:
I cannot find a sample wiring diagram for LoRa SPX1276 to correctly connect to a MEGA.

The sample sketch: SX127x_Transmit makes no mention of the SPI connections. Are these also required? if so does the SPI.h library also need to be added to the sample sketch.

This sketch also uses pin 0 which is the RX0 pin for UART0.

jgromes, 2024-08-06:
> I cannot find a sample wiring diagram for LoRa SPX1276 to correctly connect to a MEGA.

Then I would suggest proceeding in the following order:

1. Finding out which communication interface the SX1276 has.
2. Locating said interface on your board - Google is your friend here.
3. Connecting them, keeping in mind that the logic voltage of your board and the SX1276 will likely be different. If you connect them without some level conversion, you may damage the SX1276.

> The sample sketch: SX127x_Transmit makes no mention of the SPI connections. Are these also required?

Obviously so, the sketch does not mention them since on every microcontroller they will be at different pins, and sometimes even configurable. It seems pointless to state in every example "Make sure you have the SPI pins connected"; if you need such a note I would highly suggest to start with a simpler project.

> if so does the SPI.h library also need to be added to the sample sketch.

No, it is added by RadioLib. Compilation of all example sketches is checked automatically on every release, which would have caught a missing library.

> This sketch also uses pin 0 which is the RX0 pin for UART0.

Where?

alehvalelabs, 2024-08-07:
Thanks for answering my questions.

 

I am using an Arduino MEGA. I am following the pin numbers that are quoted in the sample sketch:

 

// SX1276 requires the following connections:

int pin_cs = 41;

int pin_dio0 = 0;  // was 0

int pin_nrst = 33;

int pin_dio1 = 34;

SX1276 radio = new Module(pin_cs, pin_dio0, pin_nrst, pin_dio1);

 

I used these pin numbers on the MEGA. This is where I used MEGA pin 0 to connect to SPX1276 di0. On the MEGA, pin0 is the UART0 RX.

 

With the information that you have provided, I will investigate more correct wiring details.

jgromes, 2024-08-07:
> This is where I used MEGA pin 0 to connect to SPX1276 di0

That will not work - it will interfere with your Serial port and also prevent you from using interrupt driven transmission and reception. Seems quite pointless on the Mega which has some 50 other digital pins, 6 of which support interrupts.

Since this is not actually an issue, I will convert this to a discussion instead.