jgromes/RadioLib #1183: SX127x in continous mode has packet<>length mismatch

mycosd, 2024-08-09:
Hi,

I'm using the SX1278 (but the whole series has this issue) in LoRa mode. When using the continuous operation mode and an interrupt to then pull the received data, there are cases where you can receive multiple packets before actually reading the last one. 

The `getPacketLength` returns the length of the last packet, but the read pointer in the FIFO is still at the previous packet. So when you read the packet either using `getPacketLength` or 0 as length for `readData`, you are actually reading the previous packet with the wrong length and all packets after that get scrambled as well.

Here is a very exaggerated example

TX:
```
#include <Arduino.h>
#include <RadioLib.h>

#define LORA_PIN_NSS    5
#define LORA_PIN_DIO0   4
#define LORA_PIN_RESET  17
#define LORA_PIN_DIO1   16

SX1278 radio = new Module(LORA_PIN_NSS, LORA_PIN_DIO0, LORA_PIN_RESET, LORA_PIN_DIO1);
uint8_t count = 0;

void setup() {
    Serial.begin(115200);
	radio.begin();
}
void loop() {
    String s(++count);
    if (count & 1)s += "x";
    radio.transmit(s);
    Serial.println(s);
    delay(1000);
}
```
RX:
```
#include <Arduino.h>
#include <RadioLib.h>

#define LORA_PIN_NSS    5
#define LORA_PIN_DIO0   4
#define LORA_PIN_RESET  17
#define LORA_PIN_DIO1   16

SX1278 radio = new Module(LORA_PIN_NSS, LORA_PIN_DIO0, LORA_PIN_RESET, LORA_PIN_DIO1);
volatile bool receivedFlag = false;

#if defined(ESP8266) || defined(ESP32)
ICACHE_RAM_ATTR
#endif
void setFlag(void) {
    receivedFlag = true;
}
void setup() {
    Serial.begin(115200);
    radio.begin();
    radio.setPacketReceivedAction(setFlag);
    radio.startReceive();
}
void loop()
{
    if (receivedFlag)
    {
        receivedFlag = false;
        String s;
        if (radio.readData(s) == RADIOLIB_ERR_NONE)
            Serial.println(s);
    }
    delay(2000);
}
```

I think in continous mode, the library should follow the datasheet and advance the FIFO read pointer to the current packet. That way the packet length actually matches the data:
> Set RegFifoAddrPtr to RegFifoRxCurrentAddr. This sets the FIFO pointer to the location of the last packet received in
the FIFO. The payload can then be extracted by reading the register RegFifo, RegRxNbBytes times.
Alternatively, it is possible to manually point to the location of the last packet received, from the start of the current
packet, by setting RegFifoAddrPtr to RegFifoRxByteAddr minus RegRxNbBytes. The payload bytes can then be read
from the FIFO by reading the RegFifo address RegRxNbBytes times.

I've tested this, by inserting the following line into `readData` before the FIFO is getting read:
```
auto start = this->mod->SPIgetRegValue(RADIOLIB_SX127X_REG_FIFO_RX_CURRENT_ADDR);
this->mod->SPIsetRegValue(RADIOLIB_SX127X_REG_FIFO_ADDR_PTR, start);
  ```
You'll lose some packets but at least you get the most recent and every following packet is correct.

Regards
Maik

jgromes, 2024-08-09:
This seems similar in concept to what was reported in #1159 (though that one is about the SX126x series). It will happen in case multiple packet are recieved without user handling the Rx interrupt. That will also happen in your example, since after each read there is a 2 second wait (and packets are being sent with 1 second delay).

I was not aware that the SX127x had the option to manipulate the FIFO pointer - probably missed it in the datasheet.

The solution proposed in that discussion by @GUVWAF was I think the same, that is: to advance the pointer to read out the latest received packet. I will happily merge a PR fixing this.

GUVWAF, 2024-08-09:
Indeed it looks the same as for the SX126x. 

However, I ran into the issue that the returned pointer address was always non-zero if you don't go to `standby()` in between, but the data does actually start at zero when setting `setBufferBaseAddress()`. I believe it would work correctly when not setting `setBufferBaseAddress()` in `readData()`, and then always reading from the returned address. I didn't have time yet to test this thoroughly (especially with and without going to `standby()` in between).

It looks like for the SX127x it automatically resets it to 0 if you read the data, so then this approach would work.

mycosd, 2024-08-09:
> It looks like for the SX127x it automatically resets it to 0 if you read the data, so then this approach would work.

The SX127x has a ringbuffer, the read and write pointers move forward one byte at a time when reading/writing respectively. Although the problems both issues are causing look the same, they are caused by 2 different issues, as I understand it. For the SX127x it's caused by the read pointer not being advanced because of a missing read, for the SX126x it's because write pointer is not reset to 0, while the read address being fixed at 0. It probably should be reading from `RxStartBufferPointer`, that appears to be the base base for `PayloadLengthRx` returned from `getPacketLength`. I agree, the base address shouldn't be reset, in the that solution... in fact it would operate nearly the same way as the SX127x.

jgromes, 2024-08-09:
@mycosd thank you very much for the fix, it was merged.

@GUVWAF feel free to open a PR for the SX126x issue once you have the time :)

closed 2024-08-09 by jgromes