jgromes/RadioLib #1192: BW16 RTL8720DN  support

test3210-d, 2024-08-20:
#warning "RadioLib might not be compatible with this Arduino board - check supported platforms at https://github.com/jgromes/RadioLib!"

StevenCellist, 2024-08-20:
Now that's a new board for me, interesting.
But a warning is just a warning, not an error. Have you tried using it? What is the behaviour?

test3210-d, 2024-08-21:
trouble one SPI_MODE0 not defined 
easy fix - #define SPI_MODE0 0
trouble two 
error about yield()
in basic code for rtl8720dw 
AmebaD\3.1.7\cores\ambd/Arduino.h:55:18: error: expected unqualified-id before '{' token
 #define yield(x) {}

I remove all about yield using, work fine

jgromes, 2024-08-21:
@test3210-d with what you're describing, it looks like a dedicated section in BuildOpts.h would be needed to support this MCU. 

> I remove all about yield using, work fine

What does that mean? Does it compile, or does it run on the actually hardware, correctly transmitting/receiving etc.?

> Hardware like esp 8266 but can work 2.4 and 5.8 Ghz

It might looks like an ESP8266 module, but the internals are entirely different. The RTL8720DW seems to be a dual core ARM-Cortex M, whereas ESP8266 uses Xtensa core, it's a completely different architecture.

There's aan Arduino core for it, but it doesn't seem very popular, so I'm a bit worrie about long-term support of that project: https://github.com/ambiot/ambd_arduino

HeadBoffin, 2024-08-21:
Which radio that RL supports are you going to drive with this board? Just wondering if you are hoping for WiFi or BLE to be used?

test3210-d, 2024-08-21:
yes 5.8 wifi to lora gate

StevenCellist, 2024-08-21:
I am very much wondering what usecase requires blazing-fast 5.8GHz WiFi being relayed through snail-paced LoRa..? If you could care to explain, I'm very much interested.

HeadBoffin, 2024-08-21:
> yes 5.8 wifi to lora gate

That's the overview - what is the actual radio you are thinking of attaching to your board?

test3210-d, 2024-08-22:
Sx1276

jgromes, 2024-09-06:
There has been no update explaining what is this support needed for, so it seems to me like this is a case of "wouldn't it be cool if". In addition, it looks like the Arduino core does not adhere strictly to Arduino API specification, as evident by the missing SPI defines and absence of `yield`. I will close this as not planned, if in future support is requested by more people, or if the board becomes more widespread, I will consider adding it.

closed 2024-09-06 by jgromes