jgromes/RadioLib #1201: LR1121 Custom PCB without SKY13588-460LF

gc9n, 2024-08-30:
Hi there , 
i use a LR1121 , and in my design i do not use SKY13588-460LF  Switch , 
as an MCU i use STM32L4  

I can talk to LR1121 , i can update the Radio firmware , i can send all commands , however , the RSSI is stuck on -127, 
i assume that In order to Set LR1121  in MODE_RX    DIO5 have to be HIGH  DIO6 have to be Low ,   DIO5 and DIO6 is handled by SKY13588...

any help?

jgromes, 2024-08-30:
I can't really offer any guidance on the particular RF switch you have selected. It is possible the -127 value is caused by some problem in the library. However, to help you further, I will need the information requested by the issue template, mainly the code you are running and all output from the serial console.

gc9n, 2024-08-31:
Hi again , and i am sorry i double checked with the Evaluations Board of LR1121 SEMTECH     
and With My design ,  and i found the issue.
my Design ignores this command , 

```
lr11xx_status_t lr11xx_radio_set_rx_with_timeout_in_rtc_step( const void* context, const uint32_t timeout )
{
    lr11xx_status_t status                                  = LR11XX_STATUS_ERROR;
    const uint8_t   cbuffer[LR11XX_RADIO_SET_RX_CMD_LENGTH] = {
        ( uint8_t ) ( LR11XX_RADIO_SET_RX_OC >> 8 ),
        ( uint8_t ) ( LR11XX_RADIO_SET_RX_OC >> 0 ),
        ( uint8_t ) ( timeout >> 16 ),
        ( uint8_t ) ( timeout >> 8 ),
        ( uint8_t ) ( timeout >> 0 ),

    };
```

or if you prefer better this 
```
int16_t LR11x0::setRx(uint32_t timeout) {
  uint8_t buff[3] = {
    (uint8_t)((timeout >> 16) & 0xFF), (uint8_t)((timeout >> 8) & 0xFF), (uint8_t)(timeout & 0xFF),
  };
  return(this->SPIcommand(RADIOLIB_LR11X0_CMD_SET_RX, true, buff, sizeof(buff)));
}
```

i have the exact behavior in Semtech board when i dont set the LR1121 to RX mode.  which is dead RSSI ,i change channels etc but the RSSI is dead. is -127.  its not in RX mode , its on the default state i guess standby

i want to confirm if its a HArdware issue

jgromes, 2024-08-31:
Can you post the code you are running, as I have asked in my previous post? I'm guessing you are calling `LR11x0::getRSSI`, but since you ahve not posted your code, I cannot be sure.

I also invite you to examine the RadioLib documentation for `LR11x0::getRSSI`:

```c++
    /*!
      \brief Gets RSSI (Recorded Signal Strength Indicator) of the last received packet. Only available for LoRa or GFSK modem.
      \returns RSSI of the last received packet in dBm.
    */
    float getRSSI() override;
```

The comment clearly says "of the last received packet". If you have yet to receive a packet, then I don't know what will the result be.

jgromes, 2024-09-14:
Closed due to inactivity.

closed 2024-09-14 by jgromes