jgromes/RadioLib #1212: [LoRaWAN] Getting the maximum payload size using ADR

rtamse12, 2024-09-10:
The maximum payload size in LoRaWAN depends on the current data rate. When using ADR, the data rate may be set by the network server. How do I get the current data rate or the corresponding payload size so that I may fragment my payload accordingly?

StevenCellist, 2024-09-10:
Ah, it is called `maxPayloadDwellTime()`, but is bugged anyway. Your solution sort-of works but indeed definitely isn't a good idea (and also doesn't account for the LoRaWAN header with optional FOpts). The next release of RadioLib (quite soon!) will fix this problem with an actual implementation of `maxUplinkLen()`. Thinking about it, the name `maxPayloadLen()` may be more appropriate after all.

rtamse12, 2024-09-10:
I'm looking forward to it. Thanks for the help.

StevenCellist, 2024-09-10:
Resolved in master.

closed 2024-09-10 by StevenCellist