jgromes/RadioLib #1219: LoRaWAN, uplink fails when no downlink on previous uplink

MortenGuldager, 2024-09-15:
LoRaWAN Uplink fail if previous uplink didn't respond with a downlink.

I have an ESP32C3 wired to a RFM95 module. Keys are installed on a chirpstack server and communication initially flows fine, uplinks are seen on server and downlinks are processed too. I can see ADR is doing its thing.
Then when ADR is completed and no more downlinks are sent, the following uplinks fail with  -1108 (RADIOLIB_ERR_UPLINK_UNAVAILABLE)

I can delay this lock-up by keep queuing downlink messages.

I notice this is only a problem if I dont enter deep-sleep between cycles.

Mainloop is simple (the increasing fPort is just to make it easier to track on the LNS):

```
void loop()
{
    static int fport = 50;
    
    String str = "ABCDEF";
    int16_t rl_err;

    rl_err = node.uplink(str, fport);

    Serial.print(fport);
    Serial.print(" : tx res: ");
    Serial.println(rl_err);

    node.downlink();

    delay(30000);
    fport++;
}
```
the full example code is attached if anybody want to try it out
[main.txt](https://github.com/user-attachments/files/17006262/main.txt)

My environment is: ESP32, Arduino framework (Platform.io), RFM95, jumperwires, chirpstack, EU868, RadioLib:jgromes/RadioLib@^6.6.0

MortenGuldager, 2024-09-15:
I have just reproduced the error on a different set of hardware:

- different RFM95 module
- an esp32 wroom dev kit

Code is the same except for new pin mappings

HeadBoffin, 2024-09-16:
I'll replicate your setup to reproduce, will be easier to comment then.

MortenGuldager, 2024-09-16:
New observation:
I upgraded from 6.6.0 to "master" and after making minor changes to get my application to build, it now seems to keep going downlinks or not.
My `platformio.ini`
```
lib_deps = 
    ; jgromes/RadioLib@^6.6.0
    https://github.com/jgromes/RadioLib.git#master
```

StevenCellist, 2024-09-16:
As a new RadioLib version is upon us (as in: a few hours, likely), I was honestly waiting for the release and would then ask you to upgrade ;)
Glad to hear that upstream resolves this issue. Those are the easiest issues to fix!

@HeadBoffin I wouldn't bother testing it.

StevenCellist, 2024-09-16:
Resolved with RadioLib 7.0 being released.

closed 2024-09-16 by StevenCellist

MortenGuldager, 2024-09-17:
And I can confirm that it indeed also works with an official 7.0.0.