jgromes/RadioLib #1224: Raspberry PI 4 100% CPU

longtomin, 2024-09-22:
Hello,

I found this Lib because I use it at the ESP32S3 Boards and was thinking why nor use it at the Raspberry PI as well.
Good work and it is working without any bigger issue and long explanations.

I wrote a receiver program for the PI using cmake and the libs. Everything is working find.
The only thing is, I use the PI for other things as well and I recognized 60° C or more.
So I found that the Receiver is running with 100% or a little bit more CPU usage, see:
[image]

Question: Is it possible to send the program into a sleep mode which is just processing when data is arrived?
I looked for the radio.sleep and radio.standby functions.
I have this hat for the PI: [WaveShare Lora Hat](https://www.waveshare.com/wiki/SX1262_XXXM_LoRaWAN/GNSS_HAT)

jgromes, 2024-09-22:
> I wrote a receiver program

Withtout sharing that program, it is not possible for me to tell you why is it behaving in this way. The library is just a tool, CPU utilization is a direct result of how that tool is being used.

longtomin, 2024-09-22:
Thanks for the quick reply.
Here is my code:
```
/*
   RadioLib SX126x Receive with Interrupts Example

   This example listens for LoRa transmissions and tries to
   receive them. Once a packet is received, an interrupt is
   triggered. To successfully receive data, the following
   settings have to be the same on both transmitter
   and receiver:
    - carrier frequency
    - bandwidth
    - spreading factor
    - coding rate
    - sync word

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
// include the hardware abstraction layer
#include "PiHal.h"

#include <cstring>
#include <cstdlib>
//#include <stdio.h>
//#include <string.h>
//#include <stdlib.h>
#include <time.h>
#include <curl/curl.h>

#define FREQUENCY 868.0
#define BANDWIDTH 125.0
#define SPREADING_FACTOR 7
#define TRANSMIT_POWER 0
#define CODING_RATE 5

// ------------------- LoRa  Sending & Receiving Paramaters --------------------------

#define OWNID 0xAA
#define SRCAUSSEN 0xB1 // Sender 1 Aussen Süd
#define SRCBAD 0xB2 // Sender 2 Bad
#define SRCVERONIKA 0xB3 // Sender 3 Veronika
#define SRCALENA 0xB4 // Sender 4 Alena
#define SRCSZ 0xB5  // Sender 5 Schlafzimmer
#define SRCNORD 0xB6  // Sender 6 Aussen Nord

#define ACKNOWLEDGE 0x414B

static const uint8_t PH_packetSize = 20;
static const uint8_t PH_ackPacketSize = 6;

union PH_dataPacket {
  uint8_t packetByte[PH_packetSize];
  struct packetContent {
    uint8_t dest;             // B // 1 byte
    uint8_t source;           // B // 1 byte
    uint16_t sequenceNumber;  // H // 2 bytes
    int16_t temperature;      // h // 2 bytes Temperature x 100, °C 2 decimal digits
    uint16_t humidity;        // H // 2 bytes %  x 100, °C 2 decimal digits
    uint32_t pressure;        // I // 4 bytes kPa x 100, 2 decimal digits
    uint32_t gas;             // I // 4 bytes unknown  x 100, °C 2 decimal digits
    uint16_t voltage;         // H // 2 bytes V  x 100, °C 2 decimal digits
  } content;
[... 231 lines cut ...]
          sendAcknowledge();
          // Continue receiving or listening
          state = radio.startReceive();
          saveData(myRSSI, mySNR, myFQE);
//          printf("Radio.startTransmit with code  #%d\n", state);
          // Just for me to see something
//          printRawPacket();
//          printPacket();

          // print RSSI (Received Signal Strength Indicator)
          printf("RSSI:\t\t%0.2f  dBm\n", radio.getRSSI());
          // print SNR (Signal-to-Noise Ratio)
          printf("SNR:\t\t%0.2f  dB\n", radio.getSNR());
          // print frequency error
          printf("Frequency error:\t%0.2f  Hz\n", radio.getFrequencyError());
        }
      } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
        // packet was received, but is malformed
        fprintf(stderr, "CRC error!\n");
      }
    }
  }
}
`
It is just a receiver running on the PI with an endless loop. I used your examples/NonArduino/Raspberry example to set it up. As mentioned bevor, It works really good.

0x975, 2024-09-22:
Hi @longtomin I believe that the reason is the difference in pooling between ESP32 and Raspberry.

adding something like hal->delay(1000); to your endless loop should solve some your issue

jgromes, 2024-09-22:
@longtomin now it's clear, you have an infinite loop which can always execute (since the only thing it does is check the flag set from ISR), there is no point at which the thread the proggram is running in stops being able to execute. And since the RPi has nothing else to do, that loop is always being run. PS: I edited your post so the comment isn't that long. Feel free to take a look how I did that.

The solution is to adapt your program to take into account the fact that you are now running in Unix-based OS instead of an MCU like ESP32. For example, by using threads from the POSIX `pthread` library, and having a thread that waits for a signal from the radio. The example doesn't include that because it's well out of scope of this library.

@longtomin

> adding something like hal->delay(1000); to your endless loop should solve your issue.

That's not a good idea. It will appear to help, since the thread running the program will now stop itself for 1000 ms (approximately, sicne we're in non-realtime OS). But it will also mean that you will always have at least 1 second delay before each packet is received.

This is not an issue in the library, but rather a question about usage, hence I will convert it to discussion now.