jgromes/RadioLib #1233: Issue with AFSK on the Arduino Uno

gorsheep, 2024-09-25:
Hi there,

I am trying to run the "AFSK_Tone.ino" example on the Arduino Uno and the SX1278 module. Here’s how I have my SX1278 wired to my Arduino Uno:

SX1278 Pin --> Arduino Uno Pin
3V3 --> 3V3
GND --> GND
RESET --> 9
NSS --> 10
DIO0 --> 2
DIO1 --> 3
DIO2 --> 5
MOSI --> 11
MISO --> 12
SCK --> 13

FYI, I am able to run the “APRS_Position_LoRa.ino” example. I can see on the waterfall plot that it’s actually transmitting some kind of data. I haven’t tried actually decoding the data but the waterfall looks good (so I assume it works properly).

However, when I run the “AFSK_Tone.ino” example, it doesn’t quite work as expected. It makes the 400 Hz tone initially, then silence for a second, then the tone for a second, but then it remains silent forever. If I comment out the block responsible for silence, it will transmit the 400 Hz tone, then silence for a second, then again transmit the tone non-stop. It is able to produce tones of different frequencies (400 Hz, 800 Hz) individually, so AFSK must be working properly. Is there anything glaring I'm missing? Thanks!

jgromes, 2024-09-26:
I would suggest checking the result of the compilation. You will likely see a very high utilization of program and data memory, along with a note that a low-end platform has been detected. The reason is quite simple, Arduino Uno does not have enough RAM/flash to run the latest versions of the library. I don't know which version you are running, since you have not filled in the issue template. But I'm guessing it's quite recent. So you can try to downgrade to older version of the library, or move onto a more recent microcontroller (as in, one that has been designed in the last 20 years, unlike the ATmega328).

Also, Uno has 5V logic, and SX1278 as 3.3V logic, so you will likely damage it eventually.

gorsheep, 2024-09-26:
Thank you for your response! Yes, indeed, memory utilization is quite high after compilation. And yes, I'm using the latest version of the lib. I will try using an older version of the lib and also a newer microcontroller. Are there any specific Arduino-compatible microcontrollers that you would recommend?

jgromes, 2024-09-27:
I tried to replicate this on an ESP32 with 400 Hz and 800 Hz tone, you can see the result in the screenshot below. The tones are being received on an SDR in NFM mode. You can see a 400 Hz peak in the audio spectrum. So as far as I can tell, on the ESP32 it works. There's also been a recent issue (#1230) talking about APRS using AFSK on RPi Pico, which has been resolved and verified working.

I'm not sure about the limitations of PWM/tone on Arduino Due, maybe that would be a good place to start.

[image]

jgromes, 2024-10-12:
Closed due to inactivity.

closed 2024-10-12 by jgromes